dangowrt/owrt-ubi-installer #2: Question:

BarryHuffman, 2021-04-03:
What would be the difficulting of reformatting the NAND for UBI, with OpenWRTs bootloader (uboot?) to load a stock OS payload?

The intention is to prevent future OEM updates blocking installation of OpenWRT via bootloader or signed firmware, and to get the performance of the stock OS while OpenWRT's 802.11ax support matures.

Additionally TRIM(via UBI) also extends the life of the underlaying flash.

A similar concept exists for the Nintendo 3DS called EmuNAND where you flash a custom bootloader to the NAND which chainloads the OEM bootloader which itself loads the default OS.

dangowrt, 2021-04-03:
Stock OS can only do JFFS2, it's not aware of UBI at all and hence cannot run on UBI formatted flash without destroying the UBI structures (ie. it will boot exactly once, next reboot U-Boot will fail to load anything from UBI and fall back to TFTP).
You can try to build stock OS from source and switch on UBI in the kernel and do the necessary changes -- UBI has been around for more than a decade, so support for it should be present in stock kernel sources (just not enabled).
I guess on the 3DS applications are using some OS API to access storage (rather than Linux kernel talking to the hardware directly)...

BarryHuffman, 2021-04-03:
Thanks for the clarifications and input.

closed 2021-04-03 by BarryHuffman