dangowrt/owrt-ubi-installer #6: I only see two problems.

Jefe777, 2021-04-13:
1. The script does not autodetect if we already have UBI to only update the UBOOT or bootloader when the installer is re-run, without the need for re-formatted and loose `system` and `rootfs_data` content and if it is necessary to re-formatted, then temporarily move the content of `system`, `rootfs_data` and `boot_backup` to another place and after update the bootloader put their content back there.

2. You need another script or firmware that will automatically do the steps you mentioned **to go back to stock firmware**, for people who want to go back to stock firmware for X reason and then the stock firmware can be installed without problems or the same script installs the stock firmware.

Call it: `openwrt-mediatek-mt7622-linksys_e8450-stock-recovery-firmware-installer.itb`

**A script or firmware that does this automatically:**
The installer stores a copy of the previous bootchain in a dedicated UBI volume **boot_backup**. If you want to go back to the vendor firmware, you will have to boot into recovery mode (ie. initramfs), copy those files into **/tmp**, umount the UBI volume, detach the UBI device and then write the files to the corresponding MTD partitions (**mtd write $file /dev/mtdX**).

I hope you solve these serious problems and thank you for your work.

dangowrt, 2021-04-13:
Regarding updating U-Boot and/or ARM Trusted Firmware later on:
This can already be done using the bootloader menu when connecting via the serial console.
If you want to trigger downloading updated `bl31` image via TFTP you may also do so by telling the bootloader to do that on next reboot and provide the file via TFTP.
Assuming the router is connected to `eth0` of another machine running Linux, have `tftpd-hpa` installed and do this on the host:
```
tmpdir=$(mktemp -d)
chmod 0755 $tmpdir
cd $tmpdir
wget https://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/openwrt-mediatek-mt7622-linksys_e8450-ubi-bl31-uboot.fip
chmod 0644 
sudo ip addr flush dev eth0
sudo ip addr add 192.168.1.254/24 dev eth0
sudo in.tftpd -L -v -s .
```

On the router running OpenWrt do this:
```
fw_setenv bootcmd "run boot_tftp_write_fip && run reset_factory ; reset"
reboot
```

Regarding reverting to stock bootchain:
I also don't really need that functionality myself, just wanted to keep the doors open for anyone else who might. Writing the mtd partitions back to the flash should be done in U-Boot as mtd0 and mtd1 are read-only in OpenWrt (and that's how it should be imho). Mounting the boot_backup UBIFS volume in U-Boot should work and you can use `ubifsload` to load all the files from there into the RAM, then unmount UBIFS and detach UBI, then write to flash using `mtd` command.
Providing a downloadable image containing the complete stock firmware is not possible because even though most part of the stock firmware is free open source software, some parts are not and it is not possible to build the entire vendor firmware from source (as far as I know, I haven't tried).

Jefe777, 2021-04-13:
The most serious problem that this router has is that you have to do 100 juggling with serial console, linux, to return to stock firmware.

You must understand that there are people who do not have the knowledge to do that and do not have the tools.

**If you can make it possible to revert to stock firmware** without the need of a serial console or linux, just with a script as mentioned above (`openwrt-mediatek-mt7622-linksys_e8450-stock-recovery-firmware-installer.itb`) though just roll back the bootloader to stock and then install the official or stock firmware from the web ui, it would be one of the best all-in-one router that one can recommend for OpenWrt.

**As long as you don't achieve that, this router is for a small niche.**

dangowrt, 2021-04-13:
Fine with me, I published this because I needed it for myself. If it is useful for other people, that's great. If other want to contribute to it, even better. However, I'm not going to deal with the vendor firmware at all, even if some folks may consider that a missing feature. This is free software, not a product, I'm not trying to sell anything.

Jefe777, 2021-04-13:
**Your work is half finished**, it just allows you to easily install the OpenWrt firmware, but it is too difficult to go back to the original firmware or update the bootloader, because according to you it is ok (?) and you are angry that someone tells you the truth and show you the way.

I was just giving advice, I am not forcing you, **I wanted you to understand that you did a great job**, **but it is half finished,** I hope you rectify it, because only you are the only one who can finish the job.

dangowrt, 2021-04-13:
I'm sorry if I sounded angry to you. I do appreciate you testing my code and giving feedback.
Updating to bootloader is not a problem if actually needed. I explained in my reply how you can do that by providing a TFTP server and entering two lines in the console. I don't see a need to provide images to update the bootloader right now, even people who put the installer very early already got ATF 2.2 and U-Boot 2021.01, both with MTK patches, and really not much has happened since for devices which already work well with previous versions (you can and should of course update OpenWrt). Running the installer again (using `sysupgrade -f`) is also possible, you will just loose previous UBI content and start off from scratch (which, in case there were actual relevant changes on the bootloader, you would possibly have to do anyway; I'm planning on supporting UBI in ATF bl2 -- that will overthrow the flash layout yet another time and re-installing means loosing all flash content).
This is just my opinion and you may not share it, of course.

Regarding being the only one who can "finish" the job: All this if open source, in most parts even written in the single most popular scripting language on the planet (Shell). You are very welcome to contribute any feature you desire to it or hire other people to do so if you don't want to do that yourself. I will of course help and respond to any questions. Because this is how free open source software works. I had an itch (needing to easily install OpenWrt with up-to-date bootchain on a bunch of those devices) and I scratched it. I share the results, some documentation and allow others to contribute to it. I do not have the need myself and don't consider it relevant to use stock firmware (ever, on any device). I'm not denying that other people may feel different about that. It's just that this hasn't been my motivation (selling something or pleasing anyone) in first place and hence the argument misses the point.
If this is incomplete for you, press the **Fork** button and complete it, so it's complete for you. I will help you, if needed, up to some level free of charge (because I expect the same from other people and experience a similar level generousness around me). However, I will not respond as if this was commercial product and you were the customer.

BarryHuffman, 2021-04-28:
I'm actually having issues with a "auc" upgrade that resulted in semi-bricked device. (disabled the option to restore via booting and holding the reset button, removed holding reset while powered on, and also SSH/Luci).

Prior to that I had been manually updating OpenWRT UBI-snapshots in Luci no issues.
The latest release I decided to try "auc" since I was already within a SSH connection at the time and once it said it finished I saw a message along the lines of "initializing sysupgrade" or similar, and both the Power LED and Internet LEDs turned orange.

Waiting half a hour to see if it was still writing and then I tried the reset button to allow it to reboot which it didn't respond so I did a power cycle which never recovered to the normal state despite multiple attempts.

I've hooked up serial, and tried to enter "4. Boot recovery system from flash." in the boot menu but it said none existed.

[Collected serial output](https://github.com/BarryHuffman/Output/blob/master/RT3200/UART-boot)

Is there a specific procedure you'd suggest going forward?
I want to restore OpenWRT and I care little about Stock firmware.

Would it be possible to use the "usbboot" command in u-boot with a recovery initramfs and load it externally from USB as alluded to in [release v3](https://github.com/dangowrt/linksys-e8450-openwrt-installer/releases/tag/v0.3) "USB 2.0 fixed, nice for UEFI-booting from USB"

I'm not familiar with ubi commands is there anything useful for debugging to display more info beyond "ubi part" ? 
I'd also like to help document the recovery method for others incase they fall in the same scenario.
(Also writing up a safe method to disassemble the RT3200 case without damage)

dangowrt, 2021-04-29:
@BarryHuffman I'll answer one by one:
I don't understand how `auc` managed to destory UBI during sysupgrade (as that is what seems to have happened, as recovery got missing as well). Sounds like reset hits while write hasn't yet completed or something like that... It never happened to me and I've been using `auc` almost daily for more than a month on my device. Maybe just try again and capture serial output of what's happening during the update, so we know more...?

The easiest way to recover at this point is to connect any of the LAN ports of the router to your PC, configure the PC to have static IP `192.168.1.254` netmask `255.255.255.0`. Then run TFTP server of your choice and let it serve [openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery.itb](https://github.com/dangowrt/linksys-e8450-openwrt-installer/releases/download/v0.4/openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery.itb). Then choose
```
Load recovery system via TFTP then write to flash.
```
in the bootmenu. This will write recovery back to flash and then boot it. From there you can then install the production firmware  [openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb](https://github.com/dangowrt/linksys-e8450-openwrt-installer/releases/download/v0.4/openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb) just like after running the installer.

BarryHuffman, 2021-04-29:
@dangowrt 
Thanks for your assistance!

I'm not quite sure what happened as the only variations I have beyond standard OpenWRT were luci-nginx and using openssh instead of dropbear.

The recovery method worked perfectly, I've done TFTP recoveries before but was hesistant with UBI, and not aware if it would trigger ARM Trust.

I'm going do some more testing with `auc` in the meanwhile but I'd be interested in helping test future recovery methods.
I'm also curious if the U-Boot shell command `usb` and `usbboot` could be utilized for that.
You can see them listed when typing in `help` at the shell prompt.  

Also I documented the dissasembly [procedure](https://github.com/BarryHuffman/Documentation/blob/master/Routers/Belkin/RT3200/Disassembly) for the RT3200, if you'd like to add it to the repo or the table of hardware page.

ErnyTech, 2021-05-15:
> @BarryHuffman I'll answer one by one:
> I don't understand how `auc` managed to destory UBI during sysupgrade (as that is what seems to have happened, as recovery got missing as well). Sounds like reset hits while write hasn't yet completed or something like that... It never happened to me and I've been using `auc` almost daily for more than a month on my device. Maybe just try again and capture serial output of what's happening during the update, so we know more...?
> 
> The easiest way to recover at this point is to connect any of the LAN ports of the router to your PC, configure the PC to have static IP `192.168.1.254` netmask `255.255.255.0`. Then run TFTP server of your choice and let it serve [openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery.itb](https://github.com/dangowrt/linksys-e8450-openwrt-installer/releases/download/v0.4/openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery.itb). Then choose
> 
> ```
> Load recovery system via TFTP then write to flash.
> ```
> 
> in the bootmenu. This will write recovery back to flash and then boot it. From there you can then install the production firmware [openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb](https://github.com/dangowrt/linksys-e8450-openwrt-installer/releases/download/v0.4/openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb) just like after running the installer.
> 
> I was considering to add ready-made scripts to make booting recovery image as well as UEFI system from USB, right you you can only do that using U-Boot shell using `fsload` command.

How do you select in the boot menu? Do you need the serial port?

dangowrt, 2021-05-15:
> How do you select in the boot menu? Do you need the serial port?

Not necessarily. You can also tell the bootloader to jump right into updating preloader and bootloader using `fw_setenv` inside OpenWrt (via SSH):
```
fw_setenv bootcmd "run boot_tftp_write_preloader && run boot_tftp_write_fip ; run reset_factory ; reset"
reboot
```
This makes the board try to acquire new preloader and bootloader via TFTP **once**, then reboot. To know that the operation has succeeded, you can check the build-date in `fw_printenv ver` before and after the update.

ErnyTech, 2021-05-15:
> Not necessarily. You can also tell the bootloader to jump right into updating preloader and bootloader using `fw_setenv` inside OpenWrt (via SSH):
>
> ```
> fw_setenv bootcmd "run [...]

Thank you very much

dangowrt, 2021-05-15:
@ErnyTech Forgot to mention you should set your PC to IP `192.168.1.254/24` and offer the files [openwrt-mediatek-mt7622-linksys_e8450-ubi-preloader.bin](https://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/openwrt-mediatek-mt7622-linksys_e8450-ubi-preloader.bin) as well as [openwrt-mediatek-mt7622-linksys_e8450-ubi-bl31-uboot.fip](https://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/openwrt-mediatek-mt7622-linksys_e8450-ubi-bl31-uboot.fip) on a TFTP.

taney, 2021-05-19:
@BarryHuffman, @dangowrt helped me out with the same issue. I'm running on MacOS and I enabled TFTPD (natively) for the long haul. Any time my router gets into a semi-bricked state, it will automatically flash the recovery. It works EVERY time. This device has proven to be solid and un-brickable; even with SNAPSHOTS!

Thanks again Daniel!

gongtao0607, 2021-06-05:
> The easiest way to recover at this point is to connect any of the LAN ports of the router to your PC, configure the PC to have static IP `192.168.1.254` netmask `255.255.255.0`. Then run TFTP server [...]

Another way I found (probably even easier) is to press the `reset` when powering on, until *both* Power LED and Internet LED are on. This triggers the `boot_tftp_forever` in u-boot (which is essentially `boot_tftp_recovery`). 

After that, sysupgrade to a production image.

BarryHuffman, 2021-06-06:
> Another way I found (probably even easier) is to press the `reset` when powering on, until _both_ Power LED and Internet LED are on.

This is actually listed on the main [page](https://github.com/dangowrt/linksys-e8450-openwrt-installer#post-openwrt-recovery-mode-process). This process should work 99% of the time but in the event where a recovery fails but you still have uboot access, the method provided [here](https://github.com/dangowrt/linksys-e8450-openwrt-installer/issues/6#issuecomment-828870771) should work.

BarryHuffman, 2021-06-06:
@dangowrt 
I think my prior issue with lost recovery was related to "brown outs" in my residential area at the time. (power reduction but doesn't go out)

I have since added a UPS to the router, and had no issues with `auc`.

taney, 2021-06-06:
[image]
I have this set up and it never bricks. If it ever does, it will boot into recovery ;)

gongtao0607, 2021-06-06:
> This is actually listed on the main [page](https://github.com/dangowrt/linksys-e8450-openwrt-installer#post-openwrt-recovery-mode-process). This process should work 99% of the time but in the event [...]

Sorry for any confusion. 

What I was talking about is to enter u-boot tftp-recovery menu which *loads the recovery mtd from TFTP*. - hold `reset` until both Power is orange and Internet is blue/white. You don't need serial access.  (Of course you would need to setup TFTP at 192.168.1.254)

The main page is about *Entering* the recovery mtd - hold `reset` until Power is orange.

dangowrt, 2021-06-06:
> [image]
> I have this set up and it never bricks. If it ever does, it will boot into recovery ;)
... it should actually do that anyway, as recovery is stored on flash as well. Serving it on TFTP as well is not a mistake and in case very unfortunate things happen to the UBI area on the NAND flash, this will still always work. Just right now it's not easy to trigger booting into the on-flash recovery in case the device is already  (you'd have to release the RESET button at the exact right moment). Hence I thought I might change the bootloader to always try loading recovery from TFTP first, in case that fails, try loading it from UBI and in case that fails as well, keep trying via TFTP for ever.

As @BarryHuffman mentioned, these more severe corruptions can potentially be caused by bad power (from the grid or in case power-supply if broken).

Jefe777, 2021-12-24:
Solved:
* [Restoring the vendor/official firmware](https://github.com/dangowrt/linksys-e8450-openwrt-installer#restoring-the-vendorofficial-firmware)

Thank you very much for your work.

closed 2021-12-24 by Jefe777