dangowrt/owrt-ubi-installer #10: auc failure .... cannot request packagelist from rpcd

outbackdingo, 2021-05-31:
Running:   SNAPSHOT r16700-f35dbef6d2 on mediatek/mt7622 (linksys,e8450-ubi)
Available: SNAPSHOT r16845-507257778c
Requesting package lists...
cannot request packagelist from rpcd
No error information (65)

dangowrt, 2021-05-31:
`/etc/init.d/rpcd restart` should help

dangowrt, 2021-05-31:
This should be fixed in https://github.com/openwrt/openwrt/tree/master/package/system/rpcd to reload rpcd using post-install hook of the rpcd-mod-* packages.

closed 2021-05-31 by dangowrt

outbackdingo, 2021-05-31:
ok ... though now it feels stuck.... sitting a long time......
/etc/init.d/rpcd restart
root@OpenWrt:~# auc
auc (0.1.6-4)
Running:   SNAPSHOT r16700-f35dbef6d2 on mediatek/mt7622 (linksys,e8450-ubi)
Available: SNAPSHOT r16845-507257778c
Requesting package lists...
 opkg: 2021-03-15-5936c4f9-1 -> 2021-03-15-5936c4f9-2
 libjson-script: 2021-03-02-2e52c7e9-1 -> 2021-05-16-b14c4688-1
 iw: 5.9-1 -> 5.9-8fab0c9e-2
 rpcd: 2021-03-11-ccb75178-1 -> 2021-05-05-7a560a1a-1
 busybox: 1.33.0-3 -> 1.33.1-9
 kmod-nf-reject6: 5.10.35-1 -> 5.10.41-1
 kmod-nf-flow: 5.10.35-1 -> 5.10.41-1
 kmod-lib-crc-ccitt: 5.10.35-1 -> 5.10.41-1
 getrandom: 2020-10-25-9ef88681-1 -> 2020-10-25-9ef88681-2
 kmod-pppoe: 5.10.35-1 -> 5.10.41-1
 kmod-pppox: 5.10.35-1 -> 5.10.41-1
 kmod-ipt-conntrack: 5.10.35-1 -> 5.10.41-1
 kmod-nf-reject: 5.10.35-1 -> 5.10.41-1
 base-files: 1415-r16700-f35dbef6d2 -> 1428-r16845-507257778c
 kmod-nf-nat: 5.10.35-1 -> 5.10.41-1
 kmod-mt7615e: 5.10.35+2021-04-11-bf45b30d-4 -> 5.10.41+2021-05-15-9d736545-4
 netifd: 2021-04-03-327da989-3 -> 2021-05-26-899c2a45-1
 uboot-envtools: 2021.01-24 -> 2021.01-27
 procd: 2021-05-05-021ece84-1 -> 2021-05-05-021ece84-2
 kmod-mt7615-common: 5.10.35+2021-04-11-bf45b30d-4 -> 5.10.41+2021-05-15-9d736545-4
 luci-mod-status: git-21.127.62194-d24e329 -> git-21.132.35841-3c66c5b
 libblobmsg-json: 2021-03-02-2e52c7e9-1 -> 2021-05-16-b14c4688-1
 kmod-mt76-connac: 5.10.35+2021-04-11-bf45b30d-4 -> 5.10.41+2021-05-15-9d736545-4
 kmod-nf-ipt: 5.10.35-1 -> 5.10.41-1
 wpad-openssl: 2020-06-08-5a8b3662-32 -> 2021-05-22-b102f19b-1
 kmod-ip6tables: 5.10.35-1 -> 5.10.41-1
 fstools: 2021-03-19-964d1e3a-1 -> 2021-05-16-3d40a1b5-2
 uclient-fetch: 2021-04-03-64e00d6d-1 -> 2021-05-14-6a6011df-1
 rpcd-mod-file: 2021-03-11-ccb75178-1 -> 2021-05-05-7a560a1a-1
 libubox: 2021-03-02-2e52c7e9-1 -> 2021-05-16-b14c4688-1
 luci-base: git-21.124.24916-0faf9a4 -> git-21.148.48881-79947af
 kmod-leds-gpio: 5.10.35-1 -> 5.10.41-1
 kmod-gpio-button-hotplug: 5.10.35-3 -> 5.10.41-3
 logd: 2020-10-25-9ef88681-1 -> 2020-10-25-9ef88681-2
 kmod-mac80211: 5.10.35+5.10.34-1-1 -> 5.10.41+5.10.34-1-1
 luci-proto-ipv6: git-21.074.38504-171ef77 -> git-21.148.48881-79947af
 openwrt-keyring: 2021-02-20-49283916-1 -> 2021-02-20-49283916-2
 jshn: 2021-03-02-2e52c7e9-1 -> 2021-05-16-b14c4688-1
 kmod-ipt-core: 5.10.35-1 -> 5.10.41-1
 kmod-ppp: 5.10.35-1 -> 5.10.41-1
 kmod-mt7615-firmware: 5.10.35+2021-04-11-bf45b30d-4 -> 5.10.41+2021-05-15-9d736545-4
 kmod-nf-conntrack: 5.10.35-1 -> 5.10.41-1
 kmod-nf-ipt6: 5.10.35-1 -> 5.10.41-1
 libuclient: 2021-04-03-64e00d6d-1 -> 2021-05-14-6a6011df-1
 kmod-nf-conntrack6: 5.10.35-1 -> 5.10.41-1
 kmod-mt76-core: 5.10.35+2021-04-11-bf45b30d-4 -> 5.10.41+2021-05-15-9d736545-4
 ubox: 2020-10-25-9ef88681-1 -> 2020-10-25-9ef88681-2
 rpcd-mod-iwinfo: 2021-03-11-ccb75178-1 -> 2021-05-05-7a560a1a-1
 luci-mod-network: git-21.110.65202-b7dd10d -> git-21.149.58754-e7c9c63
 hostapd-common: 2020-06-08-5a8b3662-32 -> 2021-05-22-b102f19b-1
 kmod-ipt-offload: 5.10.35-1 -> 5.10.41-1
 wireless-regdb: 2020.11.20-1 -> 2021.04.21-1
 kmod-slhc: 5.10.35-1 -> 5.10.41-1
 kmod-cfg80211: 5.10.35+5.10.34-1-1 -> 5.10.41+5.10.34-1-1
 kmod-ipt-nat: 5.10.35-1 -> 5.10.41-1
Are you sure you want to continue the upgrade process? [N/y] y
Requesting build.......
Writing to 'openwrt-885fbb90c24b-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb'

dangowrt, 2021-05-31:
Depending on your download speed from the chef.libremesh.org server this should not take more than a few minutes...

outbackdingo, 2021-05-31:
ok it finished, whew.... close!

dangowrt, 2021-05-31:
Looks like you are missing `kmod-mt7915e` though, probably you run `auc` in the recovery/initramfs (which doesn't come with all drivers due to its limited purpose) and hence `auc` would also only select the packages currently installed there... Ie. you will need to install `kmod-mt7915e` in order to have working 5 GHz WiFi.

outbackdingo, 2021-05-31:
nope, something is not right, im still seeing 16700 after runing auc on reboot, even though it says its Writing to 'openwrt-885fbb90c24b-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb'

auc
auc (0.1.6-4)
Running:   SNAPSHOT r16700-f35dbef6d2 on mediatek/mt7622 (linksys,e8450-ubi)
Available: SNAPSHOT r16845-507257778c
Requesting package lists...
cannot request packagelist from rpcd
No error information (65)
root@OpenWrt:~# /etc/init.d/rpcd restart
root@OpenWrt:~# auc
auc (0.1.6-4)
Running:   SNAPSHOT r16700-f35dbef6d2 on mediatek/mt7622 (linksys,e8450-ubi)
Available: SNAPSHOT r16845-507257778c
Requesting package lists...
 opkg: 2021-03-15-5936c4f9-1 -> 2021-03-15-5936c4f9-2
 libjson-script: 2021-03-02-2e52c7e9-1 -> 2021-05-16-b14c4688-1
 iw: 5.9-1 -> 5.9-8fab0c9e-2
 rpcd: 2021-03-11-ccb75178-1 -> 2021-05-05-7a560a1a-1
 busybox: 1.33.0-3 -> 1.33.1-9
 kmod-nf-reject6: 5.10.35-1 -> 5.10.41-1
 kmod-nf-flow: 5.10.35-1 -> 5.10.41-1
 kmod-lib-crc-ccitt: 5.10.35-1 -> 5.10.41-1
 getrandom: 2020-10-25-9ef88681-1 -> 2020-10-25-9ef88681-2
 kmod-pppoe: 5.10.35-1 -> 5.10.41-1
 kmod-pppox: 5.10.35-1 -> 5.10.41-1
 kmod-ipt-conntrack: 5.10.35-1 -> 5.10.41-1
 kmod-nf-reject: 5.10.35-1 -> 5.10.41-1
 base-files: 1415-r16700-f35dbef6d2 -> 1428-r16845-507257778c
 kmod-nf-nat: 5.10.35-1 -> 5.10.41-1
 kmod-mt7615e: 5.10.35+2021-04-11-bf45b30d-4 -> 5.10.41+2021-05-15-9d736545-4
 netifd: 2021-04-03-327da989-3 -> 2021-05-26-899c2a45-1
 uboot-envtools: 2021.01-24 -> 2021.01-27
 procd: 2021-05-05-021ece84-1 -> 2021-05-05-021ece84-2
 kmod-mt7615-common: 5.10.35+2021-04-11-bf45b30d-4 -> 5.10.41+2021-05-15-9d736545-4
 luci-mod-status: git-21.127.62194-d24e329 -> git-21.132.35841-3c66c5b
 libblobmsg-json: 2021-03-02-2e52c7e9-1 -> 2021-05-16-b14c4688-1
 kmod-mt76-connac: 5.10.35+2021-04-11-bf45b30d-4 -> 5.10.41+2021-05-15-9d736545-4
 kmod-nf-ipt: 5.10.35-1 -> 5.10.41-1
 wpad-openssl: 2020-06-08-5a8b3662-32 -> 2021-05-22-b102f19b-1
 kmod-ip6tables: 5.10.35-1 -> 5.10.41-1
 fstools: 2021-03-19-964d1e3a-1 -> 2021-05-16-3d40a1b5-2
 uclient-fetch: 2021-04-03-64e00d6d-1 -> 2021-05-14-6a6011df-1
 rpcd-mod-file: 2021-03-11-ccb75178-1 -> 2021-05-05-7a560a1a-1
 libubox: 2021-03-02-2e52c7e9-1 -> 2021-05-16-b14c4688-1
 luci-base: git-21.124.24916-0faf9a4 -> git-21.148.48881-79947af
 kmod-leds-gpio: 5.10.35-1 -> 5.10.41-1
 kmod-gpio-button-hotplug: 5.10.35-3 -> 5.10.41-3
 logd: 2020-10-25-9ef88681-1 -> 2020-10-25-9ef88681-2
 kmod-mac80211: 5.10.35+5.10.34-1-1 -> 5.10.41+5.10.34-1-1
 luci-proto-ipv6: git-21.074.38504-171ef77 -> git-21.148.48881-79947af
 openwrt-keyring: 2021-02-20-49283916-1 -> 2021-02-20-49283916-2
 jshn: 2021-03-02-2e52c7e9-1 -> 2021-05-16-b14c4688-1
 kmod-ipt-core: 5.10.35-1 -> 5.10.41-1
 kmod-ppp: 5.10.35-1 -> 5.10.41-1
 kmod-mt7615-firmware: 5.10.35+2021-04-11-bf45b30d-4 -> 5.10.41+2021-05-15-9d736545-4
 kmod-nf-conntrack: 5.10.35-1 -> 5.10.41-1
 kmod-nf-ipt6: 5.10.35-1 -> 5.10.41-1
 libuclient: 2021-04-03-64e00d6d-1 -> 2021-05-14-6a6011df-1
 kmod-nf-conntrack6: 5.10.35-1 -> 5.10.41-1
 kmod-mt76-core: 5.10.35+2021-04-11-bf45b30d-4 -> 5.10.41+2021-05-15-9d736545-4
 ubox: 2020-10-25-9ef88681-1 -> 2020-10-25-9ef88681-2
 rpcd-mod-iwinfo: 2021-03-11-ccb75178-1 -> 2021-05-05-7a560a1a-1
 luci-mod-network: git-21.110.65202-b7dd10d -> git-21.149.58754-e7c9c63
 hostapd-common: 2020-06-08-5a8b3662-32 -> 2021-05-22-b102f19b-1
 kmod-ipt-offload: 5.10.35-1 -> 5.10.41-1
 wireless-regdb: 2020.11.20-1 -> 2021.04.21-1
 kmod-slhc: 5.10.35-1 -> 5.10.41-1
 kmod-cfg80211: 5.10.35+5.10.34-1-1 -> 5.10.41+5.10.34-1-1
 kmod-ipt-nat: 5.10.35-1 -> 5.10.41-1
Are you sure you want to continue the upgrade process? [N/y] y
Writing to 'openwrt-885fbb90c24b-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb'

outbackdingo, 2021-05-31:
it is also saying the root key for ssh has changed on every reboot, like its regenrating and not saving data
ssh root@192.168.1.1
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
@    WARNING: REMOTE HOST IDENTIFICATION HAS CHANGED!     @
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
IT IS POSSIBLE THAT SOMEONE IS DOING SOMETHING NASTY!
Someone could be eavesdropping on you right now (man-in-the-middle attack)!
It is also possible that a host key has just been changed.
The fingerprint for the ED25519 key sent by the remote host is
SHA256:6HCRluLAnzNxk5W3Yk/yhCrCPYpJU41yQDbyXgM6+Ps.
Please contact your system administrator.
Add correct host key in /home/dingo/.ssh/known_hosts to get rid of this message.
Offending ED25519 key in /home/dingo/.ssh/known_hosts:85
Host key for 192.168.1.1 has changed and you have requested strict checking.
Host key verification failed.

outbackdingo, 2021-05-31:
❯ vi .ssh/known_hosts
The authenticity of host '192.168.1.1 (192.168.1.1)' can't be established.
ED25519 key fingerprint is SHA256:DREfKmuyEVsa6VsQ9l61OvamR74THRr6TdxEkv0WIME.
This key is not known by any other names
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added '192.168.1.1' (ED25519) to the list of known hosts.

BusyBox v1.33.0 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r16700-f35dbef6d2
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:~# reboot
root@OpenWrt:~# Connection to 192.168.1.1 closed by remote host.
❯ ssh root@192.168.1.1
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
@    WARNING: REMOTE HOST IDENTIFICATION HAS CHANGED!     @
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
IT IS POSSIBLE THAT SOMEONE IS DOING SOMETHING NASTY!
Someone could be eavesdropping on you right now (man-in-the-middle attack)!
It is also possible that a host key has just been changed.
The fingerprint for the ED25519 key sent by the remote host is
SHA256:iM0ryql3X6SMbQ2baIsiwwmncZChcvhuZgrOyOLgY7s.
Please contact your system administrator.
Add correct host key in /home/dingo/.ssh/known_hosts to get rid of this message.
Offending ED25519 key in /home/dingo/.ssh/known_hosts:85
Host key for 192.168.1.1 has changed and you have requested strict checking.
Host key verification failed.

dangowrt, 2021-05-31:
`auc` is not meant to be installer and used inside initramfs/recovery, and that's probably where you are now after force-flashing the non-UBI firmware. From there, first of all, you have to flash the `*ubi-sysupgrade.itb` like shown in the installer video/manual. After verifying you are no longer in initramfs (ie. check if `/rom/` is mounted or call `ubus call system board` to find out), you can install `auc` or `luci-app-attendedsysupgrade` and that should work then.

outbackdingo, 2021-05-31:
the first two things i did was flash the installer, then flash the sysupgrade, now it seems it is literally stuck at booting this image not matter what i try to flash, right now mount shows ... so how can i get past this, tftp flash or i broke it ? :)
mount
tmpfs on / type tmpfs (rw,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
tmpfs on /dev type tmpfs (rw,nosuid,relatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,mode=600,ptmxmode=000)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
none on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,noatime,mode=700)
pstore on /sys/fs/pstore type pstore (rw,noatime)

ubus call system board
{
	"initramfs": true,
	"kernel": "5.10.35",
	"hostname": "OpenWrt",
	"system": "ARMv8 Processor rev 4",
	"model": "Linksys E8450 (UBI)",
	"board_name": "linksys,e8450-ubi",
	"release": {
		"distribution": "OpenWrt",
		"version": "SNAPSHOT",
		"revision": "r16700-f35dbef6d2",
		"target": "mediatek/mt7622",
		"description": "OpenWrt SNAPSHOT r16700-f35dbef6d2"
	}
}

dangowrt, 2021-05-31:
Please see https://github.com/dangowrt/linksys-e8450-openwrt-installer/issues/11#issuecomment-851581669

outbackdingo, 2021-05-31:
LOL... Save... that worked! whew, on my image now, appreciate the help!

dangowrt, 2021-05-31:
Thank you for involuntarily testing robustness (you really went off the rails, even force-flashed something wrong and yet made it out of there without even needing serial access). And you even covered latest features (PSTORE/ramopps) :smiley_cat:

dangowrt, 2021-05-31:
I must admit that the dual-boot recovery (=initramfs) vs. production scheme and boot selection by PSTORE/ramoops content deserves some more documentation...