dangowrt/owrt-ubi-installer #11: OpenWRT trunk image - fails to flash

outbackdingo, 2021-05-31:
okay, enlighten me, i built an image from openwrt master for the E8450

ls -al bin/targets/mediatek/mt7622
total 21916
drwxr-xr-x. 1 dingo dingo      462 May 28 13:33 .
drwxr-xr-x. 1 dingo dingo       12 May 28 13:13 ..
-rw-r--r--. 1 dingo dingo     3275 May 31 20:51 config.buildinfo
-rw-r--r--. 1 dingo dingo      412 May 31 20:51 feeds.buildinfo
-rw-r--r--. 1 dingo dingo 10879140 May 31 20:59 openwrt-mediatek-mt7622-linksys_e8450-initramfs-kernel.bin
-rw-r--r--. 1 dingo dingo     5567 May 31 20:59 openwrt-mediatek-mt7622-linksys_e8450.manifest
-rw-r--r--. 1 dingo dingo 11535138 May 31 20:59 openwrt-mediatek-mt7622-linksys_e8450-squashfs-sysupgrade.bin
drwxr-xr-x. 1 dingo dingo     7230 May 31 20:58 packages
-rw-r--r--. 1 dingo dingo      615 May 31 20:59 sha256sums
-rw-r--r--. 1 dingo dingo       18 May 31 20:51 version.buildinfo

If i upload it to the gui and flash it tells me

[image]

if i force flash, it doesnt appear to flash my image, as ive added custom packages and config, and they dont exist on login ? feels like the previous firmware. So can we build a custom image with our own packages and sysupgrade? or thats a fail ?

dangowrt, 2021-05-31:
You have to select the UBI variant in the build system to generate a matching image.
[image]

outbackdingo, 2021-05-31:
Yupp that fails...

cp /home/dingo/Devel/GIT/Linksys8450/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/tmp/openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb /home/dingo/Devel/GIT/Linksys8450/bin/targets/mediatek/mt7622/openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb
cat /home/dingo/Devel/GIT/Linksys8450/staging_dir/target-aarch64_cortex-a53_musl/image/mt7622-snand-1ddr-bl2.img >> /home/dingo/Devel/GIT/Linksys8450/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/tmp/openwrt-mediatek-mt7622-linksys_e8450-ubi-preloader.bin
cat: /home/dingo/Devel/GIT/Linksys8450/staging_dir/target-aarch64_cortex-a53_musl/image/mt7622-snand-1ddr-bl2.img: No such file or directory
make[5]: *** [Makefile:41: /home/dingo/Devel/GIT/Linksys8450/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/tmp/openwrt-mediatek-mt7622-linksys_e8450-ubi-preloader.bin] Error 1
make[5]: Leaving directory '/home/dingo/Devel/GIT/Linksys8450/target/linux/mediatek/image'
make[4]: *** [Makefile:18: install] Error 2
make[4]: Leaving directory '/home/dingo/Devel/GIT/Linksys8450/target/linux/mediatek'
make[3]: *** [Makefile:11: install] Error 2

dangowrt, 2021-05-31:
When switching from one build profile to another, you have to make sure all default-packages of the new profile get selected, ie. make sure that `arm-trusted-firmware-mediatek-snand-1ddr` as well as `uboot-mt7622_linksys_e8450` are selected

[image]

closed 2021-05-31 by dangowrt

outbackdingo, 2021-05-31:
ok ive built an image for it, yet it simply refuses to flash, it always returns to 
OpenWrt SNAPSHOT r16700-f35dbef6d2 / LuCI Master git-21.124.24916-0faf9a4

-rw-r--r--. 1 dingo dingo     3279 May 31 22:14 config.buildinfo
-rw-r--r--. 1 dingo dingo      412 May 31 22:14 feeds.buildinfo
-rw-r--r--. 1 dingo dingo   933382 May 31 22:26 openwrt-mediatek-mt7622-linksys_e8450-ubi-bl31-uboot.fip
-rw-r--r--. 1 dingo dingo 11010048 May 31 22:26 openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery.itb
-rw-r--r--. 1 dingo dingo     5567 May 31 22:26 openwrt-mediatek-mt7622-linksys_e8450-ubi.manifest
-rw-r--r--. 1 dingo dingo    67445 May 31 22:26 openwrt-mediatek-mt7622-linksys_e8450-ubi-preloader.bin
-rw-r--r--. 1 dingo dingo 12808998 May 31 22:26 openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb
drwxr-xr-x. 1 dingo dingo     7230 May 31 22:26 packages
-rw-r--r--. 1 dingo dingo      874 May 31 22:26 sha256sums
-rw-r--r--. 1 dingo dingo       18 May 31 22:14 version.buildinfo

dangowrt, 2021-05-31:
If the device keeps booting into recovery/initramfs this is most likely because there is something in PSTORE/ramoops you should check (and then delete). In your case this is obviously the boot log ending with `cannot mount rootfs` created by your attempt to force-flash the non-UBI firmware.
What you should do is:
```
ls /sys/fs/pstore
# check what is there, eg. cat /sys/fs/pstore/*
rm /sys/fs/pstore/*
reboot
```