dangowrt/owrt-ubi-installer #14: How to get back to the official firmware from OPENWRT?

Athit19, 2021-06-15:
I have searched a lot of tutorials on the Internet, but there is no way to flash back to the official firmware, because the speed and stability of OPENWRT do not feel as stable and fast as the official, so I want to flash back to the official firmware version, is there any way? ?

alleyu2, 2021-06-15:
I am not sure if you have read the readme.

`WARNING This will replace the bootloader (TF-A 2.2, U-Boot 2020.10) and convert the flash layout of the device to UBI! The installer stores a copy of the previous bootchain in a dedicated UBI volume boot_backup. **If you want to go back to the vendor firmware, you will have to boot into recovery mode (ie. initramfs), copy those files into /tmp, umount the UBI volume, detach the UBI device and then write the files to the corresponding MTD partitions (mtd write $file /dev/mtdX).**`

I have personally not installed the UBI version because the back-to-stock process is complicated. 

If you would have installed the non-UBI version it would have been an easy fix. i.e. flash vendor firmware from Luci. I am not a developer but I have read this information in OpenWrt Forums.

Athit19, 2021-06-15:
I have already read the readme, because I am not a developer, but a hobbyist. I am currently flashing OPENWRT, and I really feel bad. I just want to flash back to the official firmware version. A simple and feasible method is needed.

alleyu2, 2021-06-16:
You can try taking baby steps and let us know where you are getting stuck.

1.  Do you have access to ssh? I am assuming you should.
2. If yes, can you ssh and check if you have the boot_backup folder in the UBI volume?
3. If yes, follow this `you will have to boot into recovery mode (ie. initramfs), copy those files into /tmp, umount the UBI volume, detach the UBI device and then write the files to the corresponding MTD partitions (mtd write $file /dev/mtdX)`

Let me know where do you get stuck.

Athit19, 2021-06-16:
I have to say that I am really a beginner, and as far as I know the knowledge, I think I can use SSH to enter.

But for the UBI volume you mentioned, I don’t know how to find it, and the initramfs mode you mentioned. How should I enter it?

It would be great if you can teach me patiently and provide help with sentences.

fda77, 2021-07-20:
I want also to go back to non-ubi. It worked very well, and wiht uci after saving any change (eg only set password) i cant ping device later and have to reset
I've a backup of the 15 mtd0-mtd14 files, 150MB. Have i all to write back?

> umount the UBI volume

How? Wiht recovery there is no ubi:
```
tmpfs on / type tmpfs (rw,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
tmpfs on /dev type tmpfs (rw,nosuid,relatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,mode=600,ptmxmode=000)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
none on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,noatime,mode=700)
pstore on /sys/fs/pstore type pstore (rw,noatime)
```
 - detach the UBI device

sounds like `ubidetach`should be used with some arguments

 -  `mtd write $file /dev/mtdX`

I dont have nough devices
```
ls -al /dev/mtd*
crw-------    1 root     root       90,   0 Jan  1  1970 /dev/mtd0
crw-------    1 root     root       90,   1 Jan  1  1970 /dev/mtd0ro
crw-------    1 root     root       90,   2 Jan  1  1970 /dev/mtd1
crw-------    1 root     root       90,   3 Jan  1  1970 /dev/mtd1ro
crw-------    1 root     root       90,   4 Jan  1  1970 /dev/mtd2
crw-------    1 root     root       90,   5 Jan  1  1970 /dev/mtd2ro
crw-------    1 root     root       90,   6 Jan  1  1970 /dev/mtd3
crw-------    1 root     root       90,   7 Jan  1  1970 /dev/mtd3ro
brw-------    1 root     root       31,   0 Jan  1  1970 /dev/mtdblock0
brw-------    1 root     root       31,   2 Jan  1  1970 /dev/mtdblock1
brw-------    1 root     root       31,   4 Jan  1  1970 /dev/mtdblock2
brw-------    1 root     root       31,   6 Jan  1  1970 /dev/mtdblock3
```

dangowrt, 2021-07-20:
As the total size of the flash chip is only 128MiB, 150MiB certainly has some overlap.
You have to use the initramfs image of the **non-UBI firmware** to have the same MTD partitioning which matches your backup files. Then you can write them back one by one. I only see mtd0...mtd13, so no idea what mtd14 is all about. However, when I try reverting to  factory (in order to test updated versions of the installer) I only write back 
```
Preloader (mtd0)
ATF (mtd1)
Bootloader (mtd2)
Config (mtd3)
```
of the stock firmware and then use the stock bootloader to load `Kernel1 (mtd8)` via TFTP using the stock loader.

Would also be interesting to know what actually went wrong, but that's up to you to tell if you are even into investigating that.

fda77, 2021-07-20:
Maybe i can not save because of differen layout?
```
 du -sh *
516K    mtd0.bin
260K    mtd1.bin
516K    mtd2.bin
516K    mtd3.bin
1,1M    mtd4.bin
132K    mtd5.bin
132K    mtd6.bin
132K    mtd7.bin
31M     mtd8.bin
3,7M    mtd9.bin
27M     mtd10.bin
13M     mtd11.bin
31M     mtd12.bin
26M     mtd13.bin
21M     mtd14.bin
```
Im also interest to find the cause, but someone has to tell me how :)  

Currently this is mounted:
```
df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                19.8M     19.8M         0 100% /rom
tmpfs                   241.4M      4.1M    237.3M   2% /tmp
/dev/ubi0_5              67.3M    340.0K     63.5M   1% /overlay
overlayfs:/overlay       67.3M    340.0K     63.5M   1% /
tmpfs                   512.0K         0    512.0K   0% /dev
/dev/sda1                 7.1G     55.1M      6.7G   1% /mnt/sda1
```

> initramfs image of the non-UBI firmware
Thanks. I'll try it later

dangowrt, 2021-07-20:
The mounted filesystems look like they should. Can you share output `logread` right after boot completed (make sure to censor MAC addresses and other unique bits, in case you care)?

fda77, 2021-07-20:
The mtd files where downloaded by luci, and I took all available

I flashed a new image and unselect the "keep settings". Hopeffully related is included:

```
Tue Jul 20 16:01:04 2021 kern.info kernel: [    0.521039] spi-nand spi2.0: Fidelix SPI NAND was found.
Tue Jul 20 16:01:04 2021 kern.info kernel: [    0.526403] spi-nand spi2.0: 128 MiB, block size: 128 KiB, page size: 2048, OOB size: 64
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    0.534771] 4 fixed-partitions partitions found on MTD device spi2.0
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    0.541142] Creating 4 MTD partitions on "spi2.0":
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    0.545939] 0x000000000000-0x000000080000 : "bl2"
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    0.551652] 0x000000080000-0x0000001c0000 : "fip"
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    0.558258] 0x0000001c0000-0x0000002c0000 : "factory"
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    0.564903] 0x000000300000-0x000008000000 : "ubi"
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    0.603204] random: fast init done
Tue Jul 20 16:01:04 2021 kern.info kernel: [    0.725007] libphy: Fixed MDIO Bus: probed
Tue Jul 20 16:01:04 2021 kern.info kernel: [    0.753083] libphy: mdio: probed
```

```
Tue Jul 20 16:01:04 2021 kern.info kernel: [    1.501479] hub 2-0:1.0: 1 port detected
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    1.506495] UBI: auto-attach mtd3
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    1.509824] ubi0: attaching mtd3
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    1.854814] ubi0: scanning is finished
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    1.865108] ubi0: attached mtd3 (name "ubi", size 125 MiB)
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    1.870649] ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    1.877536] ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    1.884321] ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    1.891290] ubi0: good PEBs: 999, bad PEBs: 1, corrupted PEBs: 0
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    1.897305] ubi0: user volume: 6, internal volumes: 1, max. volumes count: 128
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    1.904525] ubi0: max/mean erase counter: 7/4, WL threshold: 4096, image sequence number: 1438217621
Tue Jul 20 16:01:04 2021 kern.info kernel: [    1.905897] usb 1-1: new high-speed USB device number 2 using xhci-mtk
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    1.913662] ubi0: available PEBs: 0, total reserved PEBs: 999, PEBs reserved for bad PEB handling: 19
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    1.929826] ubi0: background thread "ubi_bgt0d" started, PID 745
Tue Jul 20 16:01:04 2021 kern.debug kernel: [    1.937078] FIT: Default configuration: "config-1" (OpenWrt linksys_e8450-ubi)
Tue Jul 20 16:01:04 2021 kern.debug kernel: [    1.944316] FIT:           kernel sub-image 0x00001000..0x004d48c1 "kernel-1" (ARM64 OpenWrt Linux-5.10.51)
Tue Jul 20 16:01:04 2021 kern.debug kernel: [    1.954160] FIT:          flat_dt sub-image 0x004d5000..0x004dc3b9 "fdt-1" (ARM64 OpenWrt linksys_e8450-ubi device tree blob)
Tue Jul 20 16:01:04 2021 kern.debug kernel: [    1.965558] FIT:       filesystem sub-image 0x004dd000..0x0186efff "rootfs-1" (ARM64 OpenWrt linksys_e8450-ubi rootfs)
Tue Jul 20 16:01:04 2021 kern.debug kernel: [    1.976341] FIT: selecting configured loadable "rootfs-1" to be root filesystem
Tue Jul 20 16:01:04 2021 kern.info kernel: [    1.983642]  ubiblock0_4: p1(rootfs-1)
Tue Jul 20 16:01:04 2021 kern.info kernel: [    1.983948] block ubiblock0_4: created from ubi0:4(fit)
Tue Jul 20 16:01:04 2021 kern.err kernel: [    1.995478] squashfs: SQUASHFS error: Xattrs in filesystem, these will be ignored
Tue Jul 20 16:01:04 2021 kern.err kernel: [    2.003008] unable to read xattr id index table
Tue Jul 20 16:01:04 2021 kern.info kernel: [    2.009132] VFS: Mounted root (squashfs filesystem) readonly on device 259:1.
```
Are "Xattrs" needed?

```
Tue Jul 20 16:01:04 2021 user.info kernel: [    7.346779] mount_root: loading kmods from internal overlay
Tue Jul 20 16:01:04 2021 user.info kernel: [    7.360223] kmodloader: loading kernel modules from //etc/modules-boot.d/*
Tue Jul 20 16:01:04 2021 user.info kernel: [    7.368235] kmodloader: done loading kernel modules from //etc/modules-boot.d/*
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    7.846721] UBIFS (ubi0:5): default file-system created
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    7.852283] UBIFS (ubi0:5): Mounting in unauthenticated mode
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    7.858368] UBIFS (ubi0:5): background thread "ubifs_bgt0_5" started, PID 853
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    7.891393] UBIFS (ubi0:5): UBIFS: mounted UBI device 0, volume 5, name "rootfs_data"
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    7.899289] UBIFS (ubi0:5): LEB size: 126976 bytes (124 KiB), min./max. I/O unit sizes: 2048 bytes/2048 bytes
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    7.909227] UBIFS (ubi0:5): FS size: 77074432 bytes (73 MiB, 607 LEBs), journal size 3809280 bytes (3 MiB, 30 LEBs)
Tue Jul 20 16:01:04 2021 kern.notice kernel: [    7.919671] UBIFS (ubi0:5): reserved for root: 3640415 bytes (3555 KiB)
[... cut ...]
 ls -al  `find / -name fstab`
-rw-r--r--    1 root     root           266 Jul 20 16:01 /etc/config/fstab
-rw-r--r--    1 root     root            61 Jul 20 16:01 /etc/fstab
-rwxr-xr-x    1 root     root           169 Jul 20 16:01 /etc/init.d/fstab
-rw-r--r--    1 root     root           266 Jul 20 16:01 /overlay/upper/etc/config/fstab
-rw-r--r--    1 root     root            61 Jul 20 16:01 /rom/etc/fstab
-rwxr-xr-x    1 root     root           169 Jul 20 16:01 /rom/etc/init.d/fstab
```

Btw, are you on irc?

dangowrt, 2021-07-20:
/etc/config/fstab is auto-generated on first-boot, so not surprising you find you (connected) pendrive there, that's the intended outcome.

The block errors you see are not really errors (we should silcence them hence), it just means that you haven't configured rootfs to be on an external device (USB pendrive), ie. what we call `extroot` in the OpenWrt world.

I don't see that xattr warning on my system, so that is a bit odd and may be related to the packages you built into the rootfs (or you enabled stuff like SELinux, I guess you'd know...)

I'm dangole on OFTC network.

tkso1997, 2021-09-26:
Could somebody share the the backup Partitions/ mtd files? Stupid as I am I haven’t made any backups …

sk23, 2021-11-20:
@Athit19 @tkso1997 

## [AT YOUR OWN RISK] If you want to flash back to the vendor firmware, the following steps are for reference only, I have not tested it.

#### 1. Flash the initramfs recovery firmware and restart.
boot into openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery.itb

#### 2. Write back the vendor mtd
* Mount ubifs `mount -t ubifs /dev/ubi0_3 /mnt`
* Copy the original mtd backup, `cp /mnt/mtd* /tmp`
* umount the UBI volume `ubidetach -d 0`
* Flash into the original factory mtd(0/1/2/3), `mtd write /tmp/mtdX /dev/mtdX`

#### 3. After restarting, use TFTP to flash the vendor firmware 
Follow https://www.linksys.com/us/support-article?articleNum=137928
Router default IP: **192.168.1.1**, default password is **admin**.
The original factory firmware will flash into Kernel1 (mtd8). 

#### 4. Done.

dangowrt, 2021-11-20:
Important to mention that `umount /mnt` is needed after `cp /mnt/mtd* /tmp` before `ubidetach -d 0`.

Also important to note that this will only work with the files in `boot_backup` if the installer has run for *the first time*. Users have to copy the files off the device after first installation is done. Once the installer is run again, `boot_backup` will be overwritten with whatever is in the flash at that point.
As many people did re-run the installer in order to fix OOB/ECC data of `factory` partition (or even just to update U-Boot) by running v0.5.3 or later installer on a system which had already been converted to the new bootchain, what is in their `boot_backup` filesystem now is just the previous version of the new (as in: already UBIfied) bootchain.

To always be fine I would hence suggest to upload backup files you should have copied off the router right after installation instead of using what you find in `boot_backup` at a later point.

dangowrt, 2021-11-20:
@JRock37 looking at your backup files, unfortunately they are not a backup of the original bootchain but contain files extracted from a system already running UBI bootchain. Hence you can **not** use them to return to stock.

JRock37, 2021-11-20:
Well, damn was hoping to just go back to stock since I been having so many
problems.

KA2107, 2021-11-21:
@JRock37 and @dangowrt: I had backed up these files after flashing 0.6.0 directly from OEM firmware, backing up mtdblock* from LuCI interface, and then flashing OpenWRT SNAPSHOT sysupgrade, then mounting boot_backup andcopying them.

0.6.0 boot_backup MTD: https://www.dropbox.com/sh/d95758qgt9eta54/AADFrShP3b_yujWwXCbE8SrWa
0.6.0 boot_backup MTDBLOCK using LuCI: https://www.dropbox.com/sh/9dofj1d60ysv9jr/AACz5B-OQKhCFHIYuKwDJ-LNa

@dangowrt  Please let me know if these files will work for reverting back to OEM firmware. **I think** only mtd3/mtdblock3 (factory.bin) differs from device to device as that contains the specific MAC address and WiFi Calibration data.

@JRock37 **I think** even if you re-ran dangowrt's installer mtd3/mtdblcok3 (factory.bin) in boot_backup should be same copy as before.

dangowrt, 2021-11-21:
The first set of files
> 0.6.0 boot_backup MTD: https://www.dropbox.com/sh/d95758qgt9eta54/AADFrShP3b_yujWwXCbE8SrWa
will work to revert to factory bootchain.

>  Please let me know if these files will work for reverting back to OEM firmware. I think only mtd3/mtdblock3 (factory.bin) differs from device to device as that contains the specific MAC address and WiFi Calibration data. Please correct me if I am wrong.

Also `mtd4/mtdblock4 (ubi)` is device specific as it covers the `devinfo` partition where the stock firmware stores device-specific information incl. serial number, addresses, SSID, wifi password, ...

KA2107, 2021-11-21:
> Also mtd4/mtdblock4 (ubi) is device specific as it covers the devinfo partition where the stock firmware stores device-specific information incl. serial number, addresses, SSID, wifi password, ...

There is no mtd4/mtdblock4 in /dev or in boot_backup. Do you mean mtd3/mtdblock3 ?

dangowrt, 2021-11-21:
> Do you mean mtd3/mtdblock3 ?

Yes :roll_eyes:
What I meant was that **both** mtd2/mtdblock2 (`factory`) and mtd3/mtdblock3 (`ubi`) cover device-specific (as in serial number, MAC addresses, ...) information.

sk23, 2021-11-21:
@dangowrt  I have a question. 
How to write back the backup mtd ? It seems that the sizes does not match. 

|  | Vendor Flash Layout | Size | boot_backup mtd Size |
|---|---|---|---|
| mtd0 | Preloader | 512 KiB | 512K  |
| mtd1 | ATF | 256 KiB | 1.3M |
| mtd2 | Bootloader | 512 KiB | 1.0M |
| mtd3 | Config | 512 KiB | 2.0M |
| mtd4 | Factory | 1 MiB | |
| mtd5 | devinfo | 128 KiB | |

dangowrt, 2021-11-21:
That backup files correspond to the partitions with UBI layout, so you have to use the UBI recovery image to flash them.
See instructions here https://github.com/dangowrt/linksys-e8450-openwrt-installer/issues/14#issuecomment-974640694

sk23, 2021-11-21:
> That backup files correspond to the partitions with UBI layout, so you have to use the UBI recovery image to flash them. See instructions here [#14 (comment)](https://github.com/dangowrt/linksys-e8450-openwrt-installer/issues/14#issuecomment-974640694)

So, after the backup mtds are restored, it is exactly the same as the vendor layout, or there are some differences?

dangowrt, 2021-11-21:
@sk23 The layout is embedded into the device-tree (for new bootchain) or bootloader parameters (`mtdparts`) (for old bootchain). Other than on block devices (think: SATA, SD card, eMMC, ...) the partitioning on raw flash is in most cases **not** decided by a partition table which is written to the flash itself (exception: eCos/RedBoot, not relevant here). Instead, the layout is defined in a way as embedded parameters for the bootloader as well as the Linux kernel.
The vendor bootloader and kernel expect things according to their layout, OpenWrt with UBI expects things differently.
So if you take a backup of the relevant regions of the flash it is crucial that you use the **same layout** which was used to create the back also to restore it. As the files you find in `boot_backup` volume correspond to partitions as defined for OpenWrt with UBI, the best thing you can do is to update (to have `kmod-mtd-rw`) and then use the `recovery` (initramfs) image.

sk23, 2021-11-21:
@dangowrt  Thank you for your patient explanation, I have learned a lot.

closed 2021-12-24 by dangowrt