dangowrt/owrt-ubi-installer #15: Unable to install UBI version on stock RT3200 (AX)

nickwood, 2021-06-23:
I've followed the instructions [here](https://github.com/dangowrt/linksys-e8450-openwrt-installer) to install on a stock router. Using the precompiled (v0.5) files [here](https://github.com/dangowrt/linksys-e8450-openwrt-installer/releases)

The first flash is successful, and I'm met with an OpenWRT login, I'm not convinced the second is successful though as it doesn't preserve settings on a reboot. Running df over SSH shows a bunch of tmpfs entries.

Any suggestions for how to proceed? Can I safely flash it using the 0.4 version of the sysupgrade file?

Many thanks,

dangowrt, 2021-06-23:
You can, but it shouldn't make a difference. The v0.5 generally works and as the router also apparently comes up in recovery mode, the most critical part has already succeeded.
Could be caused by not unchecking the 'keep settings' box or accidentally flashing the `*installer.itb` instead of the `*sysupgrade.itb` image.

Can you post the output of the following commands after login via SSH:

ubus call system board

mount

logread

This hasn't happened before and by now probably quite a few people have successfully flashed the v0.5 image (I see them in the forums), so I'd like to know what's going on :)

Don't worry, your device will not be lost, I will help you to recover it and, if needed, replace it for you if you send it to me by mail.

nickwood, 2021-06-23:
It seems like redownloading the sysupgrade file and retrying seemed to do the trick. Perhaps it got corrupted in transfer

dangowrt, 2021-06-23:
Sounds plausible, and I'm glad to see that in that case the system just went straight back into recovery, no brick :)

nickwood, 2021-06-25:
I'm really impresed! I even did the first flash over wifi (without thinking) but thankfully went off without an issue. Do you have a way for people to support you? Patreon/buymeacoffee or similar? I'd love to chip in a few quid

dangowrt, 2021-06-25:
Thank you, it feels good to get positive feedback for work I did.

If you like, donate to SFC as that helps us covering running costs of the OpenWrt.org project, see

https://openwrt.org/donate

nickwood, 2021-06-25:
Done! Thanks so much for all your hard wrok

closed 2021-06-25 by nickwood