dangowrt/owrt-ubi-installer #16: Difference between official build system and this script

bovi, 2021-07-10:
I have a quick question regarding the difference between your installer script and the official build tooling of OpenWRT.

I didn't use the build system of OpenWRT for a very long time (4+ years) but in the past I remember that when I choose the target in the "make menuconfig" it also will generate a release image. Is my understanding correct that your image has some advantages (using UBI) compared to the official build tooling or is the difference somewhere else?

Either way thanks a lot for your work here, I actually notice the E8450 router due to your work and I already ordered one and waiting now for delivery (-:

dangowrt, 2021-07-15:
This installer wraps around OpenWrt snapshot binaries, what you get **is** just a normal OpenWrt snapshot build.
While the OpenWrt build system allows to generate images compatible with vendor firmware, it cannot (yet) generate modified initramfs content for something like an installer. The script in this repository mainly just does that: Take binaries from https://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/ and re-wrap them, so we can have an easy installer instead of having to manually replace the bootloader, create UBI, ... when coming from the non-UBI vendor layout (ie. first installation when still running stock firmware). Once that's done, you can then again just use the official build system unmodified, select the UBI-variant of the device there and compile yourself or use the binaries from downloads.openwrt.org.

bovi, 2021-07-18:
Thanks for the explanation!

closed 2021-07-18 by bovi