dangowrt/owrt-ubi-installer #17: how to specify additional packages

ferboiar, 2021-07-14:
hi, i know this is not the place to look for support, but i haven't seen that there is a specific thread in the openwrt forum, so i try to ask here ...

I'm trying to prepare the environment to generate an image with your script ... But I have a couple of doubts with the "build_installer.sh" file:

To add new packages, should I include them in "OPENWRT_ADD_PACKAGES"? And should I include the package and its dependencies? for example, if I want to include adblock, just add "luci-app-adblock"? or should I also include "adblock" ?:

OPENWRT_ADD_PACKAGES = "luci-app-adblock" or
OPENWRT_ADD_PACKAGES = "luci-app-adblock adblock" ?

And... What's the point of "OPENWRT_REMOVE_PACKAGES"? I see that it deletes packages installed by "OPENWRT_UPG_PACKAGES" ...

dangowrt, 2021-07-14:
The scripts generates 3 images in total:
First, it uses ImageBuilder to generate the `sysupgrade` That's where `OPENWRT_UPG_PACKAGES` is being used as a parameter passed to the ImageBuilder as `PACKAGES`.

Next, the installer re-packs the initramfs image, removes packages in `OPENWRT_REMOVE_PACKAGES` and adds packages on `OPENWRT_ADD_REC_PACKAGES`. This is the `recovery` image.

Then the installer re-packs the initramfs image, removes packages in `OPENWRT_REMOVE_PACKAGES` and adds packages on `OPENWRT_ADD_PACKAGES`, and adds files for the installer itself (script, bootloader files, recovery image). This is the `installer image`

ferboiar, 2021-07-14:
thx for the answer dangowrt

So if I want to add packages to the image that is installed for the first time (installer image?), I have to add them to OPENWRT_ADD_PACKAGES var.

And what about the package list? Do I have to write both the luci-app and the CLI one? or is the luci-app package enough? I have doubts about what to do with packages like banIP or adblock, if I should indicate "luci-app-banip" alone, or "banip luci-app-banip"

dangowrt, 2021-07-14:
First run (invisisble, writes bootloader and recovery and reboots): `OPENWRT_ADD_PACKAGES`

Recovery stage (first time you see web-interface): `OPENWRT_ADD_REC_PACKAGES`

Final production system (sysupgrade): `OPENWRT_UPD_PACKAGES`

Regarding how to list packages: In your example, tt's enough to list `luci-app-banip` as it depends on `banip` and will make sure it's installed. Same for `adblock`.

ferboiar, 2021-07-15:
So, where should I indicate new packages to be included in the final producion system? I understand that i should do it in OPENWRT_UPD_PACKAGES, right?

dangowrt, 2021-07-15:
Yes it's that variable.
If you just want to generate a `sysugprade` image which can be loaded using this installer or an existing OpenWrt installation, you do not need to modify the installer for that. You can just use https://chef.libremesh.org/ or the OpenWrt ImageBuilder directly.

ferboiar, 2021-07-15:
oks, thank you very much for your help dango :)

closed 2021-07-15 by dangowrt