dangowrt/owrt-ubi-installer #18: *sysupgrade.itb file has not been created

ferboiar, 2021-07-16:
hello, after launching the script (build_installer.sh) two files are created:
```
openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb
openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery.itb
```
But sysupgrade file is missing:
`openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb
`
[image]

the script output:
https://pastebin.com/U2r1Bt8Y

is there anything i can do to find out what happened?

neilsan1366, 2021-07-17:
i didn't understand ? sorry but you will make a update a software and then install via openwrt ubi.sysupgrade.itb .

dangowrt, 2021-07-18:
In the log output you can see that the imagebuilder suddenly breaks after preparing the filesystem. Probably this is somehow related to your buildhost, I have so far only tried it on hosts running either Archlinux or Debian-based distributions. While it should work well (in theory) on every Linux x86_64 host, there are maybe obstacles I haven't been aware of.
Please let us know a bit more about your buildhost, like the distribution you are using.

If all you want is creating a `*sysupgrade` image, you anyway do not need this script. Just use the OpenWrt ImageBuilder locally or use https://chef.libremesh.org/ .

ferboiar, 2021-07-18:
the truth is I can't see where, in the log, the failure occurs

my VM runs ubuntu 21.04, I was going to use debian but I read here https://github.com/dangowrt/linksys-e8450-openwrt-installer/issues/8 that he had problems with debian and not with ubuntu.

I really don't know what I need, I want to do an installation on a new router ... And following these steps: https://github.com/dangowrt/linksys-e8450-openwrt-installer#steps talks about two files:
```
openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb 
openwrt-mediatek-mt7622-linksys_e8450-ubi-sysupgrade.itb
```

Excuse so many questions and clumsiness on my part, it is the first time that I do all this....

For now I'm going to install a new VM with Debian, let's see what happens ...

dangowrt, 2021-07-18:
That things don't work out on Ubuntu is even more weird, because that's what I also use to generate the images you find here for download on github.com, see https://github.com/dangowrt/linksys-e8450-openwrt-installer/blob/main/.github/workflows/blank.yml#L20

It's even more weird that generating the `*sysupgrade.itb` image fails, because that's the most trivial part of the whole story (just calling ImageBuilder and let it do the job), the `*initramfs-recovery.itb` and `*initramfs-recovery-installer.itb` are the more involved part. I assume you didn't modify the installer script, right?

And don't worry about asking questions, that's what's meant to be done with them :smiley_cat:

ferboiar, 2021-07-19:
Thanks Dango. I didn't modify the script, I just included additional packages in the `OPENWRT_UPD_PACKAGE`S variable. But I'm going to try two things, one to see what happens with Debian, and the second one to leave the script with the default packages....

ferboiar, 2021-07-20:
Solved.

Learned lessons:
1) with debian the script does not work, it is impossible to compile `unfit` app
2) redirecting the error output to a file helps a lot (`2>&1`)
3) if it seems that the router is bricked it is possible to do a hard reset before starting to sweat thinking about the jtag method (https://wiki.dd-wrt.com/wiki/index.php/Recover_from_a_Bad_Flash)
4) it is better to install the openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb file generated here than the one generated by yourself

1) with ubuntu everything seems to work without problems
2) I was not seeing the error generated by `Make` that finally allowed me to identify why the sysupgrade.itb file was not being generated. From the log:

```
make [2]: *** [Makefile: 167: package_install] Error 255
make [1]: *** [Makefile: 122: _call_image] Error 2
make: *** [Makefile: 240: image] Error 2
+ BINDIR = / home / ferboiar / linksys-e8450-openwrt-installer-0.5.2 / openwrt-ib / bin / targets / mediatek / mt7622
+ '[' -d /home/ferboiar/linksys-e8450-openwrt-installer-0.5.2/openwrt-ib/bin/targets/mediatek/mt7622 ']'
+ mv /home/ferboiar/linksys-e8450-openwrt-installer-0.5.2/openwrt-ib/bin/targets/mediatek/mt7622/openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb / home / ferboiar / linksys-e8450-openwr>
mv: cannot stat '/home/ferboiar/linksys-e8450-openwrt-installer-0.5.2/openwrt-ib/bin/targets/mediatek/mt7622/openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb' : No such file or directory
```

It seems that there is a package that does not install, but it does not say which one. So I had to launch manually the `make` command to see what package was (also taken from the log):
'
+ cd /home/ferboiar/linksys-e8450-openwrt-installer-0.5.2/openwrt-ib
+ make image PROFILE = linksys_e8450-ubi 'PACKAGES = package list'
'
and now i can see that the package that fails to install is `tc`.  When I have removed it the 3 files compile without problems.

3) When installing the recovery-installer.itb file on the router, it stops responding and the openwrt website does not appear, after 20 minutes I have finally restarted it, but it does not seem to start (it gives disconnected network cable and the power led blink eternally). Restarting or resetting the router does not work, until I have read the previous link, I have tried it, and the router has been resurrected with the factory firmware (honestly, I doubt that it was the 30/30/30 reset method described there, but something has happened for sure...)

4) I did not want to risk trying to install my recovery file again, so I installed yours. And then my sysupgrade file..

thx dango for your work!

neilsan1366, 2021-07-20:
ok great if you have solved but it seemed to be a rare case as I have it seemed to me,

Do you think we might have this problem in the future with the new release of dangoWRT?

the only thing i do while being european and flash the rt3200 via ethernet in recovery install .itb then update to the latest version as you showed in ubi sysuprage .itb

thank you in advance

closed 2021-07-21 by ferboiar