dangowrt/owrt-ubi-installer #19: How To Setup As Access Point / Dumb AP?

AmbiguousWisdom, 2021-07-31:
Hello, recently I've been trying to set up the router as an access point for my OPNsense firewall, but to no avail. I have been trying to make it work through configuration via the web interfacing, but I think I may be doing something wrong. The source I've been following is https://openwrt.org/docs/guide-user/network/wifi/dumbap, & I've been following everything to the letter except step 6 under “Configuration via OpenWrt Web Interface LUCI” because I don't know where that is located (it appears that the web UI of this particular firmware is quite different from the standard ones). Either way, it seemed optional due to its description.

As for the problem, I honestly don't know what is wrong. When I save & apply the changes (before I'm supposed to connect the cable into my OPNsense firewall), it just doesn't want to apply. It “rolls back” to original settings & leaves my changes pending. To be more specific, although I am able to disable & stop firewall, dnsmasq and odhcpd in the list of startup scripts, the LAN configurations just won't go through regardless of whether I disable & stop firewall, dnsmasq and odhcpd in the list of startup scripts before or after. If there are logs I can provide let me know how since I'm quite new in the OpenWrt domain & have only been messing around with it for a day or two. Thanks.

dangowrt, 2021-07-31:
If you change the network configuration you also have to change it on the machine which you use to access the LuCI web interface. This is because LuCI checks if you can reach the device with it's new address and if not, it rolls back. Ie. you got 90s seconds after the change to access LuCI again and confirm (and you must not disconnect the device from power in that time, as reboot will also cause a rollback).

AmbiguousWisdom, 2021-07-31:
The problem is, I don't exactly understand when you say, “you also have to change it [the network config] on the machine which you use to access the LuCI web interface.” Would I unplug & replug the ethernet cable, or would it be something else?

dangowrt, 2021-07-31:
That depends on your host OS, I guess. In Linux you can just add an address in the to-be-set subnet to the interface (`ip addr add ...`). I guess in the end, it doesn't matter as long as you manage to access to router on it's now address within 90 seconds.
If all that is too much effort, you can just change the network config using SSH by editing `/etc/config/network`.

AmbiguousWisdom, 2021-07-31:
Ah thank you. The “ip addr add ...” command worked for me. As for '6. Under “Physical Settings” tab, ensure “Bridge interfaces” is ticked, and ensure BOTH of your interfaces (eth0, wlan0) are selected, in order to allow traffic between wireless and wired connections.' under Configuration via OpenWrt Web Interface LUCI is that like already applied? It seems that traffic between my phone & PC works via syncthing so that's why I made such an assumption.

dangowrt, 2021-07-31:
Yes, the pre-configured wifi interfaces are bridged to `lan` by default. So if all you did was editing those existing interfaces to change channel, ssid, ... your assumption holds true.

AmbiguousWisdom, 2021-07-31:
Gotcha gotcha. Thanks a lot for your help, I appreciate it.

closed 2021-07-31 by AmbiguousWisdom