dangowrt/owrt-ubi-installer #20: Thank you!

sibradzic, 2021-08-24:
Just wanted to say BIG FAT THANKS for keeping this repo updated and documenting the process of transition to OpenWRT in a such detailed manner. Much obliged :+1:

mrkiko, 2021-08-31:
+1 as well!!

JaceAlvejetti, 2022-05-16:
Adding my thanks here as well, flawless from backup to flash.

mikehardy, 2023-12-19:
I maintain a lot of github repos and I *hate hate* +1 comments but I just can't resist on this one.

My E8450 would lose it's mind (and more specifically it's internet port connection) every reboot when configured in bridge mode (even though it worked fine after reconfig / before reboot) but I purchased it because I knew it could run OpenWRT if I needed to. And yep, I needed to

Then here this repo is with the most useful install process ever to reformat the flash and get me moving quickly

Thank you so much!