dangowrt/owrt-ubi-installer #23: Provide non-snapshot releases?

slhck, 2021-09-02:
Thanks for this useful script!

I see it's hardcoded to download a snapshot build. This leads to compatibility problems down the road, such as the one [here](https://www.reddit.com/r/openwrt/comments/jjvy4n/cannot_install_kmod/) where a kernel mod from the packages cannot be installed.

Would it be possible to provide the last OpenWRT release as well?

dangowrt, 2021-09-02:
It will be part of the next OpenWrt release, probably being branched-off later this year and released (hopefully) early 2022.

slhck, 2021-09-02:
Thanks for the info.

neilsan1366, 2021-09-05:
hello @dangowrt  you are write "Support for the device was added after 21.02 had branched off, and is based on Kernel 5.10. There might be another release coming up later this year (branching October/November), and that will have support for it. Until then, you will have to use snapshots." so his more near of october or 2022 ? thanks for your work

dangowrt, 2021-09-05:
There is hope that branching off will still happen this year (2021) or very early 2022. October 2022 would be too late to start a release based on Linux 5.10. Now that 21.02.0 is out, people will focus on getting the remaining targets lifted to Linux 5.10.