dangowrt/owrt-ubi-installer #24: Installing kmod-tun

slhck, 2021-09-03:
I would like to add `kmod-tun` to the packages built by this repo, so that we can use [Tailscale](https://tailscale.com/) (i.e. Wireguard) directly from the device.

Will it be enough to add it here?

https://github.com/dangowrt/linksys-e8450-openwrt-installer/blob/bcfe1d7e2f30cf13cfd06505675a27eda0543182/build_installer.sh#L239-L244

Is that something you would consider adding to the official build?

On a side note, I am not quite sure what the difference is between the recovery and the sysupgrade firmware files. I have never worked with OpenWRT, but I have a client who I am guiding in flashing the router, and he says that he can run OpenWRT with the recovery image, but once he flashes the sysupgrade image, the device no longer responds. Is that even supposed to work, I mean, running from the recovery image alone? (I realize this is very tangential to this issue.)

dangowrt, 2021-09-03:
Ok, so what you got here in this repo is really just for initial installation. Of course, you could use this script to generate the sysupgrade image, but that's a bit of overkill.

> Is that something you would consider adding to the official build?

No, I'd like to keep it minimal as it is easy to add things later on.

If you want to bake you own firmware image for this (and other) devices with pre-installed package (as you mentioned), the easiest way would be to use https://chef.libremesh.org/ which allows you to customize OpenWrt images for any device supported by OpenWrt.

> On a side note, I am not quite sure what the difference is between the recovery and the sysupgrade firmware files. I have never worked with OpenWRT, but I have a client who I am guiding in flashing [...]

The concept of dual-booting `recovery` vs. `production` is also quite new to OpenWrt and is more commonly found on Android devices (though it's built there in a different way).

Regarding the issue of your client:
You should **not** flash the `recovery` image when still running stock firmware or non-UBI OpenWrt. It may work, but will certainly break things once you try to use to flash something else (ie. sysugprade image).
Please use the file [openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb](https://github.com/dangowrt/linksys-e8450-openwrt-installer/releases/download/v0.5.3/openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb) (note the word "installer" in the filename!) which will perform initialization of the flash (ubiformat) and also install the recovery to flash. Then use that to flash [openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb](https://github.com/dangowrt/linksys-e8450-openwrt-installer/releases/download/v0.5.3/openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb).

slhck, 2021-09-04:
Thanks for the detailed feedback. I understand the reasoning for not wanting to include anything more than necessary, however installing kernel mods seems to be a PITA with snapshot releases … anyway, I will close this issue then!

closed 2021-09-04 by slhck

dangowrt, 2021-09-04:
If the snapshot is more or less recent installing kmods should be no problem. I'd recommend to use `auc` to keep your snapshot up-to-date, then installing what ever kernel and userspace additions you want will just work.

snow212-cn, 2022-02-02:
> 

I'm new to openwrt and still bewildered by it.  I just found https://chef.libremesh.org/ you recommended very useful. So I want to ask about the blank `Custom script to run on first boot (optional)` what should I enter. What's the difference between it and the script in your repo other than the packages.

dangowrt, 2022-02-02:
The custom script you can add via chef.libremesh.org are added to the regular root filesystem of an image. It's in optional feature which allows you to run some commands, usually to modify some default settings. If you don't need this, jsut let it be empty.

To generate an installer like the one here in this repo, it is needed to add/modify the **`initramfs`** image which is not supported (yet) by the OpenWrt ImageBuilder (and hence also not supported by chef.libremesh.org which uses that).