dangowrt/owrt-ubi-installer #25: Stuck in recovery mode

Blaok, 2021-09-08:
Hi,

My router seems to stuck in recovery mode while I try to convert it to use UBI. I built the itb files on my own machine (which probably is a bad idea?) and flashed the recovery installer successfully. I then tried to flash the sysupgrade.itb file. However, the router just keeps rebooting into recovery mode. Only tmpfs is mounted and I cannot make any persistent changes. Some (probably) relevant information:

```
root@OpenWrt:~# ubus call system board
{
        "kernel": "5.10.61",
        "hostname": "OpenWrt",
        "system": "ARMv8 Processor rev 4",
        "model": "Linksys E8450 (UBI)",
        "board_name": "linksys,e8450-ubi",
        "rootfs_type": "initramfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "SNAPSHOT",
                "revision": "r17485-ae33ce2af2",
                "target": "mediatek/mt7622",
                "description": "OpenWrt SNAPSHOT r17485-ae33ce2af2"
        }
}
root@OpenWrt:~# mount
tmpfs on / type tmpfs (rw,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
tmpfs on /dev type tmpfs (rw,nosuid,relatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,mode=600,ptmxmode=000)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
none on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,noatime,mode=700)
pstore on /sys/fs/pstore type pstore (rw,noatime)
root@OpenWrt:~# logread
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
Tue Sep  7 17:47:20 2021 kern.notice kernel: [    0.000000] Linux version 5.10.61 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 10.3.0 r17485-ae33ce2af2) 10.3.0, GNU ld (GNU Binutils) 2.36.1) #0 SMP Tue Sep 7 11:23:16 2021
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000] Machine model: Linksys E8450 (UBI)
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000] earlycon: uart8250 at MMIO32 0x0000000011002000 (options '')
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000] printk: bootconsole [uart8250] enabled
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000] Zone ranges:
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000]   DMA      [mem 0x0000000040000000-0x000000005fffffff]
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000]   DMA32    empty
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000]   Normal   empty
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000] Movable zone start for each node
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000] Early memory node ranges
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000043000000-0x000000004302ffff]
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000043030000-0x000000005fffffff]
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000005fffffff]
Tue Sep  7 17:47:20 2021 kern.debug kernel: [    0.000000] On node 0 totalpages: 131072
Tue Sep  7 17:47:20 2021 kern.debug kernel: [    0.000000]   DMA zone: 2048 pages used for memmap
Tue Sep  7 17:47:20 2021 kern.debug kernel: [    0.000000]   DMA zone: 0 pages reserved
Tue Sep  7 17:47:20 2021 kern.debug kernel: [    0.000000]   DMA zone: 131072 pages, LIFO batch:31
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000] psci: probing for conduit method from DT.
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000] psci: PSCIv1.1 detected in firmware.
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000] psci: Using standard PSCI v0.2 function IDs
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000] psci: MIGRATE_INFO_TYPE not supported.
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000] psci: SMC Calling Convention v1.2
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000] percpu: Embedded 20 pages/cpu s43672 r8192 d30056 u81920
Tue Sep  7 17:47:20 2021 kern.debug kernel: [    0.000000] pcpu-alloc: s43672 r8192 d30056 u81920 alloc=20*4096
Tue Sep  7 17:47:20 2021 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000] Detected VIPT I-cache on CPU0
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000] CPU features: detected: ARM erratum 845719
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000] CPU features: kernel page table isolation disabled by kernel configuration
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000] CPU features: detected: ARM erratum 843419
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
Tue Sep  7 17:47:20 2021 kern.notice kernel: [    0.000000] Kernel command line: earlycon=uart8250,mmio32,0x11002000 console=ttyS0,115200n1 swiotlb=512
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Tue Sep  7 17:47:20 2021 kern.info kernel: [    0.000000] Memory: 488588K/524288K available (8126K kernel code, 874K rwdata, 2212K rodata, 448K init, 303K bss, 35700K reserved, 0K cma-reserved)
[... 296 lines cut ...]
Tue Sep  7 17:47:28 2021 daemon.notice netifd: Interface 'loopback' is now up
Tue Sep  7 17:47:28 2021 kern.info kernel: [   17.467371] mt7530 mdio-bus:00 wan: configuring for phy/gmii link mode
Tue Sep  7 17:47:28 2021 kern.info kernel: [   17.474937] 8021q: adding VLAN 0 to HW filter on device wan
Tue Sep  7 17:47:28 2021 kern.info kernel: [   17.479268] mt7530 mdio-bus:00 wan: Link is Up - 1Gbps/Full - flow control off
Tue Sep  7 17:47:28 2021 daemon.notice netifd: Interface 'wan' is enabled
Tue Sep  7 17:47:28 2021 daemon.notice netifd: Interface 'wan6' is enabled
Tue Sep  7 17:47:28 2021 daemon.notice netifd: Network device 'eth0' link is up
Tue Sep  7 17:47:28 2021 daemon.notice netifd: Network device 'lan3' link is up
Tue Sep  7 17:47:28 2021 daemon.notice netifd: Network device 'lo' link is up
Tue Sep  7 17:47:28 2021 daemon.notice netifd: Interface 'loopback' has link connectivity
Tue Sep  7 17:47:28 2021 daemon.notice netifd: Network device 'wan' link is up
Tue Sep  7 17:47:28 2021 daemon.notice netifd: Interface 'wan' has link connectivity
Tue Sep  7 17:47:28 2021 daemon.notice netifd: Interface 'wan' is setting up now
Tue Sep  7 17:47:28 2021 daemon.notice netifd: Interface 'wan6' has link connectivity
Tue Sep  7 17:47:28 2021 daemon.notice netifd: Interface 'wan6' is setting up now
Tue Sep  7 17:47:28 2021 kern.info kernel: [   17.488114] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
Tue Sep  7 17:47:28 2021 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Tue Sep  7 17:47:28 2021 daemon.notice netifd: wan (2350): udhcpc: started, v1.34.0
Tue Sep  7 17:47:28 2021 daemon.err odhcp6c[2360]: Failed to send RS (Address not available)
Tue Sep  7 17:47:28 2021 daemon.notice netifd: wan (2350): udhcpc: broadcasting discover
Tue Sep  7 17:47:28 2021 daemon.notice procd: /etc/rc.d/S96led: setting up led WAN

```

Thanks!

dangowrt, 2021-09-09:
Most likely the router crashed for some reason and now boots into recovery mode for that reason (as U-Boot checks PSTORE/ramoops for previous crashes and if the system had previously crashed, starts in recovery mode).
Please check if there are any files in `/sys/fs/pstore` and if so, they should contain the kernel log including the crash, ie. `cat /sys/fs/pstore/*` should tell you what went wrong.
For the system to boot normally again you will need to remove them using `rm /sys/fs/pstore/*`.

Blaok, 2021-09-10:
@dangowrt Thank you so much for your reply! Indeed, there are kernel logs in that directory. I'll remove them and try to reboot the router when I got the chance.

About the crash, the first few lines of `/sys/fs/pstore/dmesg-ramoops-1` say that there are bad blocks. Am I correct to assume the UBI version can handle them automatically? I was running the non-UBI version of OpenWRT for a couple of months so I assume the bad blocks are not surprising, but since they were generated before the UBI version is flashed, I'm not sure what will happen when I reboot.

dangowrt, 2021-09-10:
Yes, UBI should handle bad-blocks, however, your log say
```
Tue Sep  7 17:47:20 2021 kern.notice kernel: [    3.107009] ubi0: good PEBs: 1000, bad PEBs: 0, corrupted PEBs: 0
```

Maybe share that crash log you found in PSTORE...?

Blaok, 2021-09-10:
Here are the files: [dmesg-ramoops.tar.gz](https://github.com/dangowrt/linksys-e8450-openwrt-installer/files/7145361/dmesg-ramoops.tar.gz)

I was unable to read the content of `dmesg-ramoops-0.enc.z` since its header seems corrupted. `dmesg-ramoops-1` is also somewhat corrupted but is readable.

Thanks again for your time!

Blaok, 2021-09-11:
Update: I removed the crash logs and rebooted the router. Everything appears to work fine now. Thank you for this wonderful project and your useful suggestions!

closed 2021-09-11 by Blaok