dangowrt/owrt-ubi-installer #26: Wireguard issue

rjordan, 2021-09-16:
kmod-wireguard refuses to install on 0.5.3.

Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.64-1-6c1889bac5024893a9a66c4b6f7b4252) for kmod-crypto-lib-chacha20
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.64-1-6c1889bac5024893a9a66c4b6f7b4252) for kmod-crypto-kpp
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.64-1-6c1889bac5024893a9a66c4b6f7b4252) for kmod-wireguard
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-wireguard:
 *      kernel (= 5.10.64-1-6c1889bac5024893a9a66c4b6f7b4252)
 * opkg_install_cmd: Cannot install package kmod-wireguard.

dangowrt, 2021-09-16:
Please run `auc` first to update to the most recent snapshot.

rjordan, 2021-09-16:
That solves it.  Sorry was a bit afraid to run auc on a forked disto... only been running releases.  Thanks for the help, closing the issue.

closed 2021-09-16 by rjordan

dangowrt, 2021-09-17:
You are right to be careful with `auc` on snapshots.
However, once you run the installer that also writes a kernel+initramfs system to the flash for recovery. In case something goes wrong, no matter what it's not a brick as you can always easily boot into that recovery system. If the system crashes that even happens automatically.
In the event the device boots and for what ever reason (lets say: broken Ethernet or switch driver) is not responsive but also doesn't crash you can force booting into recovery by holding the RESET button.