dangowrt/owrt-ubi-installer #27: OpenWrt r17443 5GHz radio appears to not be detected properly

GMMan, 2021-09-27:
Just a heads up that it seems my 5GHz radio disappeared when upgrading to the image in v0.5.3. Don't know if config issue or a bug.

dangowrt, 2021-09-28:
When coming from builds before  openwrt/openwrt@050621a you will need to flash the installer again, ie.
```
sysupgrade -F https://github.com/dangowrt/linksys-e8450-openwrt-installer/releases/download/v0.5.3/openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb
```

This is because the SPI-NAND driver has been replaced and the new driver is very picky...

closed 2021-10-03 by dangowrt

ShadGudmunson, 2022-01-16:
Hello,

I am having this same issue. I have attempted the above fix with versions 0.5.3, 0.6.0, and 0.6.1 with no luck. I have tried both with and without retaining settings. 

When running `iwinfo` I only see the 802.11bgn radio.

```
root@OpenWrt:~# iwinfo
wlan0     ESSID: unknown
          Access Point: C4:41:1E:F8:1F:DB
          Mode: Client  Channel: unknown (unknown)
          Center Channel 1: unknown 2: unknown
          Tx-Power: 6 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: 14C3:7622 14C3:7622 [MediaTek MT7622]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0
```
Is there another solution that I am missing? I am happy to provide any logs or command output to help troubleshooting. 

Thank you for your time!

mrkiko, 2022-01-16:
>     radio appears to not be detected properly (#27)

While in recovery mode, mt7915e driver is not present, only the mt7622's builtin radio is detected.

Pleasedo try out with latest installer version and use a standard squashfs sysupgrade image for ubi layout. The mt7915e radio should be detected in that case.

ShadGudmunson, 2022-01-16:
Unfortunately even after applying the sysupgrade the radio is undetected.

mrkiko, 2022-01-16:
May you send your dmesg and the output of the “lspci” command?

Thanks!

ShadGudmunson, 2022-01-16:
Here you go.

```
root@OpenWrt:~# lspci
00:00.0 PCI bridge: MEDIATEK Corp. Device 3258
01:00.0 Unclassified device [0002]: MEDIATEK Corp. MT7915E 802.11ax PCI Express Wireless Network Adapter
```
[dmesg.txt](https://github.com/dangowrt/linksys-e8450-openwrt-installer/files/7877912/dmesg.txt)

dangowrt, 2022-01-16:
Please share the output of `lsmod` command and make sure the `mt7915e` module appears there. If not, then for some reason the driver is not installed in the image you are using. Using the `sysupgrade` image downloadable here should fix that.

Another reason could be that the calibration data was not fixed up (it should have been by re-running the installer). To validate that, please try
```
hexdump -C /dev/mtd2
```
and share that here so we can see what's going on.

ShadGudmunson, 2022-01-16:
It does not appear that mt7915e is showing up. I will try the sysupgrade again and report back.
```
root@OpenWrt:~# lsmod
cfg80211              286720  4 mt7615_common,mt76_connac_lib,mt76,mac80211
compat                 16384  2 mac80211,cfg80211
crc_ccitt              16384  1 ppp_async
gpio_button_hotplug    16384  0
hwmon                  20480  1 mt7615_common
ip_tables              32768  4 iptable_nat,iptable_mangle,iptable_filter
ip6_tables             32768 24 ip6table_mangle,ip6table_filter
ip6t_REJECT            16384  2
ip6table_filter        16384  1
ip6table_mangle        16384  1
ipt_REJECT             16384  2
iptable_filter         16384  1
iptable_mangle         16384  1
iptable_nat            16384  1
leds_gpio              16384  0
mac80211              520192  4 mt7615e,mt7615_common,mt76_connac_lib,mt76
mt76                   53248  3 mt7615e,mt7615_common,mt76_connac_lib
mt76_connac_lib        28672  2 mt7615e,mt7615_common
mt7615_common          77824  1 mt7615e
mt7615e                24576  0
nf_conntrack           94208  8 xt_state,xt_nat,xt_conntrack,xt_REDIRECT,xt_MASQUERADE,xt_CT,nf_nat,nf_flow_table
nf_defrag_ipv4         16384  1 nf_conntrack
nf_defrag_ipv6         20480  1 nf_conntrack
nf_flow_table          32768  1 xt_FLOWOFFLOAD
nf_log_common          16384  2 nf_log_ipv4,nf_log_ipv6
nf_log_ipv4            16384  0
nf_log_ipv6            16384  0
nf_nat                 40960  4 iptable_nat,xt_nat,xt_REDIRECT,xt_MASQUERADE
nf_reject_ipv4         16384  1 ipt_REJECT
nf_reject_ipv6         16384  1 ip6t_REJECT
ppp_async              24576  0
ppp_generic            45056  3 pppoe,ppp_async,pppox
pppoe                  24576  0
pppox                  16384  1 pppoe
seqiv                  16384  0
slhc                   16384  1 ppp_generic
x_tables               28672 24 xt_state,xt_nat,xt_conntrack,xt_REDIRECT,xt_MASQUERADE,xt_FLOWOFFLOAD,xt_CT,ipt_REJECT,xt_time,xt_tcpudp,xt_multiport,xt_mark,xt_mac,xt_limit,xt_comment,xt_TCPMSS,xt_LOG,iptable_mangle,iptable_filter,ip_tables,ip6table_mangle,ip6table_filter,ip6_tables,ip6t_REJECT
xt_CT                  16384  0
xt_FLOWOFFLOAD         20480  0
xt_LOG                 16384  0
xt_MASQUERADE          16384  1
xt_REDIRECT            16384  0
xt_TCPMSS              16384  4
xt_comment             16384127
xt_conntrack           16384 14
xt_limit               16384 20
xt_mac                 16384  0
xt_mark                16384  0
xt_multiport           16384  0
xt_nat                 16384  0
xt_state               16384  0
xt_tcpudp              16384 12
xt_time                16384  0
```
[mtd2.txt](https://github.com/dangowrt/linksys-e8450-openwrt-installer/files/7878147/mtd2.txt)
[lsmod.txt](https://github.com/dangowrt/linksys-e8450-openwrt-installer/files/7878152/lsmod.txt)

ShadGudmunson, 2022-01-16:
After running the sysupgrade it is appearing in the lsmod output however, it is still not appearing in iwinfo or in LuCI.
```
root@OpenWrt:~# lsmod
autofs4                36864  1 
cfg80211              294912  5 mt7915e,mt7615_common,mt76_connac_lib,mt76,mac80211
compat                 16384  2 mac80211,cfg80211,[permanent]
crc_ccitt              16384  1 ppp_async
exfat                  65536  0 
fat                    69632  1 vfat
gpio_button_hotplug    16384  0 
hwmon                  20480  2 mt7915e,mt7615_common
ip_tables              32768  4 iptable_nat,iptable_mangle,iptable_filter
ip6_tables             32768 24 ip6table_mangle,ip6table_filter
ip6t_REJECT            16384  2 
ip6table_filter        16384  1 
ip6table_mangle        16384  1 
ipt_REJECT             16384  2 
iptable_filter         16384  1 
iptable_mangle         16384  1 
iptable_nat            16384  1 
leds_gpio              16384  0 
mac80211              528384  5 mt7915e,mt7615e,mt7615_common,mt76_connac_lib,mt76
mt76                   57344  4 mt7915e,mt7615e,mt7615_common,mt76_connac_lib
mt76_connac_lib        28672  2 mt7615e,mt7615_common
mt7615_common          77824  1 mt7615e
mt7615e                28672  0 
mt7915e                98304  0 
nf_conntrack           94208  8 xt_state,xt_nat,xt_conntrack,xt_REDIRECT,xt_MASQUERADE,xt_CT,nf_nat,nf_flow_table
nf_defrag_ipv4         16384  1 nf_conntrack
nf_defrag_ipv6         20480  1 nf_conntrack
nf_flow_table          32768  1 xt_FLOWOFFLOAD
nf_log_common          16384  2 nf_log_ipv4,nf_log_ipv6
nf_log_ipv4            16384  0 
nf_log_ipv6            16384  0 
nf_nat                 40960  4 iptable_nat,xt_nat,xt_REDIRECT,xt_MASQUERADE
nf_reject_ipv4         16384  1 ipt_REJECT
nf_reject_ipv6         16384  1 ip6t_REJECT
nls_cp437              20480  0 
nls_iso8859_1          16384  0 
nls_utf8               16384  0 
ppp_async              24576  0 
ppp_generic            45056  3 pppoe,ppp_async,pppox
pppoe                  24576  0 
pppox                  16384  1 pppoe
seqiv                  16384  0 
slhc                   16384  1 ppp_generic
uas                    28672  0 
usb_storage            65536  1 uas
vfat                   20480  0 
x_tables               28672 24 xt_state,xt_nat,xt_conntrack,xt_REDIRECT,xt_MASQUERADE,xt_FLOWOFFLOAD,xt_CT,ipt_REJECT,xt_time,xt_tcpudp,xt_multiport,xt_mark,xt_mac,xt_limit,xt_comment,xt_TCPMSS,xt_LOG,iptable_mangle,iptable_filter,ip_tables,ip6table_mangle,ip6table_filter,ip6_tables,ip6t_REJECT
xhci_plat_hcd          16384  0 
xt_CT                  16384  0 
xt_FLOWOFFLOAD         20480  0 
xt_LOG                 16384  0 
xt_MASQUERADE          16384  1 
xt_REDIRECT            16384  0 
xt_TCPMSS              16384  4 
xt_comment             16384127 
xt_conntrack           16384 14 
xt_limit               16384 20 
xt_mac                 16384  0 
xt_mark                16384  0 
xt_multiport           16384  0 
xt_nat                 16384  0 
xt_state               16384  0 
xt_tcpudp              16384 12 
xt_time                16384  0 
```

[mtd2.txt](https://github.com/dangowrt/linksys-e8450-openwrt-installer/files/7878201/mtd2.txt)

dangowrt, 2022-01-17:
`mtd2` looks good, so that worked and is not the problem.
Did `dmesg` output change since now the module at least got loaded (it should also show us something in the kernel log, `dmesg | grep mt7915` should tell)?
And does `iw list` show the 5 GHz interface?

If you kept your previous configuration while doing sysupgrade? If so (and as your previous image was apparently somehow broken), try resetting configuration using `firstboot`.
As the driver was completely missing in the image you were running before: where did that came from? Was that a local build you compiled yourself?

ShadGudmunson, 2022-01-17:
Looks like there is some error from dmesg I've included the full output of dmesg aswell in case that is helpful.
```
root@OpenWrt:~# dmesg | grep mt7915
[    8.780267] mt7915e 0000:01:00.0: assign IRQ: got 143
[    8.790434] mt7915e 0000:01:00.0: enabling device (0000 -> 0002)
[    8.796560] mt7915e 0000:01:00.0: enabling bus mastering
[    8.876015] mt7915e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20201105222230a
[    9.024512] mt7915e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20201105222304
[    9.083814] mt7915e 0000:01:00.0: WA Firmware Version: DEV_000000, Build Time: 20201105222323
[    9.217710] mt7915e 0000:01:00.0: reading EEPROM from mtd factory failed: -117
[    9.264227] mt7915e: probe of 0000:01:00.0 failed with error -22
```
```
root@OpenWrt:~# iw list
Wiphy phy0
	wiphy index: 0
	max # scan SSIDs: 4
	max scan IEs length: 2304 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports AP-side u-APSD.
	Device supports T-DLS.
	Available Antennas: TX 0xf RX 0xf
	Configured Antennas: TX 0xf RX 0xf
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x1ff
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 3839 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: No restriction (0x00)
		HT TX/RX MCS rate indexes supported: 0-31
		Frequencies:
			* 2412 MHz [1] (20.0 dBm)
			* 2417 MHz [2] (20.0 dBm)
			* 2422 MHz [3] (20.0 dBm)
			* 2427 MHz [4] (20.0 dBm)
			* 2432 MHz [5] (20.0 dBm)
			* 2437 MHz [6] (20.0 dBm)
			* 2442 MHz [7] (20.0 dBm)
			* 2447 MHz [8] (20.0 dBm)
			* 2452 MHz [9] (20.0 dBm)
			* 2457 MHz [10] (20.0 dBm)
			* 2462 MHz [11] (20.0 dBm)
			* 2467 MHz [12] (20.0 dBm) (no IR)
			* 2472 MHz [13] (20.0 dBm) (no IR)
			* 2484 MHz [14] (20.0 dBm) (no IR)
	valid interface combinations:
		 * #{ IBSS } <= 1, #{ managed, AP, mesh point, P2P-client, P2P-GO } <= 16,
		   total <= 16, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz, 80+80 MHz, 160 MHz }

	HT Capability overrides:
		 * MCS: ff ff ff ff ff ff ff ff ff ff
		 * maximum A-MSDU length
		 * supported channel width
		 * short GI for 40 MHz
		 * max A-MPDU length exponent
		 * min MPDU start spacing
	max # scan plans: 1
	max scan plan interval: 0
	max scan plan iterations: 0
	Supported extended features:
		* [ VHT_IBSS ]: VHT-IBSS
		* [ RRM ]: RRM
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
		* [ AQL ]: Airtime Queue Limits (AQL)
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ DEL_IBSS_STA ]: deletion of IBSS station support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
```

I am uncertain which image I had last flashed. It was either one of 0.5.3, 0.6.0, and 0.6.1 or the most recent snapshot from openwrt. Retrieved from https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=mediatek%2Fmt7622&id=linksys_e8450-ubi

It appears that `firstboot` did not help and seems to have broken something in LuCI as the `wireless` page will now not load.

[dmesg2.txt](https://github.com/dangowrt/linksys-e8450-openwrt-installer/files/7878236/dmesg2.txt)

ShadGudmunson, 2022-01-17:
It appears I have fixed the problem. After running `firstboot` there were some issues with LaCI. To resolve these problems I re-flashed the most recent recovery installer and sysupgrade. After this the radio reappeared and seems to be working as normal.

Sorry I don't seem to have anything more useful.

dangowrt, 2022-01-17:
Did you ever re-run the installer before? Because this was needed (scroll up in this issue ticket) to overcome problems with broken OOB data as the installer (from v0.5.3 on) would re-write the `factory` partition in order to fix that. I'm asking because I'm still seeing those errors in dmesg
```
blk_update_request: I/O error, dev mtdblock2, 
```
So that looks like you never fixed OOB data (ie. you just used sysupgrade image and never re-run the installer after v0.5.3) and it may work now due to the mt7915e wifi driver being more tolerant with this kinda situations after https://github.com/openwrt/mt76/commit/25237b19bcc134490a8e1c6b809ef7a91f1219ce and https://github.com/openwrt/mt76/commit/71e08471ab560ab621b4b96e2f11a43f08b8a9c8 .