dangowrt/owrt-ubi-installer #32: Is it possible to boot into recovery WITHOUT losing customization?

brentr, 2021-10-13:
Some Gl.inet routers have two copies of firmware.  One in NOR flash, another in NAND.
(GL-AR300M, for example)
Normally, the larger NAND flash is the rootfs, but it is possible to boot with the NOR flash as rootfs, mount the NAND flash from there and correct whatever configuration screw-ups might have broken the NAND flash copy.

This is a godsend for folks running complex setups with many extra OpenWRT packages.
Resetting to factory defaults means hours of reconfiguration and test.

It seems that this router could accomplish the same trick, if it were possible to reboot into recovery mode (root on the initramfs) WITHOUT clearing the UBI partitions.
One would boot into recovery, shell into the router, mount the UBIFS, and correct it before attempting to reboot.

Is this already possible?

If not, do you see reason it would be especially difficult to implement?

By the way, thanks for making all this.  It's been rock solid for me (despite being on the bleeding edge :-)

brentr, 2021-10-14:
That's a cool way to force a boot info recovery without physical access to the router.
If I have physical access to the router, can I not also boot into recovery by holding down the reset button while it powers up?

More importantly...
My impression was that, WHENEVER the router boots into recovery, the normal UBIFS root filesystem gets cleared.

This clearing of the UBIFS is what I'm trying to avoid.
I want to boot into recovery, mount the UBIFS without clearing it, modify it and reboot normally.

If the UBIFS is not cleared by booting into recovery, when does it get cleared?

brentr, 2021-10-14:
failsafe mode should work well for me.  Thank you!

Looking at the output for fw_printenv, I see that I'm running U-Boot 2021.04 (Aug 27 2021...)
This includes the definition for:
`bootcmd=if pstore check ; then run boot_recovery ; else run boot_ubi ; fi`

Will running "boot_recovery" will clear the production UBIFS, or will it leave it intact?

It seems that u-boot has been modified so "boot_first" somehow runs before "bootcmd":
`boot_first=if button reset ; then led $bootled_rec on ; run boot_tftp_recovery ; setenv flag_recover 1 ; run boot_default ; fi ; bootmenu`
--> This appears to be the logic booting to recovery if the reset button is pressed.

However, what is reading the 'boot_first' env variable?
I can't seem to find any reference to 'boot_first' in the u-boot sources either.

dangowrt, 2021-10-14:
> Will running "boot_recovery" will clear the production UBIFS, or will it leave it intact?

`boot_recovery` will not wipe or otherwise destroy the existing rootfs_data nor anything else. **only** if the recovery image is missing or broken, the bootloader will try to load it via TFTP and then write it to flash. In order to make sure there is enough space, rootfs_data is being deleted at this point. However, this should really only happen in case you somehow broke the recovery image stored in flash by the installer, and even in that case, it should happen only once.

Regarding the logic of the button:
There is a patch which makes the bootloader reset the environment in case it is held down during boot:

https://github.com/openwrt/openwrt/blob/master/package/boot/uboot-mediatek/patches/300-mt7622-generic-reset-button-ignore-env.patch

Regarding the complete default environment (to answer your question "what is calling boot_first?"), find it here:
https://github.com/openwrt/openwrt/blob/master/package/boot/uboot-mediatek/patches/410-add-linksys-e8450.patch#L349

brentr, 2021-10-15:
That's a relief. Thanks.

If I understand these u-boot env definitions, one must apply the 300-mt7622-generic-reset-button-ignore-env.patch to enable pressing reset during power up to reach the boot_first definition, as it is only referenced by _firstboot, which deletes itself during the first boot after the env is initialized.

Correct?

If so, is there any reason not to allow booting recovery without reinitializing the U-Boot environment?
I'm thinking of tweaking the defs such that pressing reset during power up boots to recovery without applying the 300-mt7622-generic-reset-button-ignore-env.patch

closed 2021-10-18 by dangowrt