dangowrt/owrt-ubi-installer #42: UCI reports missing library file

clutch2sft, 2021-12-05:
lua: error loading module 'uci' from file '/usr/lib/lua/uci.so':
        Error loading shared library libubox.so.20211120: No such file or directory (needed by /usr/lib/lua/uci.so)
I am not sure how to resolve this.  Is it an issue with the build release or my own issue?

dangowrt, 2021-12-11:
Did you update/install additional packages?
Please provide more information about what exactly you are running, as on the released binaries this issue does not show.
You may need to update to most recent snapshot (manually or by using `auc`) prior to installing any additional packages using `opkg`.

clutch2sft, 2021-12-11:
I don't recall what I did that caused lua to throw the error.  But I do recall that the file was not in my device when I was ruing the released build.  When I compiled on my own from the latest snapshot it works and the file exists:

/lib/libubox.so.20211120
/rom/lib/libubox.so.20211120

I think if you load the release image you will not find those files above.  That was my experience.

dangowrt, 2022-01-21:
I fail to reproduce this unfortunately. libubox version mismatch may have been caused by installing additional packages using the production system which is part of the v0.6.1 release. For that reason users are advised to always first update to the most recent release using `auc` or the *Attended Sysupgrade* feature in LuCI.

closed 2022-01-21 by dangowrt