dangowrt/owrt-ubi-installer #44: Cannot install `...-ubi-squashfs-sysupgrade.itb`

munro, 2021-12-11:
These steps all worked perfectly, and my router now always boots into the recovery firmware.

```
1. Connect any of the LAN ports of the device directly to the Ethernet port of your computer.
2. Set the IP address of your computer as `192.168.1.254` with netmask `255.255.255.0`, no gateway, no DNS.
3. Power on the device, wait about a minute for it to be ready.
4. Open a web browser, navigate to http://192.168.1.1 and wait for the wizard to come up.
5. Click *exactly* inside the radio button to confirm the terms and conditions, then abort the wizard.
6. You should then be greeted by the login screen, the stock password is "admin".
7. Navigate to __Administration__ -> __Firmware Upgrade__.
8. Upload `openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb`.
9. Wait for a minute, the OpenWrt recovery image should come up.
```

But every time I try to run the by following the steps below, I'm still in the recovery firmware after reboot!

```
9. Navigate to __System__ -> __Backup / Flash Firmware__.
10. Upload `openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb`.
12. The device will reboot, you may proceed to setup OpenWrt.
```

I've tried debugging this from what I could piece together from briefly look at the code, and I made the assumption (which could be wrong) that [install.sh](https://github.com/dangowrt/linksys-e8450-openwrt-installer/blob/main/files/installer/install.sh) is what runs when I upload the `sysupgrade.itb`.  When I uploaded it via SSH and tried to run it manually, every thing worked except for this one line:

https://github.com/dangowrt/linksys-e8450-openwrt-installer/blob/a96977bc22ad8f0dbc69f769b60ece8b88719de3/files/installer/install.sh#L153

Which gives me

```
root@OpenWrt:~# mtd write $FIP /dev/mtd1
Could not open mtd device: /dev/mtd1
Can't open device for writing!
```

It looks like there's enough space on `/proc/mtd`, so I'm not sure what the issue is!

```
root@OpenWrt:~# cat /proc/mtd 
dev:    size   erasesize  name
mtd0: 00080000 00020000 "bl2"
mtd1: 00140000 00020000 "fip"
mtd2: 00100000 00020000 "factory"
mtd3: 07d00000 00020000 "ubi"
```

```
root@OpenWrt:~# ls -l $FIP
-rw-r--r--    1 root     root        946853 Nov 20 12:46 /installer/openwrt-mediatek-mt7622-linksys_e8450-ubi-bl31-uboot.fip
```

Please send help :laughing:

dangowrt, 2021-12-11:
Please share kernel log (`dmesg`) of the system booted into recovery.

If the device always comes up in recovery mode the reasons can be:
 * production image damaged or missing
 * log of a previous crash is present in `/sys/fs/pstore`

munro, 2021-12-11:
Annnnnd I bricked it, I wish I had waited just a few minutes for your reply lol.  I was like, "oh I'm going to write to this `mtd` device" then ran `insmod mtd-rw i_want_a_brick=1`, and I guess got what I wanted lmao.

> having accidentally flashed the openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery.itb file in step 8 instead of openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb (which is a bit complicated to recover...)

I think this could be a possibility, but I'm pretty sure I did the `...-recovery-installer.itb`, I wish I hadn't bricked it so I could get those logs :sob:

dangowrt, 2021-12-11:
Shopping list:

1x FT2232H JTAG break-out
[image]

1x 1.27mm pitch 10-pin 2-row test-clamp (for JTAG pads)
[image]

1x 2.0 PH Micro JST 6-pin connector (for serial console)
[image]

You need to build [OpenOCD](https://openocd.org/) from source and apply this [patch](https://github.com/mtk-openwrt/openocd-scripts/tree/main/patch).

You can use OpenOCD scripts and in-ram jtag uboot binary from https://github.com/mtk-openwrt/openocd-scripts/tree/main/mt7622

Please don't return the bricked device. This will  encourage vendors to further lock down devices in future.

munro, 2021-12-11:
lmao I love you @dangowrt — I was thinking about buying one preemptively, and now I wish I did.

> 1x 1.27mm pitch 10-pin 2-row test-clamp (for JTAG pads)

This is cool!  I was looking for something some temporary through hole probe but nothing was coming up when searching

> Please don't return the bricked device. This will encourage vendors to further lock down devices in future.

❤️

dangowrt, 2021-12-11:
> This is cool! I was looking for something some temporary through hole probe but nothing was coming up when searching

See https://aliexpress.com/item/1005001409579446.html

dangowrt, 2021-12-11:
When you use such a test-clamp you can connect 1:1 the 20-pins to the JTAG port of the FT2232H.
Make sure to get the pinout in right order and make sure **NOT** to connect VCC 3.3V pins of the FT2232H with VCC PIN of the router (leave pin 2 disconnected) or you may fry either of the boards...

cvmiller, 2021-12-11:
> Please share kernel log (`dmesg`) of the system booted into recovery.
> 
> If the device always comes up in recovery mode the reasons can be:
> 
>     * production image damaged or missing
> 
>     * log of a previous crash is present in `/sys/fs/pstore`
> 
>     * having accidentally flashed the `openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery.itb` file in step 8 instead of `openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb` (which is a bit complicated to recover...)

So I am at this point, and I would rather not brick the router. So a couple of questions. 
1. I may have installed the recovery vs recovery-installer, now I am not sure. How does one figure out what is running (from a SSH session). 
2. I have copied off the previous log, but what is a enc.z? The `file` command says it is "OpenPGP Secret Key". Attaching here (zipped, since github won't allow a enc.z file).
3. What is my next step, since I don't want to go the JTAG route?
[dmesg.zip](https://github.com/dangowrt/linksys-e8450-openwrt-installer/files/7697898/dmesg.zip)

dangowrt, 2021-12-12:
@cvmiller This more looks like some random left-overs in RAM which got mistaken as pstore content. Please power-cycle the device and also share the current bootlog (`dmesg`) so we can judge whether the installer has already been run or not (as ubi0 won't probe if you did flash the wrong image).

cvmiller, 2021-12-12:
Power cycled, here's the dmesg.lo
[dmesg_after_power_cycle.zip](https://github.com/dangowrt/linksys-e8450-openwrt-installer/files/7698445/dmesg_after_power_cycle.zip)
g

dangowrt, 2021-12-12:
First of all, this seems to be an old build booting there:
> ```
> [    0.000000] Linux version 5.10.60 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 10.3.0 r17443-90e167abaa) 10.3.0, GNU ld (GNU Binutils) 2.36.1) #0 SMP Fri Aug 27 23:35:02 
> ```

But looks like installation succeeded as UBI is detected and even production image seems to be in place:
> ```
> [    2.700288] UBI: auto-attach mtd3
> [    2.703617] ubi0: attaching mtd3
> [    3.089278] ubi0: scanning is finished
> [    3.097224] ubi0: attached mtd3 (name "ubi", size 125 MiB)
> [    3.102716] ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
> [    3.109601] ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
> [    3.116388] ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
> [    3.123340] ubi0: good PEBs: 999, bad PEBs: 1, corrupted PEBs: 0
> [    3.129341] ubi0: user volume: 6, internal volumes: 1, max. volumes count: 128
> [    3.136557] ubi0: max/mean erase counter: 1/0, WL threshold: 4096, image sequence number: 1989374316
> [    3.145683] ubi0: available PEBs: 0, total reserved PEBs: 999, PEBs reserved for bad PEB handling: 19
> [    3.154906] ubi0: background thread "ubi_bgt0d" started, PID 797
> [    3.155945] FIT: Default configuration: "config-1" (OpenWrt linksys_e8450-ubi)
> [    3.168159] FIT:           kernel sub-image 0x00001000..0x0050e4d5 "kernel-1" (ARM64 OpenWrt Linux-5.10.82) 
> [    3.177999] FIT:          flat_dt sub-image 0x0050f000..0x00516328 "fdt-1" (ARM64 OpenWrt linksys_e8450-ubi device tree blob) 
> [    3.189393] FIT:       filesystem sub-image 0x00517000..0x0092ffff "rootfs-1" (ARM64 OpenWrt linksys_e8450-ubi rootfs) 
> [    3.200170] FIT: selecting configured loadable "rootfs-1" to be root filesystem
> [    3.207478]  ubiblock0_4: p1(rootfs-1)
> [    3.207787] block ubiblock0_4: created from ubi0:4(fit)

It also says the device has one bad block on the flash (`bad PEBs: 1`), which may have damaged the production image which is a very good and simple reason for the device to come up with the recovery image.
If you want to update it to more recent builds, you will have to re-run the installer, ie.

```
sysupgrade -F https://github.com/dangowrt/linksys-e8450-openwrt-installer/releases/download/v0.6.1/openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb
```

And once it comes up again you can then flash up-to-date sysupgrade image.

cvmiller, 2021-12-12:
Thanks for looking at my dmesg, and your analysis. Just to be abundently clear. Are you stating that this image is OK to flash?
[linksys_e8450-ubi-squashfs-sysupgrade.itb](https://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb) [from openwrt snapshots]

TIA

dangowrt, 2021-12-12:
> Are you stating that this image is OK to flash?

Yes.

cvmiller, 2021-12-12:
Great! I actually just removed `rm /sys/fs/pstore/*` and rebooted, and the snapshot I had previously loaded came up! I think I am good now. And now I understand better how your recovery image works. Thanks again!

closed 2021-12-12 by dangowrt