dangowrt/owrt-ubi-installer #46: The image file is incorrect

rasa, 2021-12-16:
On a brand new [Belkin RT3200](https://www.belkin.com/us/support-article?articleNum=208567) with the latest [1.1.00.180912](https://www.belkin.com/support/assets/belkin/firmware/FW_RT3200_1.1.00.180912_prod.img) firmware, and following [these](https://github.com/dangowrt/linksys-e8450-openwrt-installer/blob/main/README.md#installing-openwrt) instructions, and uploading release [v0.6.1](https://github.com/dangowrt/linksys-e8450-openwrt-installer/releases/tag/v0.6.1):

```
md5:    7abb0c728ab6dddb13d6d5085ce8364b *openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb
sha1:   5bd8e5920ede218253c286a262b7a269015c1643 *openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb
sha256: 3ca00588e01c6fb41084ded41929384a8a8717e830dfd2a1ba77c645f808cdb3 *openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb
```
I get a popup that says "The image file is incorrect". Here's the screenshot:

[image]

Any ideas what I'm doing wrong?

dangowrt, 2021-12-16:
Try downgrading to https://www.belkin.com/support/assets/belkin/firmware/FW_RT3200_1.0.01.101415_prod.img first and see if that helps.

rasa, 2021-12-16:
> Try downgrading to https://www.belkin.com/support/assets/belkin/firmware/FW_RT3200_1.0.01.101415_prod.img 

That fixed it. Thanks!

closed 2021-12-16 by rasa