dangowrt/owrt-ubi-installer #48: white led blinking fast, how to debrick

ashkib, 2021-12-25:
Dear Mr Golle,

Merry Christmas to you and your loved ones.

I want to thank you for your efforts to support RT3200 router , it was a remarkable work that took very valuable time and great efforts
I have a Belkin rt3200  and used your installer v0.6.1 and according to your steps every thing went smooth and got opwnwrt working.
few days later I decided to go back to vendor firmware and completed the steps 

1. Boot into on-flash recovery/initramfs instead
2. Use scp to copy the original/vendor bootchain (mtdx files) to the device's /tmp folder
3. Connect to the device via SSH and enter the following commands:
ubidetach -d 0
insmod mtd-rw i_want_a_brick=1
mtd write /tmp/mtd0 /dev/mtd0
mtd write /tmp/mtd1 /dev/mtd1
mtd write /tmp/mtd2 /dev/mtd2
mtd write /tmp/mtd3 /dev/mtd3

4. Reboot

Now I am stuck with blinking white power led ( 2 blinks per second) 
So could you please help me to recover the device ? Stock is preferable but openwrt is ok if  there is no other option

I am using windows pc and bit confused about using the factory tftp utility 
[https://www.linksys.com/us/support-article?articleNum=137928](url)
as I didn't find any detailed explanation for the steps 

I don't know how to use this software and also what are the timing needed for pressing the reset when powering on 
Dear Mr Golle,
I understand you are busy now but I am confident that someone with your kindness and passion could help me to recover this great router 

Thank you again an Merry Christmas 
Ash

dangowrt, 2021-12-25:
I'm afraid at this point you are down to wire up serial and maybe even JTAG (we'll see once serial is connected). If the TFTP client doesn't come up no matter how long and how often you are pressing the RESET button while powering on the device, that's all you can do. So you have to open the device and connect a 3.3V level TTL serial line to the JST PH 2.0 6-pin connector on the board (for pin out see https://openwrt.org/toh/linksys/e8450#serial ) and see what's happening on the console (setup at 115200 8N1).

ashkib, 2021-12-25:
Thank you very much for your kid replay, I will try to wire a serial line and update you

ashkib, 2021-12-25:
Hi Mr Golle, Please find attached info after pressing reset 

F0: 102B 0000
F6: 0000 0000
V0: 0000 0000 [0001]
00: 0000 0000
BP: 0000 0041 [0000]
G0: 0190 0000
T0: 0000 036D [000F]
Jump to BL

UNIVPLL_CON0 = 0xFE000000!!!
mt_pll_init: Set pll frequency for 25M crystal
[PMIC_WRAP]wrap_init pass,the return value=0.
[pmic_init] Preloader Start..................
[pmic_init] MT6380 CHIP Code, reg_val = 0, 1:E2  0:E3
[pmic_init] Done...................
Chip part number:7622B
MT7622 Version: 1.2.8, (iPA)
SSC OFF
mt_pll_post_init: mt_get_cpu_freq = 1350000Khz
mt_pll_post_init: mt_get_mem_freq = 1600000Khz
mt_pll_post_init: mt_get_bus_freq = 1119920Khz
[PLFM] Init I2C: OK(0)

[BLDR] Build Time: 20200522-165358
==== Dump RGU Reg ========
RGU MODE:     4D
RGU LENGTH:   FFE0
RGU STA:      0
RGU INTERVAL: FFF
RGU SWSYSRST: 8000
==== Dump RGU Reg End ====
RGU: g_rgu_satus:0
 mtk_wdt_mode_config  mode value=10, tmp:22000010
PL P ON
WDT does not trigger reboot
WDT NONRST=0x20000000
WDT IRQ_EN=0x340003
RGU mtk_wdt_init:MTK_WDT_DEBUG_CTL(590200F3)
[EMI] MDL number = 2
[EMI] DRAMC calibration start

[EMI] DRAMC calibration end

[EMI]rank size auto detect
[EMI]start_addr[0x40000000]=0x12345678, test_addr[0x48000000]= 0xEDCBA987
[EMI]start_addr[0x40000000]=0x12345678, test_addr[0x50000000]= 0xEDCBA987
[EMI]start_addr[0x40000000]=0xEDCBA987, test_addr[0x60000000]= 0xEDCBA987
[EMI]rank0 size: 0x20000000
[MEM] complex R/W mem test pass
RAM_CONSOLE wdt status (0x0)=0x0
[BBT] BMT.v2 is found at 0x3FF
[PLFM] Init Boot Device: OK(0)

[PART] blksz: 2048B
[PART] [0x0000000000000000-0x000000000007FFFF] "PRELOADER" (256 blocks)
[PART] [0x0000000000080000-0x00000000000BFFFF] "tee1" (128 blocks)
[PART] [0x00000000000C0000-0x000000000013FFFF] "lk" (256 blocks)

Device APC domain init setup:

Domain Setup (0x0)
Domain Setup (0x0)
Device APC domain after setup:
Domain Setup (0x0)
Domain Setup (0x0)
[PART] Image with part header
[PART] name : U-Boot
[PART] addr : 41E00000h mode : -1
[PART] size : 342676
[PART] magic: 58881688h

[PART] load "lk" from 0x00000000000C0200 (dev) to 0x41E00000 (mem) [SUCCESS]
[PART] load speed: 16731KB/s, 342676 bytes, 20ms
load lk (ret=0)
[PART] Image with part header
[PART] name : atf
[PART] addr : FFFFFFFFh mode : -1
[PART] size : 57936
[PART] magic: 58881688h

[PART] load "tee1" from 0x0000000000080200 (dev) to 0x43000DC0 (mem) [SUCCESS]
[PART] load speed: 14144KB/s, 57936 bytes, 4ms
load tee1 (ret=0)
[BLDR] bldr load tee part ret=0x0, addr=0x43001000
[BLDR] boot part. not found
[BLDR] part_load_images ret=0x0
[BLDR] Others, jump to ATF

[BLDR] jump to 0x41E00000
[BLDR] <0x41E00000>=0xEA00000F
[BLDR] <0x41E00004>=0xE59FF014

U-Boot 2014.04-rc1 Version: 1.0.0.1 (Jul 21 2020 - 12:07:36)

static declaration g_total_rank_size = 0x1F000000
DRAM:  496 MiB
NAND:  Recognize SNAND: ID [e5 71 ], Device Name [FM35X1GA], Page Size [2048]B Spare Size [64]B Total Size [128]MB
[mtk_snand] probe successfully!
[BBT] BMT.v2 is found at 0x3ff
128 MiB
[mtk_nand_erase_hw] mtk_nand_erase_hw @4249, ret:0x40. page:0x640
[mtk_nand_erase_hw] mtk_nand_erase_hw @4249, ret:0x40. page:0x6c0
In:    serial
Out:   serial
Err:   serial
Net:   mtk_eth
Uip activated

  *** U-Boot SPI NAND ***

     1. System Load Linux to SDRAM via TFTP.
     2. System Load Linux Kernel then write to Flash via TFTP.
     3. Boot master system code via Flash.
     4. Boot slave system code via Flash.
     5. System Load U-Boot then write to Flash via TFTP.
     6. System Load U-Boot then write to Flash via Serial.
     7. System Load ATF then write to Flash via TFTP.
     8. System Load Preloader then write to Flash via TFTP.
     U-Boot console

  Press UP/DOWN to move or Press 1~8 to choose, ENTER to select

NAND read: device 0 offset 0x500000, size 0x2000
 8192 bytes read: OK
[do_read_image_blks] Image format error,neither FIT image nor old image.
Bad Magic Number.

NAND read: device 0 offset 0x500000, size 0x0
 0 bytes read: OK
bootm flag=0, states=70f
Wrong Image Format for bootm command
ERROR: can't get kernel image!
MT7622>

dangowrt, 2021-12-25:
That looks like the old bootchain coming up properly. It's weird that you didn't manage to trigger TFTP using the button. Now that you have serial access, choose option `2. System Load Linux Kernel then write to Flash via TFTP.` in the bootmenu and serve the vendor firmware via TFTP (you will see the expected filename when trying, rename the file accordingly).

ashkib, 2021-12-25:
Thank you for replay, Do yo recommend a tftp program?
for example the Linksys one or tft6p64

dangowrt, 2021-12-25:
Any TFTP server will do. I personally use `tftp-hpa`, but that's just a (very old) habit of mine, your taste may vary.

ashkib, 2021-12-25:
Thank you very much Mr Golle
With your great support and kindness I managed to revive the device and I am grateful for all your work and help, please accept my gratitude and respect
Merry Christmas to you with love 
Ash
by the way What caused this problem and can I install your build again?

dangowrt, 2021-12-25:
Apparently the vendor boot loader didn't care much about holding the button (?)
But yes, you can of course go ahead and flash the installer another time and enjoy OpenWrt :)

closed 2021-12-25 by dangowrt

santiagobiali, 2022-09-12:
I've flashed the wrong image in a Linksys E8450 and couldn't get TFTP working by holding down the reset button while powering on.
Had to connect a serial cable and select recovery to tftp, whick worked.