dangowrt/owrt-ubi-installer #56: LuCi mtdblock download fails

Jip-Hop, 2021-12-30:
Hi,

I'm at the [backup step](https://github.com/dangowrt/linksys-e8450-openwrt-installer#assuming-the-device-is-running-stock-firmware-version-10) of the readme. Downloading each of the `mtdblock` files from LuCi is failing for me. The download doesn't complete successfully, resulting in files which are too small. I now have this:
```
-rw-rw-rw-@ 1 512K Dec 30 13:11 OpenWrt.mtd0.bin
-rw-rw-rw-@ 1 784K Dec 30 13:11 OpenWrt.mtd1.bin
-rw-rw-rw-@ 1   0B Dec 30 13:11 OpenWrt.mtd2.bin
-rw-rw-rw-@ 1 128K Dec 30 13:11 OpenWrt.mtd3.bin
-rw-r--r--  1 512K Dec 30 13:23 mtd0
-rw-r--r--  1 1.3M Dec 30 13:23 mtd1
-rw-r--r--  1 1.0M Dec 30 13:23 mtd2
-rw-r--r--  1 125M Dec 30 13:24 mtd3
```

[image]

I tried Chrome, Firefox and Safari with same results unfortunately.

How should I proceed? Is there another way to download each of the `mtdblock` files from LuCi? Or is it safe for me to continue without these files?

My device is a Belkin and it came with stock firmware 1.0.01.101415 (I didn't downgrade). I did do a factory reset before flashing `openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery.itb` from `Release v0.6.1` and I'm connecting with an ethernet cable on LAN port 1.

dangowrt, 2021-12-30:
Probably the problem is LuCI trying to read `mtdblock` instead of `mtd` devices and that may fail due to the OOB/ECC data being left behind by the vendor firmware. The best is probably to connect to the device using SSH and backup the `/dev/mtd*` (it looks like you already did this)

Jip-Hop, 2021-12-30:
Thanks for the prompt reply :) Yes, via scp I was able to transfer the mtdx files according to the readme. So if I have these files, I don't need the ones downloaded via LuCi? Or is there a difference between the two?

dangowrt, 2021-12-30:
Having what you copied using SSH is enough. Using LuCI instead is just for convenience. Maybe have a look at `dmesg` as well to get an idea of where bad blocks are located, as if they are somewhere at the beginning of the flash, reverting to stock can be more difficult (and involve using the serial console).

Jip-Hop, 2021-12-30:
Thank you 🙂 

Can you make sense of these?
```
[ 2892.732867] mt7530 mdio-bus:00 lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[ 2892.740376] br-lan: port 1(lan1) entered blocking state
[ 2892.745594] br-lan: port 1(lan1) entered forwarding state
[ 2924.212699] print_req_error: 58 callbacks suppressed
[ 2924.212707] blk_update_request: I/O error, dev mtdblock1, sector 1568 op 0x0:(READ) flags 0x84700 phys_seg 23 prio class 0
[ 2924.228872] blk_update_request: I/O error, dev mtdblock1, sector 1576 op 0x0:(READ) flags 0x84700 phys_seg 22 prio class 0
[ 2924.240050] blk_update_request: I/O error, dev mtdblock1, sector 1584 op 0x0:(READ) flags 0x84700 phys_seg 21 prio class 0
[ 2924.251226] blk_update_request: I/O error, dev mtdblock1, sector 1592 op 0x0:(READ) flags 0x84700 phys_seg 20 prio class 0
[ 2924.262405] blk_update_request: I/O error, dev mtdblock1, sector 1600 op 0x0:(READ) flags 0x84700 phys_seg 19 prio class 0
[ 2924.273583] blk_update_request: I/O error, dev mtdblock1, sector 1608 op 0x0:(READ) flags 0x84700 phys_seg 18 prio class 0
[ 2924.284760] blk_update_request: I/O error, dev mtdblock1, sector 1616 op 0x0:(READ) flags 0x84700 phys_seg 17 prio class 0
[ 2924.295934] blk_update_request: I/O error, dev mtdblock1, sector 1624 op 0x0:(READ) flags 0x84700 phys_seg 16 prio class 0
[ 2924.307108] blk_update_request: I/O error, dev mtdblock1, sector 1632 op 0x0:(READ) flags 0x84700 phys_seg 15 prio class 0
[ 2924.318282] blk_update_request: I/O error, dev mtdblock1, sector 1640 op 0x0:(READ) flags 0x84700 phys_seg 14 prio class 0
[ 2924.370591] Buffer I/O error on dev mtdblock1, logical block 196, async page read
[ 2924.378248] Buffer I/O error on dev mtdblock1, logical block 196, async page read
[ 2924.385897] Buffer I/O error on dev mtdblock1, logical block 196, async page read
[ 2928.651064] Buffer I/O error on dev mtdblock2, logical block 0, async page read
[ 2932.080688] print_req_error: 56 callbacks suppressed
[ 2932.080697] blk_update_request: I/O error, dev mtdblock3, sector 256 op 0x0:(READ) flags 0x84700 phys_seg 27 prio class 0
[ 2932.096791] blk_update_request: I/O error, dev mtdblock3, sector 264 op 0x0:(READ) flags 0x84700 phys_seg 26 prio class 0
[ 2932.107893] blk_update_request: I/O error, dev mtdblock3, sector 272 op 0x0:(READ) flags 0x84700 phys_seg 25 prio class 0
[ 2932.118982] blk_update_request: I/O error, dev mtdblock3, sector 280 op 0x0:(READ) flags 0x84700 phys_seg 24 prio class 0
[ 2932.130072] blk_update_request: I/O error, dev mtdblock3, sector 288 op 0x0:(READ) flags 0x84700 phys_seg 23 prio class 0
[ 2932.141198] blk_update_request: I/O error, dev mtdblock3, sector 296 op 0x0:(READ) flags 0x84700 phys_seg 22 prio class 0
[ 2932.152286] blk_update_request: I/O error, dev mtdblock3, sector 304 op 0x0:(READ) flags 0x84700 phys_seg 21 prio class 0
[ 2932.163374] blk_update_request: I/O error, dev mtdblock3, sector 312 op 0x0:(READ) flags 0x84700 phys_seg 20 prio class 0
[ 2932.174499] blk_update_request: I/O error, dev mtdblock3, sector 320 op 0x0:(READ) flags 0x84700 phys_seg 19 prio class 0
[ 2932.185586] blk_update_request: I/O error, dev mtdblock3, sector 328 op 0x0:(READ) flags 0x84700 phys_seg 18 prio class 0
[ 2932.233341] Buffer I/O error on dev mtdblock3, logical block 32, async page read
[ 2932.240901] Buffer I/O error on dev mtdblock3, logical block 32, async page read
[ 3505.290716] br-lan: port 1(lan1) entered disabled state
[ 3505.296237] mt7530 mdio-bus:00 lan1: Link is Down
[ 3738.252878] mt7530 mdio-bus:00 lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[ 3738.260403] br-lan: port 1(lan1) entered blocking state
[ 3738.265621] br-lan: port 1(lan1) entered forwarding state
```
Anything to worry about?

Jip-Hop, 2022-01-03:
Thanks @dangowrt, OpenWRT runs great on my Belkin RT3200! This repo was a huge help! 😄 As you mentioned the backup I made with SSH is sufficient and I'm not planning to go back to stock anyway so this issue is solved for me.

closed 2022-01-03 by Jip-Hop