dangowrt/owrt-ubi-installer #57: conversion failed

jasonintw2019, 2022-01-07:
I'm converting from stock firmware using your binary 0.61. I've already downgraded to 1.0.01 version of stock firmware before conversion. I got "192.168.1.1 says The image file is incorrect."

dangowrt, 2022-01-07:
But the downgrade did work? Ie. the firmware version shown is now 1.0.01?
Regarding the installer. you are using the ready binary provided in the releases section of this project I suppose...?

jasonintw2019, 2022-01-07:
Yes, it's now 1.0.01

jasonintw2019, 2022-01-07:
May be I should give more details. After downgrading, I uploaded the recovery-installer file, and the router seemed to be bricked (didn't come to live with power cycling). I then applied a hardware reset (strangely I wrongly held the wps button and power on and still fine) and to bring it to live again. After this, I tried to flash the file again, "image file is incorrect", then I flashed the recover (only) file, but also "image file is incorrect".

dangowrt, 2022-01-07:
That's very odd. Esp. the part that you succeeded to upload the recovery-installer file and then the router would not start it. I hope they didn't lock update the bootloader, but 1.0.01 is also not signed in any special way from what I can tell, so it's probably somethign more simple.

The best would be if you can connect a USB-to-serial converter to the internal (3.3V level) serial console of the device, so we can see what's actually going on...

jasonintw2019, 2022-01-07:
do I need to open the case for that serial console?

dangowrt, 2022-01-08:
Yes, you do need to remove the 4 screws hidden underneath the sticker on the bottom and then slide out the board to find the JST PH 2.0 6-pin connector which is the serial port. See also the [OpenWrt wiki](https://openwrt.org/toh/linksys/e8450) for details.

jasonintw2019, 2022-01-08:
Yesterday, I upgraded the router to new stock firmware again, just to try whether my router can flash image, it succeeded. But today, I want try once more to downgrade the router to 1.0, I couldn't do it anymore saying the image is incorrect again. Does it mean I can no longer flash this router to openwrt? If so, I need to buy another one for openwrt and use this router elsewhere.

dangowrt, 2022-01-08:
It more sounds like something is wrong with this specific device. However, as I said, only with the serial console connected we can shed light onto this.

jasonintw2019, 2022-01-08:
Thank you for your help. It really takes time and courage to open it up, but I think I will do it later, and ask for your help again. In fact I have a tp-link C7 running openwrt for years, and still working very well. Should I close this thread by now?

dangowrt, 2022-01-08:
I'm curiously awaiting the serial bootlog to be posted here ;)

jasonintw2019, 2022-01-17:
Hi dangowrt,
I'm writing to let you know that I do the conversion today again, and everything's going smoothly. Probably sth I've done wrong last time. I have the new toy to play now. 
Thank you for all the help.

closed 2022-01-17 by jasonintw2019