dangowrt/owrt-ubi-installer #59: Route does not reboot back to vendor firmware after backup

jmoch, 2022-01-11:
Hello!
Thanks for this very cool contribution.  I was attempting to to the first steps in:
Device flash complete backup procedure
Everything went smoothly until step 5.  In my instance the powecycle does not kick the router back into the Belkin vendor firmware.
The reboot enters into OpenWrt SNAPSHOT r18125-b764cb9e5b / LuCl Master git-21.322.38385-6507b1f
no password set and system running in recovery (initramfs) mode.

Any idea how to continue?  The guide suggests that the powercycle should return the router to the Belkin firmware.

Any help and suggestions would be greatly appreciated!  I can confirm that the original firmware was 1.0

Thanks,
Jason

dangowrt, 2022-01-11:
Try `reboot` when connected via SSH, and generally just keep trying, it's going to come back to vendor firmware after some reboots.

jmoch, 2022-01-11:
Hi David,
Great, I was going that route but gave up after a couple or cycles.  I'll try it now and report back soon.
Hope you have a great day,
Jason

jmoch, 2022-01-11:
Hi Daniel, 
Reporting back.  So last evening I powered everything down.  Brought it back up this morning and cannot connect to the router via browser or ssh.  Error reads no route to host.  I've power cycled an additional 5 times and this result still holds.  I thought it might be because of the ip of my computer but I can confirm that same result with "Set the IP address of your computer as 192.168.1.254 with netmask 255.255.255.0, no gateway, no DNS"

Thanks for your quick responses and assistance!
Jason

jmoch, 2022-01-12:
Hi Daniel,
Scratch the above comments not being able to connect.  I traced it to cable issue.  Multiple reboots in fact have resulted in it rebooting into the oem firmware so I'm moving to the next steps.

Thanks again!
Jason

closed 2022-01-12 by jmoch