dangowrt/owrt-ubi-installer #61: add AES supports

ericvwv, 2022-01-11:
As we know the A53 core supports AES, Normally, it should be about 550K, but I ran the AES test and the score was only around 55K.   It is sure that AES is work abnormally. I'm sure I've found a solution, but I don't know where to add these two options in build_installer.sh 

CONFIG_PACKAGE_libmbedtls=y
CONFIG_LIBMBEDTLS_HAVE_ARMV8CE_AES=y

AES test score==================================================
root@E8450:~# openssl speed -evp aes-128-gcm
type             16 bytes     64 bytes    256 bytes   1024 bytes   8192 bytes  16384 bytes
aes-128-gcm      37646.77k    49596.20k    54220.89k    56560.84k    55971.36k    55974.10k

mrkiko, 2022-01-12:
This issue doesn't seem related to this project.
Instead, it seems a more general OpenWRt configuration issue.

ericvwv, 2022-01-12:
U are right.  Please close this issue.
thanks for your work.

mrkiko, 2022-01-12:
Oh - it's all Daniel's :) :)

closed 2022-01-12 by mrkiko