dangowrt/owrt-ubi-installer #62: After flashing OpenWRT Snapshot doesn't response no more.

xromansx, 2022-01-12:
Hi,
 When I try to flash Openwrt Snapsot it doen't connect to network and doesn't ping no more I can see wifi but can't connect to it.
 With FW from here everything is ok but packager in Software are not compatible with current kernel. 
 Is there any posibility fo flash Openwrt snaphsot ? I'm trying to flash it from openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb file openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb

Thanks.

dangowrt, 2022-01-12:
Using snapshots from downloads.openwrt.org should work ok, but be aware that those are development snapshots and (unlike the `sysupgrade` file provided here) they come without a web interface.
Hence, to keep the selection of installed packages, it's better to use `auc` or `luci-app-attendedsysupgrade` to move to a more recent build (which will then allow to install all packages you like).

xromansx, 2022-01-12:
Thanks for reply but I can't even open ssh to it, it doesn't respond to ping after flash of openwrt snapshot. If luci-app-attendedsysupgrade will take snapshoot from here, I installed 0.6.1 version I think it's the last one, but when trying to install for example openvpn it give the kernel missmatch error.

dangowrt, 2022-01-12:
At this point, first of all, what you can do is holding the RESET button while power on to trigger booting into recovery mode. That will allow to then flash again the `sysupgrade` file you find here as part of v0.6.1 release.
Using `auc` or `luci-app-attendedsysupgrade` makes use of the sysupgrade.openwrt.org online service to request the correct updated image *incl. the same selection of packages*. Try doing that, as then you will be able to install any package you like.

xromansx, 2022-01-12:
Ok, thank you. Could restore it and installed openwrt snaphot and it's working fine now . Thanks.

closed 2022-01-12 by xromansx