dangowrt/owrt-ubi-installer #63: Belkin RT3200 bricked

BillBarry, 2022-01-13:
Hello,  Somehow while trying to install openwrt I have managed to get the router so it will not boot. I have managed to connect a serial console and I am getting the appended output. Thanks much for any help.

```
F0: 102B 0000
F6: 0000 0000
V0: 0000 0000 [0001]
00: 0000 0000
BP: 0000 0041 [0000]
G0: 0190 0000
T0: 0000 0270 [000F]
Jump to BL

UNIVPLL_CON0 = 0xFE000000!!!
mt_pll_init: Set pll frequency for 25M crystal
[PMIC_WRAP]wrap_init pass,the return value=0.
[pmic_init] Preloader Start..................
[pmic_init] MT6380 CHIP Code, reg_val = 0, 1:E2  0:E3
[pmic_init] Done...................
Chip part number:7622B
MT7622 Version: 1.2.8, (iPA)
SSC OFF
mt_pll_post_init: mt_get_cpu_freq = 1350000Khz
mt_pll_post_init: mt_get_mem_freq = 1600000Khz
mt_pll_post_init: mt_get_bus_freq = 1119920Khz
[PLFM] Init I2C: OK(0)

[BLDR] Build Time: 20200522-165358
==== Dump RGU Reg ========
RGU MODE:     4D
RGU LENGTH:   FFE0
RGU STA:      0
RGU INTERVAL: FFF
RGU SWSYSRST: 8000
==== Dump RGU Reg End ====
RGU: g_rgu_satus:0
 mtk_wdt_mode_config  mode value=10, tmp:22000010
PL P ON
WDT does not trigger reboot
WDT NONRST=0x20000000
WDT IRQ_EN=0x340003
RGU mtk_wdt_init:MTK_WDT_DEBUG_CTL(590200F3)
[EMI] MDL number = 2
[EMI] DRAMC calibration start

[EMI] DRAMC calibration end

[EMI]rank size auto detect
[EMI]start_addr[0x40000000]=0x12345678, test_addr[0x48000000]= 0xEDCBA987
[EMI]start_addr[0x40000000]=0x12345678, test_addr[0x50000000]= 0xEDCBA987
[EMI]start_addr[0x40000000]=0xEDCBA987, test_addr[0x60000000]= 0xEDCBA987
[EMI]rank0 size: 0x20000000
[MEM] complex R/W mem test pass
RAM_CONSOLE wdt status (0x0)=0x0
[BBT] BMT.v2 is found at 0x3FF
[PLFM] Init Boot Device: OK(0)

[PART] blksz: 2048B
[PART] [0x0000000000000000-0x000000000007FFFF] "PRELOADER" (256 blocks)
[PART] [0x0000000000080000-0x00000000000BFFFF] "tee1" (128 blocks)
[PART] [0x00000000000C0000-0x000000000013FFFF] "lk" (256 blocks)

Device APC domain init setup:

Domain Setup (0x0)
Domain Setup (0x0)
Device APC domain after setup:
Domain Setup (0x0)
Domain Setup (0x0)
[PART] Image with part header
[PART] name : U-Boot
[PART] addr : 41E00000h mode : -1
[PART] size : 342676
[PART] magic: 58881688h

[PART] load "lk" from 0x00000000000C0200 (dev) to 0x41E00000 (mem) [SUCCESS]
[PART] load speed: 15934KB/s, 342676 bytes, 21ms
load lk (ret=0)

NAND read: device 0 offset 0x2300000, size 0x2000
 8192 bytes read: OK
[do_read_image_blks] Image format error,neither FIT image nor old image.
Bad Magic Number.

NAND read: device 0 offset 0x2300000, size 0x0
 0 bytes read: OK
bootm flag=0, states=70f
Wrong Image Format for bootm command
ERROR: can't get kernel image!
MT7622>
MT7622>
MT7622>
MT7622> U-Boot 2014.04-rc1 Version: 3>>>> U-Boot 2014.04-rc1 Version: 3>>>> U-Boot 2014.04-rc1 Version: 3>>>> U-Boot 2014.04-rc1 Version: 3>>>> U-
```

dangowrt, 2022-01-13:
As it is `U-Boot 2014.04-rc1` coming up there, this is still the old vendor bootchain and apparently the installer has not actually been run, but an UBI-firmware was flashed and that can't work well, resulting in exactly the situation I can see here.

At this point you have to use TFTP to load something onto the router. So you need a TFTP server operating at IP address 192.168.1.254/24 serving [openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb](https://github.com/dangowrt/linksys-e8450-openwrt-installer/releases/download/v0.6.1/openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb) connected to port LAN1 (other LAN ports may work as well, I never tested).

Once you got the TFTP server setup, I would recommend to use `bootmenu` (which you can type at `MT7622>` prompt) and select option `1. System Load Linux to SDRAM via TFTP.`. You can then provide the name of the file the router will download into RAM and boot from. You can then watch the installer doing it's job in the serial console and another reboot will happen, but then with `U-Boot 2021.10` and the system will start into OpenWrt recovery firmware. From there you flash [openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb](https://github.com/dangowrt/linksys-e8450-openwrt-installer/releases/download/v0.6.1/openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb) and then everything should be fine.

Let me know if you need further help with this.

BillBarry, 2022-01-13:
I have the tftp server running but I don't know hot to stop it at the MT7622> prompt. The system runs right past that and shows the `U-Boot 2014.04-rc1 Version: 3>>>>` prompt several times then seems to not respond.

dangowrt, 2022-01-13:
Try repeatedly feeding it [CTRL]+ [C] to abort. Which terminal software are you using?

BillBarry, 2022-01-13:
I did not see any response to  [CTRL]+ [C]  or [ENTER]. I am using minicom with 
``
minicom -c on -b 115200 -D /dev/ttyUSB0
``

dangowrt, 2022-01-13:
I guess there is only a very small time window where [CTRL]+[C] may work to abort loading the corrupted image from NAND.

Also make sure flow-control (both hardware and software) is disabled and also try just typing `bootmenu` followed by [ENTER] even if the characters are not echoed back.

BillBarry, 2022-01-14:
Yahoo, it works!  Your suggestion about the cabling sent me off in the right direction. Long story short RX and TX were connected in my cable so it was acting as a loopback. I am surprised it worked as much as it did. After fixing that I was able to load the installer via tftp as you instructed at which time I discovered the original mistake which brought me here. I had installed openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery.itb  instead of  openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb. After downloading and flashing the correct one via tftp I was able to flash openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb  and now everything is working great.  Thanks a whole bunch!

closed 2022-01-14 by BillBarry