dangowrt/owrt-ubi-installer #64: Random crashes on r18125-b764cb9e5b (cross-post)

SebTM, 2022-01-17:
Hey,

I got the recommendation after I summarized all information I have to also report here as you are some kind of target expert - so I hope you are OK with this cross-post ✌🏻 (Forum: https://forum.openwrt.org/t/belkin-rt3200-aka-linksys-e8450-random-crashes/117547)

I'm facing random hangs/crashes of the device, I can reproduce manually when restarting br-lan device from luci. Changing settings of the device does not cause any problems:

```
Mon Jan 17 02:21:35 2022 kern.warn kernel: [20523.545686] print_req_error: 34353 callbacks suppressed
Mon Jan 17 02:21:35 2022 kern.err kernel: [20523.545693] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:35 2022 kern.err kernel: [20523.563301] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:35 2022 kern.err kernel: [20523.575053] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:35 2022 kern.err kernel: [20523.585826] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:35 2022 kern.err kernel: [20523.596576] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:35 2022 kern.err kernel: [20523.608276] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:35 2022 kern.err kernel: [20523.619036] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:35 2022 kern.err kernel: [20523.630770] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:35 2022 kern.err kernel: [20523.641523] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:35 2022 kern.err kernel: [20523.653218] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:35 2022 kern.warn kernel: [20523.665648] buffer_io_error: 34347 callbacks suppressed
Mon Jan 17 02:21:35 2022 kern.err kernel: [20523.666240] Buffer I/O error on dev mtdblock2, logical block 32, async page read
Mon Jan 17 02:21:35 2022 kern.err kernel: [20523.679089] Buffer I/O error on dev mtdblock2, logical block 32, async page read
Mon Jan 17 02:21:35 2022 kern.err kernel: [20523.687680] Buffer I/O error on dev mtdblock2, logical block 32, async page read
Mon Jan 17 02:21:35 2022 kern.err kernel: [20523.695278] Buffer I/O error on dev mtdblock2, logical block 32, async page read
Mon Jan 17 02:21:35 2022 kern.err kernel: [20523.703946] Buffer I/O error on dev mtdblock2, logical block 32, async page read
Mon Jan 17 02:21:35 2022 kern.err kernel: [20523.711575] Buffer I/O error on dev mtdblock2, logical block 32, async page read
Mon Jan 17 02:21:35 2022 kern.err kernel: [20523.720229] Buffer I/O error on dev mtdblock2, logical block 32, async page read
Mon Jan 17 02:21:35 2022 kern.err kernel: [20523.727848] Buffer I/O error on dev mtdblock2, logical block 32, async page read
Mon Jan 17 02:21:35 2022 kern.err kernel: [20523.735385] Buffer I/O error on dev mtdblock2, logical block 32, async page read
Mon Jan 17 02:21:35 2022 kern.err kernel: [20523.744056] Buffer I/O error on dev mtdblock2, logical block 32, async page read
Mon Jan 17 02:21:40 2022 kern.warn kernel: [20528.555613] print_req_error: 34401 callbacks suppressed
Mon Jan 17 02:21:40 2022 kern.err kernel: [20528.555621] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:40 2022 kern.err kernel: [20528.572530] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:40 2022 kern.err kernel: [20528.583283] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:40 2022 kern.err kernel: [20528.595062] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:40 2022 kern.err kernel: [20528.605814] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:40 2022 kern.err kernel: [20528.616988] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:40 2022 kern.err kernel: [20528.627737] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:40 2022 kern.err kernel: [20528.639519] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:40 2022 kern.err kernel: [20528.650263] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:40 2022 kern.err kernel: [20528.662034] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:40 2022 kern.warn kernel: [20528.675710] buffer_io_error: 34410 callbacks suppressed
Mon Jan 17 02:21:40 2022 kern.err kernel: [20528.675716] Buffer I/O error on dev mtdblock2, logical block 32, async page read
Mon Jan 17 02:21:40 2022 kern.err kernel: [20528.689535] Buffer I/O error on dev mtdblock2, logical block 32, async page read
Mon Jan 17 02:21:40 2022 kern.err kernel: [20528.697154] Buffer I/O error on dev mtdblock2, logical block 32, async page read
Mon Jan 17 02:21:40 2022 kern.err kernel: [20528.704693] Buffer I/O error on dev mtdblock2, logical block 32, async page read
Mon Jan 17 02:21:40 2022 kern.err kernel: [20528.713321] Buffer I/O error on dev mtdblock2, logical block 32, async page read
Mon Jan 17 02:21:40 2022 kern.err kernel: [20528.720942] Buffer I/O error on dev mtdblock2, logical block 32, async page read
Mon Jan 17 02:21:40 2022 kern.err kernel: [20528.729537] Buffer I/O error on dev mtdblock2, logical block 32, async page read
Mon Jan 17 02:21:40 2022 kern.err kernel: [20528.737155] Buffer I/O error on dev mtdblock2, logical block 32, async page read
Mon Jan 17 02:21:40 2022 kern.err kernel: [20528.744690] Buffer I/O error on dev mtdblock2, logical block 32, async page read
Mon Jan 17 02:21:40 2022 kern.err kernel: [20528.753279] Buffer I/O error on dev mtdblock2, logical block 32, async page read
Mon Jan 17 02:21:45 2022 kern.warn kernel: [20533.565684] print_req_error: 34421 callbacks suppressed
Mon Jan 17 02:21:45 2022 kern.err kernel: [20533.565692] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:45 2022 kern.err kernel: [20533.582670] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:45 2022 kern.err kernel: [20533.593415] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:45 2022 kern.err kernel: [20533.605165] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:45 2022 kern.err kernel: [20533.615899] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:45 2022 kern.err kernel: [20533.627631] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:45 2022 kern.err kernel: [20533.638372] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:45 2022 kern.err kernel: [20533.650111] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:45 2022 kern.err kernel: [20533.660844] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:45 2022 kern.err kernel: [20533.672607] blk_update_request: I/O error, dev mtdblock2, sector 256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Jan 17 02:21:45 2022 kern.warn kernel: [20533.685572] buffer_io_error: 34416 callbacks suppressed
[... cut ...]
root@router:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00080000 00020000 "bl2"
mtd1: 00140000 00020000 "fip"
mtd2: 00100000 00020000 "factory"
mtd3: 07d00000 00020000 "ubi"
```

Do you have any idea? Would it be recommended to use "Attended Sysupgrade" to get latest snapshot and test if it still occurs or can it be that I did something wrong after using the installer e.g. sysupgrades/resets that cause it? Is running the installer again another option/safe?

Thanks for the awesome work! 🥳

SebTM, 2022-01-20:
Hey, thanks for the answer - I checked and used v0.6.1, but I did some resets to factory settings / flashed "openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb" again after upgrading to snapshot one time.

If it helps I could rerun the installer, I now also tried upgrading to the latest snapshot and at least I cant reproduce the issue anymore when restarting the interface manual via Luci - but I can downgrade again to debug if it make sense :)

dangowrt, 2022-01-21:
If it works fine now I don't see a need to dig there other than my personal curiosity :)

SebTM, 2022-01-28:
Made different tries and can't reproduce it anymore - will have an eye on it but seems done for now - thanks 👍🏻

closed 2022-01-28 by SebTM

justinkb, 2022-01-29:
happening here on 0.6.2 pre-release

dangowrt, 2022-01-29:
@justinkb Please post the full log and also share the history of the device (ie. which version of the installer you used initially, did you ever re-run it if the initial version was prior to v0.5.3)

justinkb, 2022-01-29:
The log was essentially the same as what was posted above.

History is device (Belkin) arrived a few days ago. It was on firmware 1.0
out of the box. Did the stock mtd backup via the recovery from 0.6.1, then
installed the sysupgrade from 0.6.1. Since then I've had all kinds of
recent builds installed (troubleshooting the .94 kernel update e.g.)

dangowrt, 2022-01-29:
Just so that I understand correctly: The device was converted to UBI by the v0.6.1 installer, then worked fine, you tried some different snapshots and now you are seeing the above lines in your logs. Right so far?
Does it occur in a reproducible way?
Does the device keep working anyway?
Because to me it's weird that anything is even reading those (unused) areas in mtd2 (`factory`), having read-errors there is kinda expected as it would still be the broken OOB/ECC data from stock firmware... Shouldn't do any harm, but maybe I can find a way to silence that or just have future versions of the installer also fix OOB/ECC data in unused areas, if that's really needed.

justinkb, 2022-01-29:
I am not 100% sure the dmesg mtdblock errors are connected to the failure
to connect to luci, let me clarify that. Might just have been pure
coincidence, I was using luci at the time to mess with the wifi settings.
In my experience that can occasionally cause breakages (had that happen
with many routers over the years, stock or custom firmware). It only
happened once. Just posted this so you know it is not "gone" in the latest
0.6.2 snapshot. I'm now on a newer snapshot, hasn't happened yet since the
last sysupgrade

dangowrt, 2022-02-12:
@dietcoke73 Please decide for one place to discuss this, either here or in the forums -- as you can see I'm usually aware what's happening on both within some minutes, so please avoid the noise of cross-posting.

darmach, 2022-10-27:
Hi @dangowrt just popping up to let you know I just encountered this on https://github.com/dangowrt/owrt-ubi-installer/releases/tag/v1.0.1 after tinkering with bridges (removed some ports off br-lan)