dangowrt/owrt-ubi-installer #65: Coming from Non-UBI OpenWRT, cannot boot into factory firmware

slogen, 2022-01-31:
I have previously been on Non-UBI. That works just fine - but I would like to completely convert.

I am following the instructions, and have tried to boot into factory firmware, by: 
- power-cycles  (30-50s unplug)
- reboot (ssh into openwrt, reboot)
- reset-hold (holding reset for 10-20s as guided by linksys instructions)
- reset-cycle (poweroff 30s, boot with reset down, never gets amber light)

But all I achieve is resetting OperWRT to factory defaults, not the linksys E8450 firmware.

What am I doing wrong? I have *of* *course* not tried flashing the original FW or any of the .itb files directly from OpenWRT :)

[image]

dangowrt, 2022-01-31:
If you are using non-UBI OpenWrt and want to go back to stock, you need to flash the [stock firmware](https://downloads.linksys.com/support/assets/firmware/FW_E8450_1.0.01.101415_prod.img) using `sysupgrade` or upload it to *System* -> *Backup / Flash Firmware* -> *Flash new firmware image*.

If you want to convert the device to UBI, you can also directly upload [the installer](https://github.com/dangowrt/linksys-e8450-openwrt-installer/releases/download/v0.6.1/openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb) in the same way.

In both cases, you need to select *Force Upgrade* and ignore the red warnings.

slogen, 2022-01-31:
Thanks for *awesomely* quick reply! I am trying to convert non-UBI E8450 to UBI, which I have now successfully done due to your assistance!

I got pretty scared reading through the instructions for backup.

> Do not attempt to flash anything (that includes the stock firmware, openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb etc.) from the running initramfs system -- it will fail to reboot, possibly requiring serial console access. Instead, powercycle the device to reboot into the original non-ubi firmware, and then flash the installer version.

Maybe add that flashing the initramfs *can* be done *from* *OpenWRT* in the docs? and instruction to select the scary red -F option on this screen?

[image]

closed 2022-01-31 by slogen