dangowrt/owrt-ubi-installer #67: Booting installer.itb - bad crc, boots into recovery

BashfulBladder, 2022-02-07:
I have a Belkin RT3200, that was running 1.0.01.101415. I flashed the 0.6.2 recovery.itb image. I saved the mtd[0-3] backups. I forgot to press the reset button, but I eventually made it back to stock firmware. I flashed the installer.itb. It boots, is unreachable for 5 or 10 secs & reboots into recovery.itb. I manually copy mtd[0-3], IWantABrick & have stock firmware. I flashed 0.6.0 with the same result. I can't figure out how to get the installer.itb to boot.

So I connected my usb-to-serial RX-TX-GND and I get gobbldy-gook. I swap Rx<>Tx to --T-RG and I have meaningful output.

Below are console outputs of 3 boot attempts:
1. Unplugged, boot into installer with the above described light pattern. I did not press the reset button despite the 'button pushed' line.
2. Booting into recovery while pressing reset
3. Booting into recovery

Any ideas of how to get a working OpenWrt image to stick?

----------------------------------
Boot into installer 
```
F0: 102B 0000
F6: 0000 0000
V0: 0000 0000 [0001]
00: 0000 0000
BP: 0400 0041 [0000]
G0: 1190 0000
T0: 0000 02AF [000F]
Jump to BL

NOTICE:  BL2: v2.4(release):OpenWrt v2021-05-08-d2c75b21-1 (mt7622-snand-1ddr)
NOTICE:  BL2: Built : 22:51:20, Nov 19 2021
NOTICE:  SPI-NAND: FM35Q1GA (128MB)
NOTICE:  BL2: Booting BL31
NOTICE:  BL31: v2.4(release):OpenWrt v2021-05-08-d2c75b21-1 (mt7622-snand-1ddr)
NOTICE:  BL31: Built : 22:51:20, Nov 19 2021
serial_mtk serial@11002000: pinctrl_select_state_full: uclass_get_device_by_phandle_id: err=-19

U-Boot 2021.10 (Nov 19 2021 - 22:51:20 +0000)

CPU:   MediaTek MT7622
Model: mt7622-linksys-e8450-ubi
DRAM:  512 MiB
WDT:   Started with servicing (60s timeout)
MMC:   
Loading Environment from UBI... SPI-NAND: FM35Q1GA (128MB)
Read 126976 bytes from volume ubootenv to 000000005f7bf3c0
Read 126976 bytes from volume ubootenv2 to 000000005f7de400
OK
In:    serial@11002000
Out:   serial@11002000
Err:   serial@11002000
reset button found
button pushed, resetting environment
*** Warning - bad CRC, using default environment

Net:   eth0: ethernet@1b100000
Reading 131072 byte(s) (64 page(s)) at offset 0x00220000
## Error: Can't overwrite "ethaddr"
## Error inserting "ethaddr" variable, errno=1
Saving Environment to UBI... Writing to UBI... done
OK
Saving Environment to UBI... Writing to UBI... done
OK
on
Using ethernet@1b100000 device
TFTP from server 192.168.1.254; our IP address is 192.168.1.1
Filename 'openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery.itb'.
Load address: 0x48000000
Loading: *
F0: 102B 0000
F6: 0000 0000
V0: 0000 0000 [0001]
00: 0000 0000
BP: 0400 0041 [0000]
G0: 1190 0000
T0: 0000 02AF [000F]
[... 300 lines cut ...]
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x4888b140
     Data Size:    29481 Bytes = 28.8 KiB
     Architecture: AArch64
     Hash algo:    crc32
     Hash value:   d090dd70
     Hash algo:    sha1
     Hash value:   5a6a6edd97ebfd758f4e57902d4662cb015eaf39
   Verifying Hash Integrity ... crc32+ sha1+ OK
   Booting using the fdt blob at 0x4888b140
   Uncompressing Kernel Image
   Loading Ramdisk to 5f329000, end 5f7fa784 ... OK
   Loading Device Tree to 000000005f31e000, end 000000005f328328 ... OK
Add 'reserved-memory' node failed: FDT_ERR_EXISTS

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.10.80 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r18125-b764cb9e5b) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Fri Nov 19 22:51:20 2021
[    0.000000] Machine model: Linksys E8450 (UBI)
[    0.000000] earlycon: uart8250 at MMIO32 0x0000000011002000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
SNIP
```

dangowrt, 2022-02-07:
Did you also do the final step and flash the `sysupgrade.itb` file once the device comes up in recovery after flashing the `initramfs-installer.itb` file? Because it looks like there just hasn't ever been a production firmware loaded onto the device.

BashfulBladder, 2022-02-07:
No, I didn't flash sysupgrade.itb - in the mtd[0-3] backup process it says **Do not attempt to flash anything** (that includes the stock firmware, openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb etc.) from the running initramfs system -- it will fail to reboot, possibly requiring serial console access.

Maybe the various images could have ammended banners or something.

dangowrt, 2022-02-07:
> in the mtd[0-3] backup process it says Do not attempt to flash anything (that includes the stock firmware, openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer.itb etc.) from the running initramfs system -- it will fail to reboot, possibly requiring serial console access.

And that's exactly true. Do **not** do that, because at this point the device is running the original bootchain and will fail to do anything with whatever you flash using that image you are using **only to do the full backup**.

Once the device comes up again with the vendor firmware you need to load the **`initramfs-installer.irb`** image and after that you still **do** need to flash the **`sysupgrade.itb`** image as stated [here](https://github.com/dangowrt/linksys-e8450-openwrt-installer/#installing-openwrt) (see step 11).

BashfulBladder, 2022-02-08:
Doing as you instructed (restoring Belkin firmware, flashing installer.itb & rebooting into initramfs OpenWrt & flashing sysupgrade.itb) did in fact get me to a working OpenWrt.

From my perspective, the intent of the 'don't flash anything while in initramfs', when the installer comes up with a warning at top of initramfs, could be better explained with the bootchain issue you mentioned above.

Closing the issue & making a donation @ https://openwrt.org/donate

closed 2022-02-08 by BashfulBladder