dangowrt/owrt-ubi-installer #71: Was working, now it's not

keirak420, 2022-02-23:
I had everything installed and was using one of the snapshots. They released new drivers for the hardware and so I did a snapshot upgrade, now I get no response from the router. I have power cycled with the reset button but all I get is a solid yellow internet light.

In general, I am brand new to OpenWRT, I bought this router specifically to be used with OpenWRT. Up until now, I have used DD-WRT and I have never "bricked" a router, so I have never had to hook up a serial or JTAG connection. (yes I am a newbie!)

I purchased this device and hooked it up today: https://www.amazon.com/dp/B07WX2DSVB I made sure it was set to the 3.3v and set Putty to 115200 8n1. I then tested it with the tx/rx shorted and had everything working. Then I connected it to the router and turned it on but, I do not get any rx/tx at all. Now at this point, I do not know if I am doing something wrong or not.

At this point, I must differ to the superior collective knowledge of you guys. What do I need to do at this point? I have never used a JTAG, so if I need to go that route can someone please give me links to what I need to purchase and what I need to do as I am feeling overwhelmed and frustrated.

Thank you!

dangowrt, 2022-02-23:
I've seen your previous ticket #68 and was already wondering if things had worked out for you.
First of all, don't worry, you are only minutes away from a working system.
Your mistake was most likely to force-flash the image for `linksys,e8450` rather than `linksys,e8450-ubi`, and that will destroy most of the content of the flash (but not the bootloader -- and even if you somehow managed to wipe the bootloader the MT7622 would then report `System halted!` on the serial console). This is a common mistake and happened to a lot of people before.

So first of all, the serial adapter looks like the right thing and the only reason I can see for that not to show at least `System halted!` is that RX and TX are swapped -- you need to connect the TX on the router side to RX on the adapter side and vice versa.

If the yellow light goes on after you power on the device, that means good news: The bootloader is alive and wants to load a recovery image via TFTP. So you need to setup a TFTP server on address `192.168.1.254` (netmask `255.255.255.0`) and have that serve the file [`openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery.itb`](https://github.com/dangowrt/linksys-e8450-openwrt-installer/releases/download/v0.6.1/openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery.itb).
I'm usually using `tftpd-hpa` (ie. start `tftpd -L -v -s .` in the folder where that file is located) for that, works on all UNIX systems where tftpd-hpa is available (Linux, *BSD incl. Mac OS X, ...)
If you are on Windows, you can use for example [SolarWinds TFTP Server](https://www.solarwinds.com/free-tools/free-tftp-server). Make sure you have all firewall things disabled on the system you are using to host the TFTP server.

Once the router has loaded the recovery image (which is also stored to flash again at this point, so in future you can use the RESET button to load it without having to provide it via TFTP), you need to flash the [production image](https://github.com/dangowrt/linksys-e8450-openwrt-installer/releases/download/v0.6.1/openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb) to it, just like you did when initially running the installer.

Now, as you are back on square one, you can update to a more recent snapshot, either using `auc` or by using what you can find on [downloads.openwrt.org](https://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb). Make sure you grab the image file with `-ubi-` in it's name. There shouldn't be any warnings what-so-ever.

Let me know in case you need more assistance.

keirak420, 2022-02-24:
OMG, I feel so dumb. You were correct, I had the leads reversed. I completely forgot the tx from one is the rx on the other. As you will see further down the list, I did load the UBI version. I was able to boot it and get the following:
_______________________________________________________________________
NOTICE:  BL2: v2.4(release):OpenWrt v2021-05-08-d2c75b21-1 (mt7622-snand-1ddr)
NOTICE:  BL2: Built : 22:51:20, Nov 19 2021
NOTICE:  SPI-NAND: FM35Q1GA (128MB)
NOTICE:  BL2: Booting BL31
NOTICE:  BL31: v2.4(release):OpenWrt v2021-05-08-d2c75b21-1 (mt7622-snand-1ddr)
NOTICE:  BL31: Built : 22:51:20, Nov 19 2021
serial_mtk serial@11002000: pinctrl_select_state_full: uclass_get_device_by_phandle_id: err=-19

U-Boot 2021.10 (Nov 19 2021 - 22:51:20 +0000)

CPU:   MediaTek MT7622
Model: mt7622-linksys-e8450-ubi
DRAM:  512 MiB
WDT:   Started with servicing (60s timeout)
MMC:
Loading Environment from UBI... SPI-NAND: FM35Q1GA (128MB)
Read 126976 bytes from volume ubootenv to 000000005f7bf3c0
Read 126976 bytes from volume ubootenv2 to 000000005f7de400
OK
In:    serial@11002000
Out:   serial@11002000
Err:   serial@11002000
reset button found
Loading Environment from UBI... Read 126976 bytes from volume ubootenv to 000000005f7bf3c0
Read 126976 bytes from volume ubootenv2 to 000000005f7de400
OK
Net:   eth0: ethernet@1b100000

      ( ( ( OpenWrt ) ) )       U-Boot 2021.10 (Nov 19 2021 - 22:51:20 +0000)

     1. Run default boot command.
     2. Boot system via TFTP.
     3. Boot production system from flash.
     4. Boot recovery system from flash.
     5. Load production system via TFTP then write to flash.
     6. Load recovery system via TFTP then write to flash.
     7. Load BL31+U-Boot FIP via TFTP then write to flash.
     8. Load BL2 preloader via TFTP then write to flash.
     9. Reboot.
     a. Reset all settings to factory defaults.
     0. U-Boot console

  Press UP/DOWN to move, ENTER to select, ESC/CTRL+C to quit
Volume fit not found!
Wrong Image Format for bootm command
ERROR: can't get kernel image!
Volume recovery not found!
Wrong Image Format for bootm command
ERROR: can't get kernel image!
Volume recovery not found!
Using ethernet@1b100000 device
TFTP from server 192.168.1.254; our IP address is 192.168.1.1
Filename 'openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery.itb'.
Load address: 0x48000000
Loading: *
______________________________________________________________________

Even trying to load the recovery gave me:
______________________________________________________________________
Wrong Image Format for bootm command
ERROR: can't get kernel image!
Volume recovery not found!
______________________________________________________________________

So at this point, I followed your instructions reinstalled the recovery flash, and then upgraded the snapshot. I want to thank you for your kindness and thoroughness. You have been most helpful and now I have a better understanding of how the process works. I thank you so much! <hugs> from me.

dangowrt, 2022-02-24:
Thank you for the kind words.
Good to read you got the serial console going, and also good to see the bootloader is all fine and yes, you did run the installer correctly and it successfully replaced the stock bootloader.
Most likely the content of UBI has been destroyed/overwritten by a firmware which wasn't meant to be used on a UBI bootchain -- ie. what would happen if you confirm the warning and continue what you shared on the screenshot in #68 .

The way forward is to feed the recovery image via TFTP (which will also be stored in flash again), as that will allow you to flash the sysupgrade image afterwards. Up to this point, nothing else will work as apparently UBI area has been wiped/destroyed and all it's content is lost.

dangowrt, 2022-03-03:
@keirak420 Were you able to resolve the issue and revive the device? If so, please close this issue. If not, I can help you with whatever needed to do so.

dangowrt, 2022-03-21:
There is also https://www.solarwinds.com/free-tools/free-tftp-server
That's a graphical TFTP server for windows, with extensive logging and easy to use. just select folder, disable firewall and should work and tell you what's going on.
Just in case anyone still trusts anything which says "solarwinds" and "secure" on it...

closed 2022-04-10 by dangowrt

Driver74, 2023-02-20:
I actually found myself in the same boat but have very limited knowledge of serial/connectivity.
I'm getting the same error and anything I try to flash through LUCI interface boot loops me back to recovery (initramfs) mode.
It takes .BIN files, throws an error for .ITB files and neither do anything after reboot.

Is opening up the router and connecting to the board the only way to make it take a tftp image?
If so what hardware do I need? I presume some sort of usb-serial adapter? what commands to start the process?

The older Asus routers seem much easier, hold reset while starting and built-in tftp server would load.

Thank you,

dangowrt, 2023-02-20:
Flashing any `.bin` file will lead to crash during boot which will then result in recovery image being loaded **until you empty /sys/fs/pstore**. If the recovery image is starting you will never need to open the device or connect serial port. That's the purpose of the recovery image. So just remove the device from power for a couple of minutes to make sure pstore is empty, then power on **while holding down reset button** until the LED turns yellow, then in recovery image make sure to flash the correct image (ie. something named `*-ubi-squashfs-sysupgrade.itb`).

And please do not hijack existing unrelated bug tickets.

Driver74, 2023-02-23:
Thanks for the info! Recovery ROMs must not allow any writing to happen unless it's booted with reset held. Once I held the reset button during power on it took my ubi-squashfs-sysupgrade.itb file and came back up running the new image. I definitely tried a couple variants of itb and bin files so luckily they didn't take.

I do apologize for jumping into a thread. I considered the factors that my issue seemed identical with path and current state the same, that OP's issue was resolved, and that others coming along later that are trying to take the next step will have more information to follow. I guess the correct way would be opening a new one and linking this one which I'll do going forward.

Thanks again.