dangowrt/owrt-ubi-installer #72: Dump AP with linksys e8450 / Belkin RT3200 occasionaly client associates but no connectivity

nkrepo, 2022-02-23:
I have flashed with two e8450 / Belkin RT3200 rotuers with openwrt snapshots build (now i am on OpenWrt SNAPSHOT r18777-1847382456). I mainly followed the dump ap guide https://openwrt.org/docs/guide-user/network/wifi/dumbap.

I configured the both wlan interfaces on both routers  with the same ssid and wpa_passphrase on both routers. The only difference is that the second router is on different channels. I have tried to setup 802.11r but soon I understand that I have more basic issues and I disabled it.
I also have disabled firewall, dnsmasq, odhcpd and igmproxy.

If I reboot the router everything again is hanky dory for a while, then the issue starts again, client goes through the 4-way handshake connects successfully but no connectivity.
If it is not a configuration error, I am afraid that there is a underlying issue either with hostapd or with the hardware driver for the MT7622BV , MT7915E.
The issue is driving me mad and I don't know where to start the debugging.

I started a discussion in openwrt forum, where i provide all my config files.
https://forum.openwrt.org/t/dump-ap-with-linksys-e8450-belkin-rt3200-occasionaly-client-associates-but-no-connectivity/121106

nkrepo, 2022-02-23:
Hi @mrkiko you are correct it's a heads up in case there is an underlying problem with the specific chipset, I will close the bug.

closed 2022-02-23 by nkrepo

githubbapoopa, 2022-03-04:
Had the same problems on DumbAP.

You have to choose your country code and then use channel width 80MHz.

My setup is AX > channel 36 (as this lower frequency is not dampend as much as higher frequencies) > Width 80MHz

Since then I was able to connect my mobilephone and my laptop correctly. Channel width 160MHz did not work despite set up correctly.

Cheers