dangowrt/owrt-ubi-installer #73: Image does not build: cannot find dependency libubus20210809 for blockd

touzoku, 2022-03-01:
Hi and thank you very much for your work!

The installation of a ready image went smoothly but since it was missing a required package and would not install it anymore due to outdated kernel version, I have decided to build the image manually.

I tried to run the steps listed in the Github Action definition both manually and using Github Actions. It fails on the step `make image PROFILE=linksys_e8450-ubi PACKAGES="$OPENWRT_UPG_PACKAGES"` with:

```
... skipped ...
Configuring ca-bundle.
Configuring urngd.
Configuring ppp-mod-pppoe.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libubus20210809 for blockd
 * pkg_hash_fetch_best_installation_candidate: Packages for blockd found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package blockd.
make[2]: *** [Makefile:168: package_install] Error 255
make[1]: *** [Makefile:123: _call_image] Error 2
make: *** [Makefile:241: image] Error 2
mv: cannot stat '/home/ubuntu/linksys-e8450-openwrt-installer/openwrt-ib/bin/targets/mediatek/mt7622/openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb': No such file or directory
```

Same error happens when I try to update the snapshot using `auc`:
```
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libubus20210809 for blockd
 * pkg_hash_fetch_best_installation_candidate: Packages for blockd found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package blockd.
make[2]: *** [Makefile:168: package_install] Error 255
make[1]: *** [Makefile:133: _call_manifest] Error 2
make: *** [Makefile:252: manifest] Error 2
asu.build.PackageSelectionError
Bad message (74)
```

I think it is an issue with the openwrt package repo and not your build script, but I wonder if there is an easy way to resolve this.

touzoku, 2022-03-01:
As a workaround, removing `blockd` from the `OPENWRT_UPG_PACKAGES` (line 239 in `build_installer.sh`) list has fixed the issue.

dangowrt, 2022-03-03:
Things like this are normal and expected in snapshot builds, as buildbots run in phases and `libubus` is part of phase1 (because it is used by packages installed by default, like ubus itself and all its users) while `blockd` is part of phase2. All you have to do is to wait for a day or two (phase2 takes a long time, but typically things will converge within 48 hours).

closed 2022-03-03 by dangowrt