dangowrt/owrt-ubi-installer #74: Unable to update opkg on Release v0.6.2

githubbapoopa, 2022-03-03:
I installed Realease v0.6.2 on my brandnew Belkin RT3200 but I'm unable to update opkg packages.

My goal i to install wireguard.

Why do I get these errors ond why isn't wireguard included in the image in the first place?

See errors below.

Thanks.

------------

Executing package manager

Downloading https://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/packages/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/packages/Packages.gz

Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.gz

Downloading https://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/kmods/5.10.92-1-df7cd26da92fdd5f2ede827f1b1112a7/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/kmods/5.10.92-1-df7cd26da92fdd5f2ede827f1b1112a7/Packages.gz

Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.gz

Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.gz

Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.gz

Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.gz
*** Failed to download the package list from https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.gz

-----

Errors

Failed to send request: Operation not permitted
Failed to send request: Operation not permitted
Failed to send request: Operation not permitted
Failed to send request: Operation not permitted
Failed to send request: Operation not permitted
Failed to send request: Operation not permitted
Failed to send request: Operation not permitted
Collected errors:
 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/packages/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/kmods/5.10.92-1-df7cd26da92fdd5f2ede827f1b1112a7/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

neilsan1366, 2022-03-03:
hi you are maybe a same adress ip box fai and router ?? no

dangowrt, 2022-03-03:
The errors indeed look like the device is not connected to Internet or at least DNS is unavailable and/or data/time maybe wrong and hence HTTPS will fail (it takes a while for NTP to set the correct date/time).
Apart from that, you should first upgrade to the most recent snapshot using `auc` because you will not be able to install packages on an more than 7 days old snapshot (and 0.6.2 is more than 7 days old by now) as they will no longer be available on the download server. Also be aware that in current snapshots there appears to be a bug which causes speed of the Ethernet switch to be **degraded to the speed of the slowest connected link partner** when using VLANs.

githubbapoopa, 2022-03-03:
Setup:
Router (Provider): 192.168.0.1 (DHCP range .10 - .253)
Belkin LAN: static 192.168.0.2  IPv4 GW 192.168.0.1 DNS 192.168.0.1 DHCP disabled, RA-Service disabled, DHCPv6-service disabled, NDP-Procy disabled

setup similar as described here as "Dumb Access Point"

https://openwrt.org/docs/guide-user/network/wifi/dumbap

I don't think it's the network setup (which is working fine) rather than time limits to installing new packages with opk...

So how do I install the missing packages (wireguard)?

githubbapoopa, 2022-03-03:
Done.

Via Webinterface: Attended Sysupgrade

After Sysupgrade I see a red Warning that says
NotFoundError
Ressource not found

Software > Update lists

There's only a dismiss button (instead of ok)

But I now can install packages and go further from here.

Thanks!!!

dangowrt, 2022-03-03:
No idea what's happening with the LuCI app, what you are describing sounds a bit like https://github.com/openwrt/asu/issues/332
Please close this ticket if you were able to install packages using opkg now.

githubbapoopa, 2022-03-04:
Yes it's like  https://github.com/openwrt/asu/issues/332 but that's another story...

Thanks.

closed 2022-03-04 by githubbapoopa