dangowrt/owrt-ubi-installer #79: Kernel panics, Router constantly goes into recovery mode

t3dium, 2022-03-09:
My router works fine until it randomly goes into recovery mode and this happens at least once every day. When it does go into recovery mode i then have to use ethernet, ssh and delete the pstore log, then reboot which temporarily fixes it.

I've tried re-flashing your sysupgrade image several times, and also updated to the latest version using the luci tool, but still have this issue. I've pasted cat /sys/fs/pstore* below:

```
<6>[    9.165285] mt7915e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20201105222304
<6>[    9.187663] mt7915e 0000:01:00.0: WA Firmware Version: DEV_000000, Build Time: 20201105222323
<6>[    9.312883] PPP generic driver version 2.4.2
<6>[    9.318118] NET: Registered protocol family 24
<14>[    9.329880] kmodloader: done loading kernel modules from /etc/modules.d/*
<6>[   12.599785] mtk_soc_eth 1b100000.ethernet eth0: Link is Down
<6>[   12.603364] mtk_soc_eth 1b100000.ethernet eth0: configuring for fixed/2500base-x link mode
<6>[   12.603562] mtk_soc_eth 1b100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx
<6>[   12.603587] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
<6>[   12.607618] mt7530 mdio-bus:00 lan1: configuring for phy/gmii link mode
<6>[   12.611842] 8021q: adding VLAN 0 to HW filter on device lan1
<6>[   12.616785] br-lan: port 1(lan1) entered blocking state
<6>[   12.616800] br-lan: port 1(lan1) entered disabled state
<6>[   12.617695] device lan1 entered promiscuous mode
<6>[   12.617703] device eth0 entered promiscuous mode
<6>[   12.631432] mt7530 mdio-bus:00 lan2: configuring for phy/gmii link mode
<6>[   12.631987] 8021q: adding VLAN 0 to HW filter on device lan2
<6>[   12.636769] br-lan: port 2(lan2) entered blocking state
<6>[   12.636782] br-lan: port 2(lan2) entered disabled state
<6>[   12.637882] device lan2 entered promiscuous mode
<6>[   12.643568] mt7530 mdio-bus:00 lan3: configuring for phy/gmii link mode
<6>[   12.644158] 8021q: adding VLAN 0 to HW filter on device lan3
<6>[   12.649494] mt7530 mdio-bus:00 lan3: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[   12.652472] br-lan: port 3(lan3) entered blocking state
<6>[   12.652484] br-lan: port 3(lan3) entered disabled state
<6>[   12.654805] device lan3 entered promiscuous mode
<6>[   12.658568] br-lan: port 3(lan3) entered blocking state
<6>[   12.658583] br-lan: port 3(lan3) entered forwarding state
<6>[   12.658702] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
<6>[   12.664290] mt7530 mdio-bus:00 lan4: configuring for phy/gmii link mode
<6>[   12.664863] 8021q: adding VLAN 0 to HW filter on device lan4
<6>[   12.669830] mt7530 mdio-bus:00 lan4: Link is Up - 100Mbps/Full - flow control rx/tx
<6>[   12.672226] br-lan: port 4(lan4) entered blocking state
<6>[   12.672239] br-lan: port 4(lan4) entered disabled state
<6>[   12.674276] device lan4 entered promiscuous mode
<6>[   12.676534] br-lan: port 4(lan4) entered blocking state
<6>[   12.676547] br-lan: port 4(lan4) entered forwarding state
<6>[   12.682258] mt7530 mdio-bus:00 wan: configuring for phy/gmii link mode
<6>[   12.682801] 8021q: adding VLAN 0 to HW filter on device wan
<6>[   12.687548] mt7530 mdio-bus:00 wan: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[   12.689538] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
<6>[   15.043015] br-lan: port 5(wlan1) entered blocking state
<6>[   15.043029] br-lan: port 5(wlan1) entered disabled state
<6>[   15.043246] device wlan1 entered promiscuous mode
<6>[   15.043330] br-lan: port 5(wlan1) entered blocking state
<6>[   15.043338] br-lan: port 5(wlan1) entered forwarding state
<6>[   15.045045] br-lan: port 5(wlan1) entered disabled state
<6>[   15.170457] br-lan: port 6(wlan0) entered blocking state
<6>[   15.170470] br-lan: port 6(wlan0) entered disabled state
<6>[   15.170709] device wlan0 entered promiscuous mode
<6>[   15.170795] br-lan: port 6(wlan0) entered blocking state
<6>[   15.170804] br-lan: port 6(wlan0) entered forwarding state
<6>[   15.419884] mt7530 mdio-bus:00 lan1: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[   15.419925] br-lan: port 1(lan1) entered blocking state
<6>[   15.419935] br-lan: port 1(lan1) entered forwarding state
<6>[   16.045992] br-lan: port 6(wlan0) entered disabled state
<6>[   22.615047] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
<6>[   22.615184] br-lan: port 5(wlan1) entered blocking state
<6>[   22.615194] br-lan: port 5(wlan1) entered forwarding state
<6>[   23.036756] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
<6>[   23.036895] br-lan: port 6(wlan0) entered blocking state
<6>[   23.036903] br-lan: port 6(wlan0) entered forwarding state
<6>[   54.623461] br-lan: port 3(lan3) entered disabled state
<6>[   54.623929] mt7530 mdio-bus:00 lan3: Link is Down
<6>[ 5640.084425] br-lan: port 1(lan1) entered disabled state
[... 171 lines cut ...]
<7>[160817.599667] x19: dead0000000000f0 x18: ffffffc010cc7570
<7>[160817.605058] x17: 000000000000001e x16: 0000000b7caccada
<7>[160817.610449] x15: 000031cd89e2066e x14: ffffffc0108083e0
<7>[160817.615840] x13: 0000000000000357 x12: ffffffffffffffff
<7>[160817.621231] x11: 0000000000000040 x10: ffffff8000ce1914
<7>[160817.626622] x9 : ffffff8002a7857c x8 : 0000000000000000
<7>[160817.632013] x7 : ffffffffffffffff x6 : 0000000000000000
<7>[160817.637404] x5 : 0000000000000002 x4 : 0000000000000000
<7>[160817.642795] x3 : ffffff800034ee00 x2 : 0000000000000000
<7>[160817.648186] x1 : dead000000000100 x0 : 00000000ffffffff
<7>[160817.653576] Call trace:
<7>[160817.656103]  nf_flow_offload_ip_hook+0x778/0x1340 [nf_flow_table]
<7>[160817.662276]  nf_flow_table_offload_setup+0x468/0x660 [nf_flow_table]
<7>[160817.668709]  process_one_work+0x1d4/0x370
<7>[160817.672796]  worker_thread+0x178/0x4e0
<7>[160817.676625]  kthread+0x120/0x124
<7>[160817.679933]  ret_from_fork+0x10/0x18
<0>[160817.683589] Code: d1004013 eb0002bf 54000580 52800014 (f9401263)
<4>[160817.689759] ---[ end trace d2e0a6d64d92b633 ]---
<0>[160817.701479] Kernel panic - not syncing: Oops: Fatal exception
<2>[160817.707308] SMP: stopping secondary CPUs
<0>[160817.711313] Kernel Offset: disabled
<0>[160817.714881] CPU features: 0x0000002,04002004
<0>[160817.719226] Memory Limit: none
```

mrkiko, 2022-03-10:
Hello!

this does not look like an issue related to this project, which aims to build an easy path to transition a device to UBI layout, replacing the stock bootloader with a newer version, thereby adding some important enhancements.

Altough everything is possible, this kernel panic does not look like it's related to porting issues.

mrkiko, 2022-03-11:
Hello again!
I noticed a very similar (if not the same) issue being reported here:
https://github.com/openwrt/openwrt/issues/9440
So you may join the discussion here!

t3dium, 2022-03-12:
Someone said the issue was related to nftables firewall 3 being used in combination with iptables in dangowrt’s image, although the has added firewall 4 in the pre-release version. I’ve just disabled hardware offloading for the time being but might test the pre-release version once it gets released.

closed 2022-03-12 by t3dium