dangowrt/owrt-ubi-installer #80: Maximum transmit power seems not to work

WiredLife, 2022-03-16:
hello,
i set my region to panama (PA) and Maximum transmit power shows "Current power: 27 dBm"
but it seems it does not change the tx power
is there a way to get the max power output?
maybe change something in the firmware?

mrkiko, 2022-03-16:
>     seems not to work (Issue #80)

Hello!

This issue does not seem related to this project, which aims to create an installer to switch the device to an open-source updated bootchain and using UBI.

You may have better luck by posting on the OpenWRt forum.

dangowrt, 2022-03-17:
@WiredLife I agree with @mrkiko that this is not an issue with the installer but could either be a hardware limitation or tx-power not being handled correctly by the [mt76](https://www.github.com/openwrt/mt76) driver.

closed 2022-03-17 by dangowrt