dangowrt/owrt-ubi-installer #86: Backup Help Needed.

TuckerEck, 2022-03-30:
So recently I flashed openwrt the using the guide provided here but I forgot to backup the mtd so I can revert to the original manufacturer u boot bootloader and firmware.
Is there any way to revert a firmware without a backup or any tool or is someone has a backup?

-Thanks for your help.

dangowrt, 2022-04-05:
The installer itself stores a backup of the previous bootchain (**not** the stock firmware itself, but that you can download easily from either vendor). Unless you have re-run the installer or otherwise completely wiped the flash, this is still going to be there.
See [instructions here](https://github.com/dangowrt/owrt-ubi-installer#upgrading-to-the-latest-openwrt-snapshot) on how to copy it from the device to your computer.

The files you now got are referred to as **"minimal backup"**

You may then follow the [restore procedure](https://github.com/dangowrt/owrt-ubi-installer#restoring-the-vendorofficial-firmware)

TuckerEck, 2022-05-16:
My mtd3 is 125mb but seems like it a copy of the openwrt snapshot bc everytime I restore to the vendor firmware it goes right to the openwrt login page

dangowrt, 2022-05-24:
@TuckerEck did you ever re-run the `*installer.itb` after you had initially flashed OpenWrt?
If so, the "previous bootchain" in the `boot_backup` will be what ever is found in the flash at the time you run last run the installer.

huyz, 2023-03-05:
@ghost It sounds like you did a full backup (https://github.com/dangowrt/owrt-ubi-installer#device-flash-complete-backup-procedure) _after_ you had already installed OpenWrt. So unless I'm mistaken, that explains why `mtd3` would be of OpenWrt, not the stock firmware (which you'll have to download from the vendor).

huyz, 2023-03-05:
I wish the instructions had https://github.com/dangowrt/owrt-ubi-installer#device-flash-complete-backup-procedure at the top before the installation part.  That way people wouldn't forget to backup the proper mtd3.

dangowrt, 2023-03-05:
The thing is that you will hardly ever really need a complete backup including `mtd3`. As long as Belkin/Linksys provide their firmware files for download on their servers, you can as well just take the firmware from there and use only the partial backup of `/dev/mtd3` the installer does and puts into `boot_backup` UBI volume.

closed 2025-07-04 by dangowrt