dangowrt/owrt-ubi-installer #87: port forwarding on E8450

doug-gilbert, 2022-04-03:
I have gone down this path with a Linksys E8450 with both the firmware suggested on this page and the latest daily snapshots. Everything works okay that I want apart from one thing: port forwarding. For my application that means I need to stay with my old Buffalo router or go back to stock Linksys firmware. Is this a known issue for OpenWRT firmware with the E8450 and/or the RT3200? Could this be related to the recent shift to fw4 (i.e. nftables)?

dangowrt, 2022-04-03:
I'm using port-forwarding on this device every day and it works for me (IPv4, forwarding a few TCP and UDP ports from public v4 on WAN interface to specific hosts on the LAN side). There are still problems with `miniupnp` and firewall4, see https://github.com/openwrt/packages/issues/17871 . Forwarding rules setup manually should work fine though (with both, fw4/iptables and fw4/nftables). How did you setup the forwarding? Can you share the relevant parts of your `/etc/config/firewall`?

doug-gilbert, 2022-04-04:
I'm trying to use luci to set up port forwarding (i.e. using the same procedure that works with my WZR-HP-AG300H). Cleaned out my settings with firstboot.  Loaded today's openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb [4 April]. Changed default brlan to 192.168.48.0/24 . Turned off dhcpd on brlan (I have my own running on that subnet). Set up two port forwards on ports 22 and 443 to 192.168.48.96 . Rebooted. Used https://canyouseeme.org/ to check if those forwards worked. NO they did not! Must be the 6th time I have tried this, always with the same result. Github didn't support the extension on the "firewall" file that you asked for, so I added .txt .
Advice to others: if you have port forwarding working on a E8450 with a firewall prior to fw4 (nftables) do NOT upgrade it until this is sorted out.
[firewall.txt](https://github.com/dangowrt/owrt-ubi-installer/files/8413302/firewall.txt)

dangowrt, 2022-04-04:
Very odd. You configuration looks ok. I've tried it (obviously with my LAN /24 subnet and changing the dest IP and port to a service available on my local network) and connecting works (I'm also using port 22 and port 443 for this experiment, to make sure it also works for ports which are used locally by the router for SSH/dropbear and HTTPS/uhttpd)
I'm running OpenWrt SNAPSHOT r19351-3e0daca644
[image]

doug-gilbert, 2022-04-04:
My firmware version is:  OpenWrt SNAPSHOT r19353-839b1ff1fc / LuCI Master git-22.089.43958-7110635 . To your question: yes it does, it is one of my dhcp servers in failover mode with another box on that subnet (both RPi_s). It also does DNS for my subnet, forwarding to 192.168.48.1 (E8450) anything it can't resolve. Not sure how what 'plug' (192.168.48.96) is doing should impact the E8450's ability to forward to it. OpenWRT firmware should play nice with other boxes doing some of its chores (e.g. DHCP and DNS).

dangowrt, 2022-04-04:
Yep, that should all work then and is fairly similar to what I'm doing myself (and works nicely). As it is also not a problem related to the installer or even this specific device, maybe the [OpenWrt Forum](https://forum.openwrt.org/) is a better place to discuss this problem.

doug-gilbert, 2022-04-05:
I'm sick of getting yet another account at some place just to ask one question. Perhaps you might show the attachment to the openwrt thread your referred to above. My question (which I can't answer) is whether there is anything defective with that ruleset.
Also are snapshots archived somewhere so I can get one for the E8450 _before_ fw4 screwed the pooch?
[nft_list_ruleset.txt](https://github.com/dangowrt/owrt-ubi-installer/files/8414208/nft_list_ruleset.txt)

dangowrt, 2022-04-05:
You can actually do a one-click sign-up to the forum using your Github account (via OAuth).
If you still don't want to use the forums, please move this ticket to https://github.com/openwrt/openwrt as this repository is really just about the **installer**. It is **not** a community build what-so-ever. The device is officially supported by OpenWrt, just the installer is not (yet).

The ruleset looks fine to me, just like your firewall configuration (which I have tried and works for me -- doesn't mean that there isn't a problem, but I'm not the author/maintainer of firewall4 and actually don't know much about nftables...)

closed 2022-04-10 by dangowrt