dangowrt/owrt-ubi-installer #90: E8450 not booting. I only getting a flashing blue power light

Eddyswu, 2022-04-07:
Hi,

I just got my E8450 and try to install openwrt on it.

I download release v0.6.3 and follow the steps to backup the linksys firmware without problem. I followed all the steps, get the linksys firmware to boot again, did a factory reset to get ready for openwrt installation.

I followed the steps under Installing OpenWrt. I finished step 1 to 11 and waiting for the router to reboot.

After waiting for over 10 minutes, I am still getting a flashing blue power light. what can I do now? Did I brick my device?

dangowrt, 2022-04-07:
v0.6.3 is entirely untested at this point and likely to not work yet -- hence marked as a pre-release. You will have to wire up a serial port to the router to find out what's going on (I was going to give it some testing in the next days, but didn't do that yet).

dangowrt, 2022-04-07:
Ok, there was a bug in the installer which made it abort and not actually replace the bootloader. So your device is now still with the vendor bootchain, but without any firmware it can load. At this point you will unfortunately have to open the device, connect the serial port (3.3V TTL level; connect RX, TX, and GND only) and manually load the stock firmware or the (fixed) installer to revive the device.
Please do **not** return the bricked device to the store! If you don't want to get involved with the procedure connecting to the serial port, contact me by email and mention your location, I'll find someone in your country to fix it for you.

Sorry for the mess :(

dangowrt, 2022-04-07:
Fixed the problem https://github.com/dangowrt/owrt-ubi-installer/commit/26e2af9a1bb173bafaefdebb8091dd2f8fbff27e, deleted v0.6.3 (broken) pre-release, tagged v0.6.4, did lots of testing and released v0.6.4.

Eddyswu, 2022-04-07:
dangwrt, Thanks for your quick reply. I think I will have to fix it myself, since I am living in Hong Kong. it will be great if you can give me some pointer to how to connect to the serial port. I managed to open the router and localed the serial port already. I understand I need to get a UART to USB connector. If you can get me some information on this, that will help me a lot. 😀

Once I get connection over the seral port, can I just load new 0.6.4 installer? or should I load the 0.6.1 installer?

Eddyswu, 2022-04-07:
I think I have enough info to start. Thanks a lot.  I will try to pickup a USB to serial converter at a local electronic shop and give it a try. Thanks

dangowrt, 2022-04-08:
Hehe, that's the cool thing about where you are: You can have a USB to serial adapter within one day :) (that's far from possible in most places in the world, you'd have to wait at least a week until it arrives in the mail)

dangowrt, 2022-04-10:
@Eddyswu Did you get it working by now? If so, please close this issue.