dangowrt/owrt-ubi-installer #91: Unable to install Openvpn

mgupta80, 2022-04-10:
I am unable to install Openvpn, it complains that hash is not matching. 

root@OpenWrt:~# opkg install openvpn-openssl
Installing openvpn-openssl (2.5.5-1) to root...
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/aarch64_cortex-a53/packages/openvpn-openssl_2.5.5-1_aarch64_cortex-a53.ipk
Installing liblzo2 (2.10-4) to root...
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/aarch64_cortex-a53/packages/liblzo2_2.10-4_aarch64_cortex-a53.ipk
Installing libopenssl1.1 (1.1.1n-1) to root...
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/aarch64_cortex-a53/base/libopenssl1.1_1.1.1n-1_aarch64_cortex-a53.ipk
Configuring liblzo2.
Configuring libopenssl1.1.
Configuring openvpn-openssl.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.110-1-ecdf38a0bcca08e02a6b27f1f92cf94b) for kmod-tun

Could someone please guide if I am missing something ?

dangowrt, 2022-04-11:
The problem is that the kernel version installed on the device no longer matches what is offered on our download servers.
Please install a more recent build from https://downloads.openwrt.org/releases/22.03-SNAPSHOT/targets/mediatek/mt7622/

Alternatively you can install `auc` or `luci-app-atttendedsysupgrade` to semi-automatically do so without having to manually reinstall the packages you have already selected.

If it turns out not to work due to (other) unfulfilled dependencies, you will unfortunately have to wait for a day or two for our [buildbot infrastructure](https://buildbot.openwrt.org/openwrt-22.03/packages/#/builders/27) to have re-built all packages to include the dependency on the [recently updated kernel version](https://github.com/openwrt/openwrt/commit/662d1f9f8db3e8bd295dc93b9fd5383741bbadf5) (5.10.110).

mgupta80, 2022-04-12:
Thanks, I got this working by using a more recent build as suggested

closed 2022-04-12 by mgupta80