dangowrt/owrt-ubi-installer #92: Devices keeps disconnecting on 2.4Ghz band

mgupta80, 2022-04-19:
Hi, My Nest IQ outdoor cameras keeps disconnecting very frequently. They only support 2.4 GHz wifi band in UK, I have also tried giving them a static IP address but that didn't helped either. 

Could someone please suggest if they are having a similar issue and are there any logs that might be useful for debugging?

[image]

dangowrt, 2022-04-19:
Please discuss this in the [OpenWrt forums](https://forum.openwrt.org) -- this repository is merely the installer and that has apparently worked well for you :wink:

closed 2022-04-19 by dangowrt

neilsan1366, 2022-05-05:
do you have found a solution because i has same problem unfortunatly i'm desesperate

mrkiko, 2022-05-06:
The problem is not related to this repository, which provides an installer to convert this device to UBI layout with a fresh bootchain with great features.
Your problem might bebetter discussed in the OpenWRt forum as suggested, maybe after trying to install the latest master snapshot for ubi layout!