dangowrt/owrt-ubi-installer #93: How to upgrade to self-compiled firmware ？

RottenLeaf24, 2022-04-20:
I upgraded my router using the itb image from the release, but I can no longer use the bin image to upgrade.
I want to compile some software myself and I end up with an image called ```openwrt-mediatek-mt7622-linksys_e8450-squashfs-sysupgrade.bin```, how can I use it for firmware upgrade?

dangowrt, 2022-04-20:
No, the `openwrt-mediatek-mt7622-linksys_e8450-squashfs-sysupgrade.bin` image is generated for the device when running the stock bootloader.
You have to select the UBI variant in menuconfig, that will then result in a file called `openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb` which works with sysupgrade.

RottenLeaf24, 2022-04-20:
> No, the `openwrt-mediatek-mt7622-linksys_e8450-squashfs-sysupgrade.bin` image is generated for the device when running the stock bootloader. You have to select the UBI variant in menuconfig, that [...]

I also want to build an image of itb，But the target image only has these kinds in the picture below:
[image]
I don't know how to build an image in itb format？

dangowrt, 2022-04-20:
As I said, you just have to select the right `Target Profile`:
[image]

RottenLeaf24, 2022-04-20:
@dangowrt 
Thank you very much for your help,I found It should select "Multiple Devices" in Target Profile first.
[image]
And then I can see the ubi option in the target devices.
[image]

I have compiled the firmware in itb format，It's really appreciate for solving my problem.

dangowrt, 2022-04-20:
I'm glad it was easy to solve and I'm happy to help. I'm closing this issue and it is resolved.

closed 2022-04-20 by dangowrt