dangowrt/owrt-ubi-installer #97: Add Warning About New Vendor Firmware?

timedroid, 2022-05-04:
From the [changelog](https://www.belkin.com/support/assets/belkin/releasenotes/RT3200_1.2.00.273012Customer_ReleaseNotes.txt), it appears firmware 1.2 and above only allow signed upgrades and do not allow downgrading to a previous firmware version.

The RT3200 I bought a few weeks ago came with 1.0.01.101415. I went through the out of the box setup, but did not check the box for automatic firmware upgrades. I'm assuming doing so would eventually leave one without the ability to install OpenWRT? Besides opening it up and connecting to the UART (or maybe only the JTAG).

dangowrt, 2022-05-11:
I had a look at the new firmware and also the intermediate firmware (1.1.01.272918 (Unsigned)) which allows flashing the new firmware (1.2.00.273012 (Signed)) and there is no need to issue warnings.
The bad news: They are using GnuPG v1.4.7 (because it's still under GPLv2 licence, newer versions of GnuPG are under GPLv3)

The good news: Linksys was kind enough to leave the also the **private key** next to the door and use **the same key for encryption and signature**. See here
```
$ binwalk -e FW_RT3200_1.1.01.272918_PROD_unsigned.img
...
$ ls  _FW_RT3200_1.1.01.272918_PROD_unsigned.img.extracted/squashfs-root/root/.gnupg/
gpg.conf  pubring.gpg  random_seed  secring.gpg  trustdb.gpg
$ gpg --import < FW_RT3200_1.1.01.272918_PROD_unsigned.img.extracted/squashfs-root/root/.gnupg/pubring.gpg 
gpg: key 99DAC426DCF76BA1: public key "Aruba recipient <aruba_recipient@linksys.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1
$ gpg --import < FW_RT3200_1.1.01.272918_PROD_unsigned.img.extracted/squashfs-root/root/.gnupg/secring.gpg 
gpg: key 99DAC426DCF76BA1: "Aruba recipient <aruba_recipient@linksys.com>" not changed
gpg: key 99DAC426DCF76BA1: secret key imported
gpg: Total number processed: 1
gpg:              unchanged: 1
gpg:       secret keys read: 1
gpg:   secret keys imported: 1
```

I'm pretty sure uploading the secret key to this repository would be a copyright violation.
Having the script download the firmware from Linksys and use binwalk to extract the keyring, import the keys and use them to sign and encrypt the installer would probably be an option...

dangowrt, 2022-05-12:
Done. Now shipping signed/encrypted images in [v0.6.5](https://github.com/dangowrt/owrt-ubi-installer/releases/tag/v0.6.5).

closed 2022-05-12 by dangowrt

dangowrt, 2022-10-11:
I have recently tried flashing v1.0.0 installer on all publicly available and historic versions of the stock firmware, and it all worked. So up to now we are fine.