dangowrt/owrt-ubi-installer #100: [question] Regarding attended sysupgrade (auc)

MarcoBuess, 2022-05-13:
I'm fairly new to OpenWRT in general and trying to figure out the easiest and safest way for me to be up to date. I installed `luci-app-attendedsysupgrade` and `auc` as described [in this issue](https://github.com/dangowrt/owrt-ubi-installer/issues/98) after I found it wasn't coming with the snapshot I'm using.

Now I'm wondering if it will retain my configuration or if I need to take further steps to backup and restore the configuration like described in [Upgrading OpenWrt firmware using LuCI and CLI](https://openwrt.org/docs/guide-user/installation/generic.sysupgrade) and [Backup and restore](https://openwrt.org/docs/guide-user/troubleshooting/backup_restore). I spent quite some time to set everything up the way I need it and I don't want to lose any of the configurations I made.

Thanks for taking your time to (hopefully) guide me through this.

dangowrt, 2022-05-16:
Using `auc` does keep your settings (and additionally installed packages).
Anyway, if you have invested a lot of time into your setup it is **always** a good idea to keep configuration backups as well...

MarcoBuess, 2022-05-17:
Thanks for you input highly appreciated

closed 2022-05-17 by MarcoBuess