dangowrt/owrt-ubi-installer #99: Help

EthanTu, 2022-05-13:
Navigate to Administration -> Firmware Upgrade.
Upload openwrt-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery.itb. (No word installer)

Navigate to System -> Backup / Flash Firmware.
Upload openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb.

Hold down the "reset" button (below the "WPS" button) whilst powering on the device.
Release the button once the power LED turns into orange/yellow.(awalys blue)

Now, only power light is blue, netowrk is unreachable.
What can i do now?

dangowrt, 2022-05-13:
You will need to open the device and connect the serial console to recover using U-Boot TFTP.
Next time read the manual first :wink:

incendiary, 2022-09-11:
Incase anyone else runs into this, there is a video walk through here: https://www.youtube.com/watch?v=KgDbRJMG6uE

john-clark, 2023-12-30:
This video is not  perfect, but it is close. I just went through this and was able to get my router repaired. He hints that documentation tells how to do but does not point to the instructions .

Ultimately if your uboot is working (you can test this blindly), setup a tftpserver on windows, make sure to turn off the firewall, set the computer ip to 192.168.1.254 and make sure the filename is correct and test get the file (from another machine). Then boot the router and after the lights blink a bit it should show up on 192.168.1.1 so you can upload the sysupgrade file. Then reboot the router without the ethernet connected and see if it is fully recovered.

closed 2025-07-04 by dangowrt