dangowrt/owrt-ubi-installer #102: 1.2 installed and unable to load original firmware FW_RT3200_1.0.01.101415_prod

flowersrj, 2022-05-26:
When I try to degrade firmware from 1.2.00.273012 all I get after a minute or so is the message "the image file is incorrect".  

Any advice on how to get router to reload earlier firmware?

mrkiko, 2022-05-26:
Hi!

so - it depens on the reasons why you are downgrading.

If you wanted to install OpenWrt via the UBI installer, then you may not actually need to downgrade, since support for signed images is available!! :)
Otherwise, you may try to hack things up to build a signed stock image if no u-boot modifications happened in the meantime from vendor side.

flowersrj, 2022-05-27:
Just looking for a way to install Openwrt--Sent from my Android phone with GMX Mail. Please excuse my brevity.

flowersrj, 2022-05-27:
Where to I get the UBI version and instructions on installing it?

flowersrj, 2022-06-02:
> It's the same as for earlier versions, just that you have to use the _signed.itb images instead. No downgrading needed what-so-ever.
> Ie. just download and flash openwrt-22.03-snapshot-r19338-ae64d0624c-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery-installer_signed.itb.

Danel,

 

Would you know if a GUI(Luci) is included ?

 

Thanks,

Riochi
 

 
 

Sent: Tuesday, May 31, 2022 at 9:26 AM

dangowrt, 2022-06-02:
Yes, all images offered for download here come with LuCI preinstalled.

mrkiko, 2022-06-11:
Closing this issue right now. Feel free to re-open it in case it's needed!

closed 2022-06-11 by mrkiko

org0000h, 2022-06-24:
> 

I think you should change the readme file . The  downgrading content part  of it  is confusing.

flowersrj, 2022-07-08:
Daniel,Install worked great.  Could you provide a link where I could look for updates?Also, am I correct in assuming that backups taken from snapshots can't be restored?Thank you,Rich--Sent from my Android phone with GMX Mail. Please excuse my brevity.

dangowrt, 2022-10-11:
> Any advice on how to get router to reload earlier firmware?

You can't (and that's intentional, complain at Belkin/Linksys).

You can, however, use the `_signed` version of the installer and install
OpenWrt **without having to downgrade to v1.0**.