dangowrt/owrt-ubi-installer #104: stuck in  recovery (initramfs) mode

Tom-Neverwinter, 2022-06-04:
woke up and noticed wifi was down. found out that the device is stuck in recovery.

Rebooting [undervolt issue?] or attempting to overwrite ubi partition with installer or installer signed:

openwrt-22.03-snapshot-r19338-ae64d0624c-mediatek-mt7622-linksys_e8450-ubi-initramfs-recovery_signed.itb

openwrt-22.03-snapshot-r19338-ae64d0624c-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb

etc.

leads back to:

OpenWrt SNAPSHOT r17758-b118efa0d2

probably have to flash over console?

dangowrt, 2022-06-05:
Please check if there is a crashlog in pstore:
```
ls /sys/fs/pstore
```
If so, investigate **and delete those files**, then the router will boot again to production firmware.

**NEVER** re-run the installer. This won't fix anything if you are already running OpenWrt at this point.

Tom-Neverwinter, 2022-06-06:
`Please append a correct "root=" boot option; here are the avai[10;12H56337] 1f00             512 mtdblock0                                [11;15H43]  (driver?)                    [12;12H62875] 1f01             256 mtdblock1[13;16H9]  (driver?)                    [14;13H9401] 1f02             512 mtdblock2[15;16H5]  (driver?)                    [16;12H75933] 1f03             512 mtdblock3[17;16H6]  (driver?)                    [18;12H82463] 1f04            1024 mtdblock4[19;16H7]  (driver?)                    [20;13H8988] 1f05             128 mtdblock5[21;15H92]  (driver?)                    [22;12H95520] 1f06             128 mtdblock6[23;16H4]  (driver?)                    [23;1H[24;1H[K- dmesg-ramoops-0 [Readonly] 115/130 88%[23;1H[1;2H5>[    2.256866] UBI: auto-attach mtd10      [2;12H60284] ubi0: attaching mtd10 [3;11H317075] ubi0: scanning is finished[4;2H3>[    2.320831] ubi0 error: ubi_read_volume_table: the layout volume was not f[5;13H8108] ubi0 error: ubi_attach_mtd_dev: failed to attach mtd10, error [6;12H35269] UBI error: cannot attach mtd10                               [7;2H7>[    2.340441] VFS: Cannot open root device "(null)" or unknown-block(0,0): e[8;13H7955] Please append a correct "root=" boot option; here are the avai[9;12H56337] 1f00             512 mtdblock0                                [10;15H43]  (driver?)                    [11;12H62875] 1f01             256 mtdblock1[12;16H9]  (driver?)                    [13;13H9401] 1f02             512 mtdblock2[14;16H5]  (driver?)                    [15;12H75933] 1f03             512 mtdblock3[16;16H6]  (driver?)                    [17;12H82463] 1f04            1024 mtdblock4[18;16H7]  (driver?)                    [19;13H8988] 1f05             128 mtdblock5[20;15H92]  (driver?)                    [21;12H95520] 1f06             128 mtdblock6[22;16H4]  (driver?)                    [23;11H402050] 1f07             128 mtdblock7[23;1H[24;1H[K- dmesg-ramoops-0 [Readonly] 116/130 89%[23;1H[1;12H60284] ubi0: attaching mtd10 [2;11H317075] ubi0: scanning is finished[3;2H3>[    2.320831] ubi0 error: ubi_read_volume_table: the layout volume was not f[4;13H8108] ubi0 error: ubi_attach_mtd_dev: failed to attach mtd10, error [5;12H35269] UBI error: cannot attach mtd10                               [6;2H7>[    2.340441] VFS: Cannot open root device "(null)" or unknown-block(0,0): e[7;13H7955] Please append a correct "root=" boot option; here are the avai[8;12H56337] 1f00             512 mtdblock0                                [9;15H43]  (driver?)                    [10;12H62875] 1f01             256 mtdblock1[11;16H9]  (driver?)                    [12;13H9401] 1f02             512 mtdblock2[13;16H5]  (driver?)                    [14;12H75933] 1f03             512 mtdblock3[15;16H6]  (driver?)                    [16;12H82463] 1f04            1024 mtdblock4[17;16H7]  (driver?)                    [18;13H8988] 1f05             128 mtdblock5[19;15H92]  (driver?)                    [20;12H95520] 1f06             128 mtdblock6[21;16H4]  (driver?)                    [22;11H402050] 1f07             128 mtdblock7[23;16H4]  (driver?)                    [23;1H[24;1H[K- dmesg-ramoops-0 [Readonly] 117/130 90%[23;1H[1;11H317075] ubi0: scanning is finished[2;2H3>[    2.320831] ubi0 error: ubi_read_volume_table: the layout volume was not f[3;13H8108] ubi0 error: ubi_attach_mtd_dev: failed to attach mtd10, error [4;12H35269] UBI error: cannot attach mtd10                               [5;2H7>[    2.340441] VFS: Cannot open root device "(null)" or unknown-block(0,0): e[6;13H7955] Please append a correct "root=" boot option; here are the avai[7;12H56337] 1f00             512 mtdblock0                                [8;15H43]  (driver?)                    [9;12H62875] 1f01             256 mtdblock1[10;16H9]  (driver?)                    [11;13H9401] 1f02             512 mtdblock2[12;16H5]  (driver?)                    [13;12H75933] 1f03             512 mtdblock3[14;16H6]  (driver?)                    [15;12H82463] 1f04            1024 mtdblock4[16;16H7]  (driver?)                    [17;13H8988] 1f05             128 mtdblock5[18;15H92]  (driver?)                    [19;12H95520] 1f06             128 mtdblock6[20;16H4]  (driver?)                    [21;11H402050] 1f07             128 mtdblock7[22;16H4]  (driver?)                    [23;13H8575] 1f08           30720 mtdblock8[23;1H[24;1H[K- dmesg-ramoops-0 [Readonly] 118/130 90%[23;1H[1;2H3>[    2.320831] ubi0 error: ubi_read_volume_table: the layout volume was not f[2;13H8108] ubi0 error: ubi_attach_mtd_dev: failed to attach mtd10, error [3;12H35269] UBI error: cannot attach mtd10                               [4;2H7>[    2.340441] VFS: Cannot open root device "(null)" or unknown-block(0,0): e[5;13H7955] Please append a correct "root=" boot option; here are the avai[6;12H56337] 1f00             512 mtdblock0                                [7;15H43]  (driver?)                    [8;12H62875] 1f01             256 mtdblock1[9;16H9]  (driver?)                    [10;13H9401] 1f02             512 mtdblock2[11;16H5]  (driver?)                    [12;12H75933] 1f03             512 mtdblock3[13;16H6]  (driver?)                    [14;12H82463] 1f04            1024 mtdblock4[15;16H7]  (driver?)                    [16;13H8988] 1f05             128 mtdblock5[17;15H92]  (driver?)                    [18;12H95520] 1f [... cut ...] atures: 0x00003000,00000802[23;13H4861] Memory Limit: none               [23;1H[24;1H[K- dmesg-ramoops-0 [Readonly] 130/130 100%[23;1H[23;1H[23;1H[23;1H[23;1H[23;1H[23;1H[23;1H[23;1H[23;1H[23;1H[23;1H[23;1H[23;1H[23;1H[23;1H[23;1H[23;1H[23;1H[23;1H[23;1H[23;1H[23;1H[23;1H[23;1H[24;1H[K:q[23;1H[23;1H[24;1H[K- dmesg-ramoops-0 [Readonly] 130/130 100%[23;1H[24;1H[K[?1049l]0;root@OpenWrt: /sys/fs/pstoreroot@OpenWrt:/sys/fs/pstore# exit`

dangowrt, 2022-06-06:
This looks like you have flashed the non-UBI firmware despite already having converted the device to UBI layout.
MTD devices do **not** have a partition table (such as MBR or GPT), instead the partition layout is given with the device tree and part of the OS you flash. Hence, if you get to see too many MTD devices, that's because you have booted with a device tree which defines them (but, obviously, the right content is not in place).
Just make sure to flash the correct image (with `-ubi-` in it's name) and you'll be fine.

Tom-Neverwinter, 2022-06-06:
flashing from "http://192.168.1.1/cgi-bin/luci/admin/system/flash" 

using: "openwrt-22.03-snapshot-r19338-ae64d0624c-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb"

still loads into recovery (initramfs) mode

Tom-Neverwinter, 2022-06-06:
solved. issue is "I'm an idiot. "

closed 2022-06-06 by Tom-Neverwinter

dangowrt, 2022-06-06:
Good to read that the issue has been resolved!