dangowrt/owrt-ubi-installer #105: Kernel update

fever365, 2022-06-10:
Kernel Version is 5.10.113
how can i update it to 5.10.120

because i install the dnsmasq-full,notice the message like:

The installed version of package kernel is not compatible, require 5.10.120-1-556ff95e… while 5.10.113-1-d3a5d2af… is installed.
yellow warning :  kernel VERSION INCOMPATIBLE

mrkiko, 2022-06-11:
Hi!

This issue does not seem to be related to this project. As an example, youcan install any newer snapshot sysupgrade image for the UBI layout to get a more recent kernel, or a release image to get little bit more time to install kernel-dependent packages.
This project aims to create an installer to convert the device to UBI layout. Once the conversion is done, you can install any OpenWrt image, provided it is made for the UBI layout of this device.

dangowrt, 2022-06-12:
I recommend to use `auc` or `luci-app-attendedsysupgrade` to update to the most recent 22.03-SNAPSHOT or alternatively install the current 22.03.0 release candidate. Either way should allow to install kernel module packages matching the installed kernel.

closed 2022-09-12 by dangowrt