dangowrt/owrt-ubi-installer #106: Image check failed

jottr, 2022-06-10:
I'm currently on an older snapshot of `22.03` and would like to upgrade to the latest snapshot. 
I downloaded the latest sysupgrade image: [openwrt-22.03-snapshot-r19338-ae64d0624c-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb](https://github.com/dangowrt/owrt-ubi-installer/releases/download/v0.6.5/openwrt-22.03-snapshot-r19338-ae64d0624c-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb)

When I upload the image, I get the following warning.

[image]

I'm unsure if it is safe to ignore the warning in this case. 

I can't afford router downtime, so any advice is highly appreciated.
@dangowrt can you help?

dangowrt, 2022-06-11:
Looks like release candidate images are unsigned and the image you are currently running does signature checks with `ucert`.
To make sure that this is really the cause, please remove the `ucert` package (either in LuCI Software tab or using `opkg remove ucert`) and then try uploading the image again.

jottr, 2022-06-11:
The snapshot image that was previously installed was from this repository.
I can't recall having `ucert` installed myself, but maybe it was added as a dependency?
Anyways, removing `ucert` disabled the warning.
Flashing procedure ran without issues after. I guess I could've simply ignored the warning without removing `ucert` in the first place, no?
Anyways, thanks for chipping in.

closed 2022-06-11 by jottr