dangowrt/owrt-ubi-installer #111: Unable to restore stock firmware in Belkin AX3200

suarezjulian, 2022-07-22:
Hi, first of all thank you for this project! I was able to install OpenWRT and test it out, great instructions!

I completed the backup process and made sure that mtd3 was 125MB and the other files were not empty (they were a few MB each)

Then I installed OpenWRT did my testing and now I want to go back to stock.

I was able to complete all the steps in https://github.com/dangowrt/owrt-ubi-installer#restoring-the-vendorofficial-firmware, until step 5. I did a power cycle but nothing happened

Now the power led is on but the Internet led is flashing orange, but if I go to 192.168.1.1, I get redirected to https://192.168.1.1/cgi-bin/luci/ but it never loads (Error: This site cannot be reached), so I think it is not completely bricked, but I'm not sure how to recover from here

I cannot connect via SSH either

Any help would be greatly appreaciated

suarezjulian, 2022-07-22:
Ok managed to solve it. Turns out the restore process was successful, I just needed to use the correct URL to get to the router admin page https://192.168.1.1 (the https is required), so probably Belkin/Linksys are using a customized version of Luci OpenWRT

closed 2022-07-22 by suarezjulian

dangowrt, 2022-07-23:
Good to read that you managed and the device is all fine :+1: 
The issue was probably also related to the browser cache which still contained the redirect to LuCI which came from OpenWrt running on the device. I guess I should add "clear browser cache" to the instructions...

austonpramodh, 2023-05-27:
Hi @dangowrt,

I have lost the backup files for RT3200. Would be great if I can get a way to restore my router to stock without the backup files. Please help me out. 

Thanks.